Koenkk/zigbee2mqtt #18961: Innr RS 227 T not working after restart

FWeinb, 2023-09-12:
### What happened?

Similar as #15528, the Innr RS 227 T will stop working when restarting zigbee2mqtt, but strangely only when state was changed before zigbee2mqtt was restarted. 

### What did you expect to happen?

Like any other device, it should be possible to just use your devices without having to restart zigbee2mqtt before you can configure them

### How to reproduce it (minimal and precise)

1. Add a Innr RS 227 T to Zigbee2mqtt
2. Turn it on / off 
3. Restart Zigbee2mqtt
4. Innr RS 227 T will stop working 

If you skip step 2. and restart zigbee2mqtt directly after the interview, everything works fine even after restarts. 

### Zigbee2MQTT version

1.33.0

### Adapter firmware version

0x26780700

### Adapter

ConBee2

FWeinb, 2023-09-13:
Found out that if I use the "Dev Console" i can at least get information (like current brightness) from the device if I query:
`Endpoint: 1, Cluster: LevelCtrl, Attribute: currentLevel` 
but if I try to use the refresh button on the "Exposes" Tab, I get a Timeout

julianfs, 2023-09-13:
Possibly also related to #12985 & #16100

FWeinb, 2023-09-13:
I did some further investigating, when the bulbs are initially added they don't have the Endpoint 0 in the cluster overview in zigbee2mqtt, there is only Endpoint 1 and 242. Another curious thing is that interviewing is not creating any bind/reporting configuration for these bulbs. 

Reading the [this configuration](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/innr.ts#L314-L321) i find it really hard to find the spot where the endpoint to be interviewed is specified, it looks like it is done in dynamically in [libs/(light.ts#L84-L99](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/lib/light.ts#L84-L99) but I don't really see the bind/reporting configuration setup.

Koenkk, 2023-09-14:
@FWeinb can you provide the herdsman debug log starting from the point where you pair the bulb until endpoint 0 appears?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

FWeinb, 2023-09-14:
@Koenkk thanks for locking into it! I think I found the place where the endpoint 0 appears:

<details>
<summary>Before</summary>
```
zigbee2mqtt      | 2023-09-14T20:13:20.722Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
zigbee2mqtt      | 2023-09-14T20:13:20.722Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
zigbee2mqtt      | 2023-09-14T20:13:20.739Z zigbee-herdsman:deconz:driver query aps data indication
zigbee2mqtt      | 2023-09-14T20:13:20.750Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 44
zigbee2mqtt      | 2023-09-14T20:13:20.753Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 44 srcAddr: 0x3b1d destAddr: 0x0 profile id: 0x82b8 cluster id: 0x7706 lqi: 255
zigbee2mqtt      | 2023-09-14T20:13:20.753Z zigbee-herdsman:deconz:frameParser response payload: 78,165,68,32,87,2
zigbee2mqtt      | 2023-09-14T20:13:20.753Z zigbee-herdsman:deconz:adapter could not parse zclFrame: Error: Cluster '30470' has no command response '87'
zigbee2mqtt      | 2023-09-14T20:13:20.754Z zigbee-herdsman:controller:log Received 'raw' data '{"clusterID":30470,"data":{"type":"Buffer","data":[78,165,68,32,87,2]},"address":15133,"endpoint":0,"linkquality":255,"groupID":null,"wasBroadcast":false,"destinationEndpoint":155}'
```
</details>

```
zigbee2mqtt      | Zigbee2MQTT:debug 2023-09-14 22:13:20: Received Zigbee message from '0x44a54e000082f262', type 'raw', cluster '30470', data '{"data":[78,165,68,32,87,2],"type":"Buffer"}' from endpoint 0 with groupID null
zigbee2mqtt      | Zigbee2MQTT:debug 2023-09-14 22:13:20: No converter available for 'RS 227 T' with cluster '30470' and type 'raw' and data '{"data":[78,165,68,32,87,2],"type":"Buffer"}'
```

<details>
<summary>After</summary>
```
zigbee2mqtt      | 2023-09-14T20:13:20.759Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
zigbee2mqtt      | 2023-09-14T20:13:23.377Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
zigbee2mqtt      | 2023-09-14T20:13:23.378Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
zigbee2mqtt      | 2023-09-14T20:13:23.379Z zigbee-herdsman:deconz:driver query aps data indication
zigbee2mqtt      | 2023-09-14T20:13:23.384Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 45
zigbee2mqtt      | 2023-09-14T20:13:23.387Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 45 srcAddr: 0xbc3 destAddr: 0x0 profile id: 0x104 cluster id: 0xb04 lqi: 255
zigbee2mqtt      | 2023-09-14T20:13:23.388Z zigbee-herdsman:deconz:frameParser response payload: 24,24,10,8,5,33,83,0
zigbee2mqtt      | 2023-09-14T20:13:23.413Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":24,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":1288,"dataType":33,"attrData":83}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":3011,"endpoint":1,"linkquality":255,"groupID":null,"wasBroadcast":false,"destinationEndpoint":1}'
```
</details>

Koenkk, 2023-09-16:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/978cb30d31a6ba15583e9f551de77fd3

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

@shanetuohy please also try this but replace `zigbeeModel: ['RS 227 T'],` with `zigbeeModel: ['BY 266'],`

FWeinb, 2023-09-17:
@Koenkk Thanks for that! Did some (limited) testing now but can confirm that this is fixing this behaviour. Will see if they keep working.

julianfs, 2023-09-17:
A 'BY 266' user here. I restarted z2m without making the external converter change & found I could *not* turn on/off the light. 

After adding the external converter above & restarting z2m the light was working again.

Thanks for this fix, it's looking good so far.

Koenkk, 2023-09-17:
Integrated the fix! You can remove the ext converter after updating to the next release.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-09-17 by Koenkk

julianfs, 2023-09-17:
Just for reference, this also fixes an issue where I could not update firmware on a bulb, that also now works as expected.

Slx-hub, 2023-10-30:
Same Issue with the [RB 266](https://www.zigbee2mqtt.io/devices/RB_266.html)
That fix works like a charm!

Koenkk, 2023-10-30:
@Slx-hub integrated the fix!