Koenkk/zigbee2mqtt #18963: [New device support]: TS0601,[_TZE204_mhxn2jso]

MasterFeige, 2023-09-13:
### Link

https://www.aliexpress.com/item/1005005590094557.html

### Database entry

{"id":7,"type":"Router","ieeeAddr":"0xa4c1387b77e1a0c5","nwkAddr":9648,"manufId":4417,"manufName":"_TZE204_mhxn2jso","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�@n,i\u001fAn,i An,i!An,i"An,i3An,i8An,i�@n,i","65506":56,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE204_mhxn2jso","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1692079890196,"defaultSendRequestWhen":"immediate"}

### Comments

<html><body>
<h2 dir="auto">Here is the dp point information I obtained from the supplier:</h2>

DP ID | 标识符 | 数据传输类型 | 数据类型 | 功能点属性
-- | -- | -- | -- | --
1 | presence_state | 只上报（ro） | enum | 枚举值: none, presence
12 | presence_time | 可下发可上报（rw） | value | 数值范围: 1-3600,        间距: 1, 倍数: 0,        单位: s
19 | dis_current | 只上报（ro） | value | 数值范围: 0-1000,        间距: 1, 倍数: 0,        单位: cm
20 | illuminance_value | 只上报（ro） | value | 数值范围: 0-10000,        间距: 1, 倍数: 0,        单位: lux
101 | sensitivity | 可下发可上报（rw） | value | 数值范围: 0-10,        间距: 1, 倍数: 0,        单位:
102 | presence_delay | 可下发可上报（rw） | value | 数值范围: 5-3600,        间距: 1, 倍数: 0,        单位: s
111 | minimum_range | 可下发可上报（rw） | value | 数值范围: 0-1000,        间距: 50, 倍数: 0,        单位: cm
112 | maximum_range | 可下发可上报（rw） | value | 数值范围: 50-1000,        间距: 50, 倍数: 0,        单位: cm

</body>
</html>

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [
        {modelID: 'TS0601', manufacturerName: '_TZE204_mhxn2jso'},
    ],
    model: 'rtsc11r', 
    vendor: 'TuYa',
    description: '5.8G human presence sensor with relay',
    configure: tuya.configureMagicPacket,
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, 
    exposes: [
      e.enum('presence_state',ea.STATE,['none','presence']).withDescription('none : did not checkout '),
      e.numeric('dis_current',ea.STATE).withValueMin(0).withValueMax(1000).withValueStep(1)
          .withDescription('Entry distance indentation').withUnit('cm'),
      e.numeric('illuminance_value',ea.STATE).withValueMin(0).withValueMax(10000).withValueStep(1)
          .withDescription('Illumination threshold for switching on').withUnit('lux'),

      e.numeric('presence_time',ea.STATE_SET).withValueMin(1).withValueMax(3600).withValueStep(1)
          .withDescription('Entry filter time').withUnit('s'),
      e.numeric('sensitivity',ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(1)
          .withDescription('Sensitivity of sensors'),
      e.numeric('presence_delay',ea.STATE_SET).withValueMin(5).withValueMax(3600).withValueStep(1)
        .withDescription('Turn off delay').withUnit('s'),
      e.numeric('minimum_range',ea.STATE_SET).withValueMin(0).withValueMax(1000).withValueStep(50)
        .withDescription('Detection range').withUnit('m'),
      e.numeric('maximum_range',ea.STATE_SET).withValueMin(50).withValueMax(1000).withValueStep(50)
        .withDescription('Detection range').withUnit('m'),
    ],
    meta: {
        tuyaDatapoints:[
          [1, 'presence_state', tuya.valueConverterBasic.lookup({'none': tuya.enum(0), 'presence': tuya.enum(1)})],
          [12,'presence_time',tuya.valueConverter.divideBy10],
          [19,'dis_current',tuya.valueConverter.raw],
          [20,'illuminance_value',tuya.valueConverter.raw],
          [101,'sensitivity',tuya.valueConverter.divideBy10],
          [102,'presence_delay',tuya.valueConverter.raw],
          [111,'minimum_range',tuya.valueConverter.divideBy100],
          [112,'maximum_range',tuya.valueConverter.divideBy100],
        ],
    },
};

module.exports = definition;
```

Koenkk, 2023-09-13:
converter looks good, whats not working?

MasterFeige, 2023-09-19:
Thanks for your reply, this script works fine. I want to know, what do I need to do so that consumers can add this product to their home assistant tool and use it directly when they purchase it?looking forward to your answer, thank you very much.

Koenkk, 2023-09-19:
I can add it but can you explain me the `presence_time` and `dis_current` options?

MasterFeige, 2023-09-20:
‘presence_time’ means how long the sensor continues to detect a signal before it is considered someone is present.
'dis_current' (distance_current)means the distance at which the sensor currently detects people.

Koenkk, 2023-09-20:
thanks, added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-09-20 by Koenkk

roastcom, 2023-11-02:
Should this work OK? I copied the code from release into an external converter and when I try to alter any parameters in MQTT I get a message "missing converter".

[image]

The HA isn't even showing most of those parameters, and the luminance cannot be used as condition (only allows detection distance and presence).

MasterFeige, 2023-11-02:
I think you can solve the problem by replacing "presence_delay" in the script with "keep_time".

roastcom, 2023-11-02:
Thank you, I ended up upgrading to the latest docker image so I don't have to keep loads on external converters (I had others too) but that caused a few other things to break (and this one didn't work regardless) so I raised a couple of issues.

roastcom, 2023-11-15:
Fixed all other issues by upgrading HA, but the illuminance on this one still doesn't work - HA just doesn't see Illuminance at all, even though in MQTT it seems to work ok.

[image]

Works the same whether I use the external converter or the 'internal' one (which I suppose are the same as this one had simple been added to MQTT).

3dbluebot, 2023-12-06:
Hi
I wrote a quirk for this device on ZHA (I use both flavours of zigbee on my HA yellow). I see the data points are the same. So this should work. 
I can't get the built in converter to work, which I think is this one.  I think it may be because the sliders use the correct data points, but present the units in a way the device can't handle. It should be noted that not all devices that share model and manufacturer are the same these days. So it was probably great for the guy that wrote it. 

I can tell you about function of the device when it works,  I'm on safer ground ;)

**Presence delay**,  effects function.> _Below 65_ seconds and the device only looks for large movements. _65 seconds_ and above and it detects breathing within the central detection cone.

It only reports  **current distance** with any acuracy, within the central detection cone. I've never known the device to report a value and not trigger. Unless outside the min max distance.  And this min max function usually doesn't work (see below).

A minimum detection delay below 1 second, or with decimal point,  can make it crash. It only seems to accept full seconds

All distance are expected and reported back in full centermeteres. All times likewise in complete seconds. For the device I worked on,  otherwise things broke. 

Min and Max detection distance,  don't work correctly in Tuya. This is not in HA, but with the device attached to a proper Tuya gateway (that's how I get data points etc via their developer portal)  so I think the firmware is broken. 

Hope this helps.  I'm going to try my hand at tweaking this file, it looks great, and a good starting point for a noob of a few months like me.

roastcom, 2023-12-06:
> **Presence delay**, effects function.> _Below 65_ seconds and the device only looks for large movements. _65 seconds_ and above and it detects breathing within the central detection cone.

Thank you, that is so weird! 
What does the sensitivity do then?

roastcom, 2023-12-06:
> Just need to find out how to use the file.

If you mean how to run the manual decoder then easy:
In z2m config folder create a file with ".js" extension, copy the contents into that file, and in the configuration.yaml (z2m) add

external_converters:
  - filename.js

After that restart z2m and it will start using the external converter from that point.

That is at least if you run everything in docker like I do, if you use HASSOS and m2q as a plugin, then I'm not quite sure. But they probably have that described in the manuals somewhere. The main idea is that you want to run an external converter for that device.

3dbluebot, 2023-12-06:
Thank you,  sounds refreshingly easy

3dbluebot, 2023-12-08:
> I am yet to test it on a Tuya hub (probably won't even bother to be
> honest) but the behaviour on z2m is definitely different for me. I think
> the overall issue is that the detecting hardware/firmware is just
> unreliable, which in the end causes all sorts of issues.
>
> First, all entities and settings are more or less properly supported (both
> in z2m and HA), which isn't a big surprise.
>
> The *min/max distance* definitely works for me most of the time - it
> affect the presence state value, if the detection distance is not in the
> given interval the presence state doesn't change. The problem for me is
> that I saw 'presence' switched on 0m detections when I had 50m as min, and
> more often I saw detections at a distance within the interval but presence
> didn't change.
>
> The *detection delay* - a bit murky that one. I suspect it's the one
> responsible for those unregistered detections above. The device seems a bit
> more 'responsive' at 0 and less responsive as the value increases. My guess
> is that is for how long the motion should be detected before the presence
> state changes. It doesn't seem like a linear dependency, but less is less
> and more is more.
>
> *Keep time* - definitely affect the length of time the detected presence
> state lasts. Again, doesn't seem like a 1-to-1 relationship, but close
> within 5-10 sec. Minimum is 5 and at that value some of devices (I have a
> [... 15 lines cut ...]
> might be ok, or if you wanna make sure a rat doesn't crawl through a vent
> (although I'm not quite sure it would pick up a rat).
> Under realistic conditions detections are pretty unreliable, and drop to
> near zero in the dark for some reason. I have no clue why, all of mine just
> turn near blind in the dark.

Hi, that sucks.

I wonder if we have the same format (mine are long thin angled rectangles,
good for fitting in corners, not round, as shown in 2M).
I have multiple  manufacturers and they work the same way.  My devices are
totally reliable for detection.  But that's under ZHA using my own quirk.
Everything works reliably except min max value, which is erratic. But
luckily I don't need that to be reliable. Maybe the erratic nature is just
the inaccuracy you speak of.

It does need a presence delay (not detection delay) of over 65 seconds, so
that it detects me sleeping on the couch though. But that's also in the
instructions (on the one device that had instructions) as how to get
breathing / micro detection working.

Detection delay, is accurate and repeatable on the devices I use in ZHA.
I'm sure they will be on 2M also, once I load up the converter on this
page. As it will only need minor changes for one device.

Interestingly the manufacturer version named on this page does not support
the presence countdown data point according to Tuya development platform.

The other manufacturer version of the same rectangular format, reports the
presence countdown, and it is 100% accurate.You can see you get exactly
what you set in presence delay. Very handy.

I think you mentioned presence delay is not accurate on this version, but I
may be confused.  If that is what you said, that's disappointing.   I
wonder if that indicates an issue with this version?
Thanks for the heads up.

I'm still using the inbuilt converter from zigbee2mqtt for this version
though, and frankly the sensor doesnt work well enough for me to judge if
anything works as intended. It crashes my device repeatedly, but then it is
also being fed 10ths of seconds, not seconds, and the distance is setable
by 1cm, not 50cm.  This screws the versions I own.

It should be noted that the picture for the device in 2M shows a round
device, mounted on a short stalk, which is a different format to mine,  so
maybe the wrong converter is loading. It recognised my device model and
manufacturer correctly though,  so it's odd it's shows,  and links to a
different physical model.

I hope the converter version on this page will work better. It is very
close to what I would expect, but I'm new to 2M, so don't read anything
into that.

As you say the lux reading (the only thing that works reliably) appears to
be presented as a raw value,  not calculated by the converter to a lux
value, another frustration for me.

Thanks for filling me in.

roastcom, 2023-12-08:
From what I've seen m2q rarely has correct pictures of niche devices.

The ones I have are these:
[image]
https://www.aliexpress.com/item/1005005590094557.html

I know you have high hopes for this converter, but the way it works in m2q is the external converter is released first and later it's just committed into the 'master branch'. 
You can look up the actual code here, it's nearly identical to this converter (as it should be)
https://github.com/Koenkk/zigbee-herdsman-converters/commit/54ffe76c8e6ad6d22c402d16b525cd4d273d3c0d

I am surprised though that you say the device works well in ZHA (perhaps you have a different make?). From what I've surmised it's really the device that seems to be the issue - not the drivers. I've abandoned ZHA early on back when it was noticeably lagging in device support (and because it also seemed much easier for me to make a custom converted in z2m by myself), but maybe I should give it another try...

roastcom, 2023-12-11:
One other thing I noticed - these sensors smash HA database like there's no tomorrow. They make 5 updates **per second**, I had to completely remove them from the recorder.

3dbluebot, 2023-12-11:
I knew it was bad,  but I didn't know it was that bad.
They are not well designed, or at least the firmware is flaky.   I have one
per room.  So I ended up creating an extra network just for them. So my
yellow has one ZHA and now two zigbee2mqtt, rather than one.   Luckily it
all works like well oiled clockwork.

Your solution sounds much better, but I'm very much still learning, hence
my brute force approach  😅

roastcom, 2023-12-12:
Pretty simple, this is what I put in the configuration.yaml:

`recorder:
  auto_purge: true
  auto_repack: true
  purge_keep_days: 5
  exclude:
    entity_globs:
      - binary_sensor.*presence*
      - sensor.sun*
      - weather.*
`

3dbluebot, 2023-12-12:
I am learning so much, thank you for your help.
I'm really enjoying my move from Samsung to Homeassistant

roastcom, 2023-12-15:
To capture all entities from those devices you probably would need to extend to 
```
      - binary_sensor.*presence*  
      - sensor.*presence*
      - number.*presence*
``` 
where *presence* fits the names of those devices.

PS: I'm pretty new to the whole HA thing myself, not even a year in. Honestly at first it was quite infuriating at times. :)

3dbluebot, 2023-12-15:
I really appreciate the help,  I read as much as I can.  But it's nuggets
like yours that push me in the correct direction, preventing me reading the
wrong stuff👍

thepecca, 2023-12-26:
> Hi I wrote a quirk for this device on ZHA (I use both flavours of zigbee on my HA yellow). I

@3dbluebot I know this might not be the best repo to share it, but would you mind linking your ZHA Quirk for this device? I've tried adapting multiple ones without success.

Thanks in advance!

kaischmerer, 2024-02-17:
https://www.zigbee2mqtt.io/devices/rtsc11r.html
_Minimum range (numeric)

Minimum detection range. Value can be found in the published state on the minimum_range property. It's not possible to read (/get) this value. To write (/set) a value publish a message to topic zigbee2mqtt/FRIENDLY_NAME/set with payload {"minimum_range": NEW_VALUE}. The minimal value is 0 and the maximum value is 1000. The unit of this value is m.
Maximum range (numeric)

Maximum detection range. Value can be found in the published state on the maximum_range property. It's not possible to read (/get) this value. To write (/set) a value publish a message to topic zigbee2mqtt/FRIENDLY_NAME/set with payload {"maximum_range": NEW_VALUE}. The minimal value is 50 and the maximum value is 1000. The unit of this value is m._

>The unit of this value is m
Should be cm not meter.

checkiecheck, 2025-02-16:
> Hi I wrote a quirk for this device on ZHA (I use both flavours of zigbee on my HA yellow). I see the data points are the same. So this should work. I can't get the built in converter to work, which I thought was this one. It is not, this one uses units of 1 second. The built in version used tenths of a second. It should be noted that not all devices that share model and manufacturer are the same these days. So it was probably great for the guy that wrote in built version.
> 
> I can tell you about function of the device when it works, I'm on safer ground ;)
> 
> **Presence delay**, effects function.> _Below 65_ seconds and the device only looks for large movements. _65 seconds_ and above and it detects breathing within the central detection cone.
> 
> **current distance** is only reported with any acuracy, within the central detection cone. I've never known the device to report a _current distance_ value and not trigger. Unless _outside the min max distance,_ or if you have a _presence delay above one second_. Note min and max detection distance function usually doesn't work (see below).
> 
> **detection delay**, how long it detects you before reporting your presence, can't be below 1 second, or with decimal point, as this can make the device crash. It only seems to accept full seconds
> 
> All distance are expected and reported back in full centermeteres. All times likewise in complete seconds. For the device I worked on, otherwise things broke.
> 
> **Min and Max detection distance**, don't work correctly for this device (for me) even in Tuya. This is not in an HA integration, but with the device attached to a proper Tuya gateway (that's how I get data points etc via their developer portal) so I think the firmware is broken.
> 
> **Sensitivity** This goes from 10 (least sensitive) to 0 (most sensitive). I use 3-4, if I have the _presence delay_ at 65 seconds or above. And 1-3 for below 65 seconds
> 
> Hope this helps. I'm going to try my hand at installing this file, it looks great, and a good starting point for a noob of a few months like me.

any info where that quirk might be? I had this device for a year on the shelf , now i wanted to see if i can make it work.
I'm in the process gathering all kinds of code/scripts for knowledge building for getting more devices to work once i understand more of the ecosystem