Koenkk/zigbee2mqtt #18964: [Feature request]: a branch that supports new devices for Home Assistant < 2023.08

cocoabox, 2023-09-13:
### Is your feature request related to a problem? Please describe

https://github.com/Koenkk/zigbee2mqtt/issues/18445 and merges related have rendered zigbee2mqtt>=1.33.0 to be incompatible with HomeAssistant < 2023.08 ; however many users including me are still using 2023.08 as a set-and-forget setup.

### Describe the solution you'd like

Can there be a branch that continues to support new devices but is compatible with HomeAssistant < 2023.08 ?

### Describe alternatives you've considered

I have considered upgrading HomeAssistant but newer versions do not work well with mqtt+tls. MQTT Configuration done in configuration.yaml have been deprecated and all mqtt configuration needs to be done from the UI. When setting up MQTT+TLS from the UI, it gives the following error:

[image]
 
I hence could not upgrade my HomeAssistant to a newer version

### Additional context

n/a

Drafteed, 2023-09-21:
You can try manually override `name` for HA devices in **configuration.yaml** to match the old requirements:

```yaml
devices:
  "0x12345678":
    friendly_name: my_light
    homeassistant:
      light:
        name: "Light"
  "0x12345678":
    friendly_name: my_switch
    homeassistant:
      switch_left:
        name: "Switch Left"
  "0x1112231":
    friendly_name: my_temperature
    homeassistant:
      temperature:
        name: "Temperature"
```

See https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#overriding-discovery-properties

Hypfer, 2023-09-27:
> Can there be a branch that continues to support new devices but is compatible with HomeAssistant < 2023.08 ?

If you look at the issue tracker, it's already quite obvious that there are not enough resources to even keep up with the demands of the main branch. The only way stuff like that will happen is if you start maintaining that branch and backport the changes yourself.

Koenkk, 2023-09-27:
@Hypfer off topic: didn't expect you here, just wanted to let you know I'm a happy Valetudo user! ❤️

closed 2023-11-04 by github-actions[bot]