Koenkk/zigbee2mqtt #18965: Frontend not launching/accessible

drpau, 2023-09-13:
### What happened?

When browsing to http://10.0.1.5:8080 the webpage doesnt load

### What did you expect to happen?

i expected that the webpage would load

### How to reproduce it (minimal and precise)

1. recreate docker image and allow it to start
2. try to launch front end (http://10.0.1.5:8080) 

- Confirmed in `configuration.yaml` that `frontend: true` is on
- Confirmed the correct port is published in docker
- Confirmed other docker images are working fine
- Confirmed that functionality of mqtt requests are being sent and issued, but the webpage isnt loading.

Nothing in logs about access to website being blocked/dropped connections.

### Zigbee2MQTT version

:latest

### Adapter firmware version

unable to access

### Adapter

CC2652

Hypfer, 2023-09-27:
Do you really expect the maintainer to debug your network setup?

j3kz, 2023-10-21:
Same problem here frontend seems ready but not reachable.

`docker compose logs zigbee2mqtt`
```
zigbee2mqtt  | Using '/app/data' as data directory
zigbee2mqtt  | Zigbee2MQTT:info  2023-10-20 15:32:21: Logging to console and directory: '/app/data/log/2023-10-20.15-32-21' filename: log.txt
zigbee2mqtt  | Zigbee2MQTT:info  2023-10-20 15:32:21: Starting Zigbee2MQTT version 1.33.0 (commit #7ee207f)
zigbee2mqtt  | Zigbee2MQTT:info  2023-10-20 15:32:21: Starting zigbee-herdsman (0.18.5)
zigbee2mqtt  | Zigbee2MQTT:info  2023-10-20 15:32:21: zigbee-herdsman started (resumed)
```

`docker compose exec zigbee2mqtt wget --spider http://localhost:8080`
```
Connecting to localhost:8080 (127.0.0.1:8080)
wget: can't connect to remote host (127.0.0.1): Connection refused
```

exabird, 2024-05-11:
Same error for me now i dont find any solution ...

dmike3, 2024-07-22:
You ever figure out why the frontend isn't responding?

closed 2025-01-13 by github-actions[bot]