Koenkk/zigbee2mqtt #18968: Invalid color mode/value in logs

chench00, 2023-09-13:
### What happened?

Invalid or incomplete color value '{'brightness': 254, 'color_mode': 'xy', 'linkquality': 65, 'state': 'OFF', 'update': {'installed_version': [1057832](tel:1057832), 'latest_version': [1057832](tel:1057832), 'state': 'idle'}, 'update_available': None}' received for entity light.garden
Invalid color mode 'xy' received for entity light.sink

### What did you expect to happen?

Nothing in logs. 

### How to reproduce it (minimal and precise)

I don’t think I did anything?

### Zigbee2MQTT version

1.33.0

### Adapter firmware version

20230507

### Adapter

Sonoff P

Hypfer, 2023-09-27:
Can confirm the same thing happening with https://github.com/Koenkk/zigbee2mqtt/commit/a7bb3fc

I created a new zigbee group, joined two identical hue signe lights and was faced with no proper state reporting for the home assistant entity. It just stayed at 0% brightness because of that error message in the HA logs.

This was solved by setting a color to that group light entity via HA. 
With that, the state reporting started to work

larronni, 2023-10-12:
My HA log is full of "invalid color mode" messages... It's not only groups, but individual lights as well. 

---
Logger: homeassistant.components.mqtt.light.schema_json
Source: components/mqtt/light/schema_json.py:285
Integration: MQTT (documentation, issues)
First occurred: 8 October 2023 at 09:13:14 (7303 occurrences)
Last logged: 09:40:39

Invalid color mode 'xy' received for entity light.0xf082c0fffe0fa143
Invalid color mode 'xy' received for entity light.0xf082c0fffe0fa041
Invalid color mode 'xy' received for entity light.0xf4b3b1fffe9894e7
Invalid color mode 'xy' received for entity light.0xf082c0fffe13d5ec
Invalid color mode 'xy' received for entity light.mirror

MichaelRuzicka, 2023-10-16:
Same here.

Adapter fw version
20230507

Adapter
Sonoff-P

Zigbee2MQTT version
1.33.1

asfalots, 2023-10-22:
Same here.

Zigbee2MQTT version: [1.33.1]
Coordinator type: EZSP v11 
Coordinator revision: 7.2.2.0 build 190 (HA skyconnect)

larronni, 2023-10-26:
Looks like (I do not know why/how) that the state changes from "color_mode": "color_temp" to "color_mode": "xy"
As long as the state says "color_temp" it is not a problem, but as soon as it changes to "xy" the trouble starts...

The issue is on several types of bulbs/lights. Among IKEA LED2101G4 and IKEA L1527.

Are there some logs you need to look into this, or is it something we can try to set in Z2M to solve this issue?

sirjmann92, 2023-11-06:
I'm seeing the same thing in my logs, for 2 out of 4 Sylvania LED panels, and 1 out of 3 Hue bulbs.

`2023-11-05 18:12:31.872 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid or incomplete color value '{'color_mode': 'xy', 'linkquality': 83, 'power_on_behavior': None, 'state': 'OFF', 'update': {'installed_version': 16785162, 'latest_version': 16785162, 'state': 'idle'}, 'update_available': None}' received for entity light.hue_bulb_01
2023-11-05 18:12:31.906 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid or incomplete color value '{'brightness': 254, 'color_mode': 'xy', 'level_config': {'on_level': 'previous'}, 'linkquality': 102, 'state': 'OFF'}' received for entity light.led_panel_02
2023-11-05 18:12:31.909 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid or incomplete color value '{'brightness': 254, 'color_mode': 'xy', 'level_config': {'on_level': 'previous'}, 'linkquality': 87, 'state': 'OFF'}' received for entity light.led_panel_03`

Heggeg, 2024-01-27:
Same for me 

I use this lights:

https://www.zigbee2mqtt.io/devices/50049_500.63.html#paulmann-50049%252F500.63
https://www.amazon.de/gp/product/B071WZZBCD/

All lights are in groups which we control with Hue Remote Controls 
https://www.zigbee2mqtt.io/devices/324131092621.html#philips-324131092621

Latest stable available HA Version and z2m is running in my homelab. 
HA:
- Core 2024.1.5
- Supervisor 2023.12.1
- Operating System 11.4
- Frontend 20240104.0

z2m:

- Current version: 1.35.1-1

`
2024-01-27 10:42:53.639 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid or incomplete color value '{'brightness': 254, 'color_mode': 'xy', 'last_seen': '2024-01-27T03:14:33+01:00', 'linkquality': 168, 'power_on_behavior': None, 'state': 'OFF'}' received for entity light.kuche_deckenlampe
2024-01-27 10:42:53.669 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid or incomplete color value '{'brightness': 254, 'color_mode': 'xy', 'last_seen': '2024-01-27T09:50:35+01:00', 'linkquality': 108, 'power_on_behavior': None, 'state': 'OFF'}' received for entity light.kuche_esstisch_deckenlampe
2024-01-27 10:42:53.670 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid or incomplete color value '{'brightness': 254, 'color_mode': 'xy', 'last_seen': '2024-01-26T21:40:16+01:00', 'linkquality': 148, 'power_on_behavior': None, 'state': 'OFF'}' received for entity light.schlafzimmer_deckenlampe_links
2024-01-27 10:42:53.673 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid or incomplete color value '{'brightness': 254, 'color_mode': 'xy', 'last_seen': '2024-01-27T09:48:53+01:00', 'linkquality': 124, 'power_on_behavior': None, 'state': 'OFF'}' received for entity light.flur_oben_deckenlampe
2024-01-27 10:42:53.674 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid color mode 'hs' received for entity light.schlafzimmer_deckenlampe_rechts
2024-01-27 10:42:53.674 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid or incomplete color value '{'brightness': 254, 'color_mode': 'xy', 'last_seen': '2024-01-27T09:50:42+01:00', 'linkquality': 172, 'power_on_behavior': None, 'state': 'OFF'}' received for entity light.flur_unten_deckenlampe_hinten
2024-01-27 10:42:53.675 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid or incomplete color value '{'brightness': 254, 'color_mode': 'xy', 'last_seen': '2024-01-26T17:03:13+01:00', 'linkquality': 124, 'power_on_behavior': None, 'state': 'OFF'}' received for entity light.flur_unten_deckenlampe_vorne
2024-01-27 10:42:53.676 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid or incomplete color value '{'brightness': 254, 'color_mode': 'xy', 'last_seen': '2024-01-27T03:14:33+01:00', 'linkquality': 164, 'power_on_behavior': None, 'state': 'OFF'}' received for entity light.wohnzimmer_deckenlampe
2024-01-27 10:42:53.677 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid or incomplete color value '{'brightness': 97, 'color_mode': 'xy', 'last_seen': '2024-01-27T03:14:33+01:00', 'linkquality': 144, 'power_on_behavior': None, 'state': 'OFF'}' received for entity light.kuche_stripe
2024-01-27 10:42:53.696 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid or incomplete color value '{'brightness': 254, 'color_mode': 'xy', 'state': 'OFF'}' received for entity light.flur_unten_lampen
2024-01-27 10:42:53.697 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid or incomplete color value '{'brightness': 254, 'color_mode': 'xy', 'state': 'OFF'}' received for entity light.flur_oben_lampen
2024-01-27 10:48:54.246 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid or incomplete color value '{'brightness': 254, 'color_mode': 'xy', 'state': 'OFF'}' received for entity light.flur_oben_lampen
2024-01-27 10:48:54.246 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid or incomplete color value '{'brightness': 254, 'color_mode': 'xy', 'last_seen': '2024-01-27T10:48:54+01:00', 'linkquality': 128, 'power_on_behavior': None, 'state': 'OFF'}' received for entity light.flur_oben_deckenlampe
`

Milfeldt, 2024-02-06:
I'm using Z2M, had two HUE bulps and a HUE strip, both with the same 'fault' 
In Setting (specific) set color sync to true.
Problem solved :slight_smile:

Chaoscontrol, 2024-02-06:
Same here. Device worked fine for a year. I activated the "Color sync" option after a reset and started getting these errors repeatedly. Came looking for an answer. I disabled the option now, and I think it will go away.

dajomas, 2024-02-15:
Z2M version: [1.35.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.35.0) commit: [b5a90e92](https://github.com/Koenkk/zigbee2mqtt/commit/b5a90e92)
HA Core: 2024.2.0

Same issue with Hue bulbs (color) and IKEA Tradfri (white)

dajomas, 2024-08-14:
Funny that for one person, setting Color Sync to true works and for the other person setting Color Sync to false works and then for me, neither works....

And after half a year, this issue still exists.

VanDoni, 2024-08-15:
Same issue here. For me it seems to occur with Gledopto devices.

hardwareadictos, 2024-08-20:
Same here with two phillips ambiance LED strips

corchat, 2024-08-24:
I had the same problem. What solved it for me was setting the power_on_behaviour. I had to do this in home assistant as well as zigbee2mqtt

henriec, 2024-08-29:
I have the same issue. At least for one ikea color bulb. On the tab settings(specific) there are two options "color sync" and "state action" which seem to have no value.. When i click on true/false the value is set. However, while this is fixing the problem for others (remove the warning from the logs) this does not change it for me. Whichever value is set: i still get the error `[homeassistant.components.mqtt.light.schema_json] Invalid or incomplete color value '{'brightness': 66, 'color_mode': 'xy', 'color_options': None, 'last_seen': '2024-08-29T09:00:55.314Z', 'linkquality': 96, 'power_on_behavior': None, 'state': 'OFF', 'update': {'installed_version': 268572245, 'latest_version': 268572245, 'state': 'idle'}, 'update_available': None}' received for entity light.0xe..........`

dajomas, 2024-08-30:
> After this: my conclusion is that the error could be thrown when one of the "settings" has an empty or unknown value....

Ran a test on this and I agree, tried it out with a couple of different brands and looking at the "Exposes" tab and filling in the empty/unknown/unselected values solved the issue. In fact, setting a value for only "Color temp" and leaving "Color temp startup" and "Power-on behavior" alone (and any other unset value), fixed the issue for me.

So question to the developer: Can you ensure a default value is set for fields that are unset? (Like default neutral for Color temp, for instance)

Merril-r, 2024-10-24:
I am having the same issue with several bulbs

lukbe, 2025-03-06:
Hello,
after update to 2.1.3-1 I have the same problem.

2025-03-06 20:51:19.163 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid color mode 'xy' received for entity light.led_dekorace_tuya

closed 2025-05-13 by github-actions[bot]