Koenkk/zigbee2mqtt #18973: Continously restant the addon

petoblat, 2023-09-13:
### What happened?

The service continously restart After more or less 1hr after the update 

### Zigbee2MQTT version

1.33.0-1

### Adapter firmware version

20210708

### Adapter

zStack3x0

mijnmodelbaan, 2023-09-17:
Same here. After updating (via openHAB) every 6 seconds zigbee2mqtt is restarting itself. Herdsman doesn't recognise my coordinator anymore, although the same coordinator has been working error free for about a year. I'll try an older version.

lux73, 2023-09-21:
without any Logfiles there is nobody here who could help

mijnmodelbaan, 2023-09-25:
> without any Logfiles there is nobody here who could help

I know, but I was hoping there would be lots of people with the same problem, more or less, so it was an obvious problem. I tried a lot of solutions given in the OpenHAB forum, Googled a lot, changed ports and settings, all to no avail. I have a headless Pi, so all is going via SSH. And as I'm not a real Linux user a lot of searching is going into finding the correct commands to get things done. The only solution I get from the Z2M website is flashing my coordinator, but why would I do that if the coordinator is working fine if I install the Zigbee binding, is working fine if I create a Things file for it and started to give me problems after updating Z2M to the latest version. Unfortunately downgrading Z2M to a version from last year didn't work so in my perspective the upgrade changed something which doesn't get changed back with the downgrade. I'm looking through all the updates to see if there is some breaking change regarding the coordinator but didn't find one, yet.
In the meantime I stumbled about Home Assistant and decided to give it a try. It worked straight from the box, so to speak, and after installing ZHA all my Zigbee devices are working too. With the same hardware, no coordinator flashing necessary.

closed 2023-11-02 by github-actions[bot]