Koenkk/zigbee2mqtt #18977: PJ-MGW1203 - does not report energy, all other fields are OK

papperone, 2023-09-14:
### What happened?

I just bought this device (https://it.aliexpress.com/item/1005004948381800.html) and connected to test its functionality, all seems ok, I can see all values exposed with right numbers but "energy" is always "null", even with a load; I left it working from more than 1h and no changes...

### What did you expect to happen?

I expected "energy" value to appears after a while

### Zigbee2MQTT version

1.33.0

### Adapter firmware version

6.7.9.0 build 405

### Adapter

ZB-GW04 based on EFR32MG21

### Debug log

this is the paylog I keep getting
payload '{"current":0.42,"energy":null,"linkquality":152,"power":94.3,"voltage":227.1}'

closed 2023-10-22 by github-actions[bot]

JanButter, 2024-03-09:
Hi, i have the same.  Connected two of these devices and voltage, current and power are working but the amount of energy is not calculated.  
[image]

Poupi1972, 2024-03-09:
Hello
Same issue for me
Is someone have a solution
Regards

Douglasfc, 2024-05-03:
Same issue here.
Hope someone can help us.

BR

wolfwander, 2024-10-02:
Same issue here, it seems it is wrongly calculated.