Koenkk/zigbee2mqtt #18979: PTM 215Z missing action

joinn1710, 2023-09-14:
### What happened?

I am running z2m standalone with a separate mosquitto server and home assistant instance.
the home assistant integration is active
maybe unrelated, but i got a warning in home assistant that after HA update 2024.9, it was unsupported to have the same name for the device and entity or something, don't remember fully.
Anyway, because of that, I tried to see if there was something wrong, and noticed both legacy options in the home assistant integrations were on. i disabled them, and the day after, the switches did not work. everything else works the same, but all three of my ptm 215Z switches stopped working
Using an MQTT explorer, i found that the switches exist in the homeassistant topic with lqi, but not action, and it does not get updated. most likely, it is the values recieved when pairing. when i press a button, it shows up in the zigbee2mqtt topic instead of the homeassistant topic with both action and lqi.
the attached log is from pressing all 4 buttons after removing, resetting, and re-pairing one switch.

### What did you expect to happen?

i expected the switch to show up in the sensor topic under the homeassistant topic with action and link quality.

### How to reproduce it (minimal and precise)

pair any switch containing an EnOcean PTM 215Z switch in a standalone Z2M installation with the home assistant integration enabled

### Zigbee2MQTT version

1.32.2

### Adapter firmware version

7.1.1.0 build 273

### Adapter

Home assistant SkyConnect

### Debug log

info 2023-09-14 21:57:11MQTT publish: topic 'zigbee2mqtt/0x0000000001737d02/action', payload 'press_1'
info 2023-09-14 21:57:11MQTT publish: topic 'zigbee2mqtt/0x0000000001737d02/linkquality', payload '211'
info 2023-09-14 21:57:11MQTT publish: topic 'zigbee2mqtt/0x0000000001737d02/action', payload 'press_1'
info 2023-09-14 21:57:11MQTT publish: topic 'zigbee2mqtt/0x0000000001737d02/action', payload 'release_1'
info 2023-09-14 21:57:11MQTT publish: topic 'zigbee2mqtt/0x0000000001737d02/linkquality', payload '211'
info 2023-09-14 21:57:11MQTT publish: topic 'zigbee2mqtt/0x0000000001737d02/action', payload 'release_1'
info 2023-09-14 21:57:12MQTT publish: topic 'zigbee2mqtt/0x0000000001737d02/action', payload 'press_3'
info 2023-09-14 21:57:12MQTT publish: topic 'zigbee2mqtt/0x0000000001737d02/linkquality', payload '212'
info 2023-09-14 21:57:12MQTT publish: topic 'zigbee2mqtt/0x0000000001737d02/action', payload 'press_3'
info 2023-09-14 21:57:12MQTT publish: topic 'zigbee2mqtt/0x0000000001737d02/action', payload 'release_3'
info 2023-09-14 21:57:12MQTT publish: topic 'zigbee2mqtt/0x0000000001737d02/linkquality', payload '212'
info 2023-09-14 21:57:12MQTT publish: topic 'zigbee2mqtt/0x0000000001737d02/action', payload 'release_3'
info 2023-09-14 21:57:13MQTT publish: topic 'zigbee2mqtt/0x0000000001737d02/action', payload 'press_2'
info 2023-09-14 21:57:13MQTT publish: topic 'zigbee2mqtt/0x0000000001737d02/linkquality', payload '210'
info 2023-09-14 21:57:13MQTT publish: topic 'zigbee2mqtt/0x0000000001737d02/action', payload 'press_2'
info 2023-09-14 21:57:13MQTT publish: topic 'zigbee2mqtt/0x0000000001737d02/action', payload 'release_2'
info 2023-09-14 21:57:13MQTT publish: topic 'zigbee2mqtt/0x0000000001737d02/linkquality', payload '210'
info 2023-09-14 21:57:13MQTT publish: topic 'zigbee2mqtt/0x0000000001737d02/action', payload 'release_2'
info 2023-09-14 21:57:13MQTT publish: topic 'zigbee2mqtt/0x0000000001737d02/action', payload 'press_4'
info 2023-09-14 21:57:13MQTT publish: topic 'zigbee2mqtt/0x0000000001737d02/linkquality', payload '209'
info 2023-09-14 21:57:13MQTT publish: topic 'zigbee2mqtt/0x0000000001737d02/action', payload 'press_4'
info 2023-09-14 21:57:13MQTT publish: topic 'zigbee2mqtt/0x0000000001737d02/action', payload 'release_4'
info 2023-09-14 21:57:13MQTT publish: topic 'zigbee2mqtt/0x0000000001737d02/linkquality', payload '209'
info 2023-09-14 21:57:13MQTT publish: topic 'zigbee2mqtt/0x0000000001737d02/action', payload 'release_4'

blastrock, 2023-09-19:
Same issue here on [1.33.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.33.0) commit: [7ee207f](https://github.com/Koenkk/zigbee2mqtt/commit/7ee207f). The device works fine and the events are published on mqtt, but not on the homeassistant topic.

blastrock, 2023-09-24:
I investigated this, it all comes down to [this line](https://github.com/Koenkk/zigbee2mqtt/blob/f224ef7b7c9ba1cc6819c3b7bf84119d7ce5ec9b/lib/extension/homeassistant.ts#L1197). So I'm not sure it's a bug or intended behavior, but enabling `legacy_triggers` in the config fixed the issue for me.

In `configuration.yaml`:

```yaml
homeassistant:
  legacy_triggers: true
```

closed 2023-11-01 by github-actions[bot]