Koenkk/zigbee2mqtt #18980: [New device support]: Maginon Smart thermostat WT-1 - TS0601

groows, 2023-09-14:
### Link

https://maginon.de/products/smartes-thermostat-wt-1-100950

### Database entry

{"id":21,"type":"EndDevice","ieeeAddr":"0xa4c138b3c5205372","nwkAddr":42048,"manufId":4417,"manufName":"_TZE200_7p8ugv8d","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1694715895027,"defaultSendRequestWhen":"immediate"}

### Comments

I tried to create a external converter, but failed hard.
I also think that this Thermostat is maybe similar to the tuya thermostat TS0601,
so maybe this can be just added to the whitelist?

If you need further informations, just contact me.

Thanks in advance for your great work
Dirk

tms99, 2023-10-03:
Hi, 

first of all I'd like to thank everybody for this great project, which helped me a lot setting up and integrating a bunch of zigbee device into my little smart home. This is the first time that I am trying to integrate an unknown device into my setup.

I have the same devices and I got them at least partially integrated here. Means, I can read and set temperature on the device, as well as a few boolean values.
Details can be found in the external_converter file attached: **tuya_maginon_wt1.txt** (rename to *.js for use in configuration.yaml)

I can't get the conversion of the daily schedules to work. It seems that this device is using a different way of handling the time in the data array. I tried to explain my assumption in the comments in the converter file.
 
Although I am pretty ok with C++, I'm an absolute beginner with JS and all the internals of zigbee2mqtt and zigbee-herdsman-converters , So currently I do not know if my assumption (see attached external_converter file) is correct and if so, how to implement that.

Maybe someone with a better understanding of the internals can have a look at my thoughts and support.

Thx for any support
Tom.
[tuya_maginon_wt1.txt](https://github.com/Koenkk/zigbee2mqtt/files/12796586/tuya_maginon_wt1.txt)

Jezz3, 2023-10-08:
thank you very mich for the config :) Helped me alot to get my thermostat integration into zigbee started.

Koenkk, 2023-10-08:
@tms99 code looks good, could you make a PR?

tms99, 2023-10-08:
@Koenkk : I'd like to do that, but 
 - the code of the external_converter is still incomplete and 
 - for handling the schedules a new function in zigbee-herdsman-converters/src/lib/tuya.ts is required, because  _**thermostatScheduleDaySingleDP**_ does not generate the correct byte sequence expected from this device.
 
I tried to anaylze the required functionality based on the debug output from the device (see my explanation in previously attached external_converter file). 
Unfortunately I am a JavaScript dummy and I was hoping that someone with better knowledge could write this function... ;-)

MarphXL, 2023-10-16:
I added the external converter, but didn't see the WT-1 in the entities or devices. It's listed in Zigbee2MQTT as a device with unknown manufacturer.

@tms99 can you help me, so I could use this WT-1 in homeassistant?

Debug 2023-10-16 16:54:26Received Zigbee message from 'TS0601_thermostat', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,230],"type":"Buffer"},"datatype":2,"dp":103}],"seq":768}' from endpoint 1 with groupID 0

Warning 2023-10-16 16:54:26Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_7p8ugv8d'

Info 2023-10-16 16:54:26MQTT publish: topic 'zigbee2mqtt/TS0601_thermostat', payload '{"device":{"applicationVersion":67,"dateCode":"","friendlyName":"TS0601_thermostat","hardwareVersion":1,"ieeeAddr":"0xa4c138e042043445","manufacturerID":4417,"manufacturerName":"_TZE200_7p8ugv8d","model":"unknown","networkAddress":25309,"powerSource":"Battery","stackVersion":0,"type":"EndDevice","zclVersion":3},"last_seen":"2023-10-16T16:54:21+02:00","linkquality":78}'

groows, 2023-11-04:
Hi everyone,

are there any updates on this topic? 
Can I delivery any further informations of the device?

Thanks in advance

HadsonB, 2023-11-14:
does anyone know,
will the new device be integrated?

Nordmann1961, 2023-11-18:
Hello everyone,

I use the ioBroker as a smarthome base.
The radiator thermostat is currently (as of 18.11.2023) on sale at the discounter Aldi. 
I get the following entries for "Message from Zigbee":

_{"type":"commandDataReport","data":{"seq":768,"dpValues":[{"dp":103,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,200]}}]},"linkquality":255,"groupID":0,"cluster":"manuSpecificTuya","meta":{"zclTransactionSequenceNumber":216,"manufacturerCode":null,"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0}},"endpoint_id":1}_

I would appreciate it if one of you could also take a look at the ioBroker Zigbee adapter to see how the integration can/could work here.
However, I have absolutely no idea about programming.

I am looking forward to your answers!

Sonsi79, 2024-01-27:
@Koenkk can you help to integrate?

Sonsi79, 2024-01-27:
> thank you very much for the config :) Helped me alot to get my thermostat integration into zigbee started.

How did you integrate the code and let it work? Can you help and tell us more?

Koenkk, 2024-01-28:
Yes, but I need a working external converter.

Sonsi79, 2024-01-28:
> Hi,
>
> first of all I'd like to thank everybody for this great project, which helped me a lot setting up and integrating a bunch of zigbee device into my little smart home. This is the first time that [...]

@tms99 Did you get progress in searching for a complete external converter?

tms99, 2024-01-28:
@Sonsi79 As mentioned before: it's running in my environment using the external_converter I provided ([tuya_maginon_wt1.txt](https://github.com/Koenkk/zigbee2mqtt/files/12796586/tuya_maginon_wt1.txt)) with the mentioned limitations. Any support for the daily schedule is appreciated.

Sonsi79, 2024-01-28:
> @Sonsi79 As mentioned before: it's running in my environment using the external_converter I provided ([tuya_maginon_wt1.txt](https://github.com/Koenkk/zigbee2mqtt/files/12796586/tuya_maginon_wt1.txt)) with the mentioned limitations. Any support for the daily schedule is appreciated.

Thnx for quick reply. I try using it without daily schedule. HA can to it a as well too :-) but would be nice to get all features in future.

MarphXL, 2024-02-13:
It's soooo complicated to add a new unsupported device. :-(
The documentation https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html is very unclear. I am fit in programming 7 languages but I can't add a device. No help here???

MarphXL, 2024-02-18:
@Koenkk I bought Sonoff, so I can send you some Maginon WT-1. Do you want some?

closed 2024-09-16 by github-actions[bot]