Koenkk/zigbee2mqtt #18986: [New device support]: _TZE204_dzuqwsyg

gciman, 2023-09-16:
### Link

https://www.zigbee2mqtt.io/devices/BAC-002-ALZB.html

### Database entry

{"id":17,"type":"Router","ieeeAddr":"0xa4c138b6afb62f9f","nwkAddr":52651,"manufId":4417,"manufName":"_TZE204_dzuqwsyg","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"9;�,i<;�,i>;�,f?;�,\u0012","65506":56,"65508":0,"stackVersion":0,"dateCode":"","appVersion":74}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1694852227688,"defaultSendRequestWhen":"immediate"}

### Comments

TS0601
Produttore Zigbee
_TZE204_dzuqwsyg

### External converter

```shell
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const ota = require('zigbee-herdsman-converters/lib/ota');
const {repInterval, defaultBindGroup, OneJanuary2000} = require('zigbee-herdsman-converters/lib/constants');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const fz_moes_thermostat_custom = {
	cluster: 'manuSpecificTuya',
	type: ['commandGetData', 'commandSetDataResponse', 'commandDataResponse'],
	convert: (model, msg, publish, options, meta) => {
            const dpValue = tuya.firstDpValue(msg, meta, 'moes_thermostat');
            const dp = dpValue.dp;
            const value = tuya.getDataValue(dpValue);
		let temperature;
		/* See tuyaThermostat above for message structure comment */
		switch (dp) {
		case tuya.dataPoints.moesSchedule:
			return {
				program: [
					{weekdays_p1: value[0] + 'h:' + value[1] + 'm ' + value[2]/2 + '°C'},
					{weekdays_p2: value[3] + 'h:' + value[4] + 'm ' + value[5]/2 + '°C'},
					{weekdays_p3: value[6] + 'h:' + value[7] + 'm ' + value[8]/2 + '°C'},
					{weekdays_p4: value[9] + 'h:' + value[10] + 'm ' + value[11]/2 + '°C'},
					{saturday_p1: value[12] + 'h:' + value[13] + 'm ' + value[14]/2+ '°C'},
					{saturday_p2: value[15] + 'h:' + value[16] + 'm ' + value[17]/2 + '°C'},
					{saturday_p3: value[18] + 'h:' + value[19] + 'm ' + value[20]/2 + '°C'},
					{saturday_p4: value[21] + 'h:' + value[22] + 'm ' + value[23]/2 + '°C'},
					{sunday_p1: value[24] + 'h:' + value[25] + 'm ' + value[26]/2 + '°C'},
					{sunday_p2: value[27] + 'h:' + value[28] + 'm ' + value[29]/2 + '°C'},
					{sunday_p3: value[30] + 'h:' + value[31] + 'm ' + value[32]/2 + '°C'},
					{sunday_p4: value[33] + 'h:' + value[34] + 'm ' + value[35]/2 + '°C'},
				],
			};
		case 1: // system mode (OFF, ON)
			if(value == 0) {
				return {state: 'OFF'};
			} else if(value == 1) {
				return {state: 'ON'};
			} else {
				return {};
			}
		case 2: // system mode (cooling, heating, fan)
			if(value == 0) {
				return {system_mode: 'cool'};
			} else if(value == 1) {
				return {system_mode: 'heat'};
			} else if(value == 2) {
				return {system_mode: 'fan_only'};
			} else {
				return {};
			}
		case 28: // fan mode
			if(value == 0) {
				return {fan_mode: 'low'};
			} else if(value == 1) {
				return {fan_mode: 'medium'};
			} else if(value == 2) {
				return {fan_mode: 'high'};
			} else if(value == 3) {
				return {fan_mode: 'auto'};
			} else {
				return {};
			}
		case 16: // temperature setpoint for both heating and cooling
			return {occupied_heating_setpoint: value};
		case tuya.dataPoints.moesChildLock:
			return {child_lock: value ? 'LOCK' : 'UNLOCK'};
		case tuya.dataPoints.moesMaxTempLimit:
			return {max_temperature_limit: value};
		case tuya.dataPoints.moesMaxTemp:
			return {max_temperature: value};
		case tuya.dataPoints.moesDeadZoneTemp:
			return {deadzone_temperature: value};
		case tuya.dataPoints.moesLocalTemp:
			temperature = value & 1<<15 ? value - (1<<16) + 1 : value;
			return {local_temperature: parseFloat((temperature / 10).toFixed(1))};
		case tuya.dataPoints.moesTempCalibration:
			temperature = value;
			// for negative values produce complimentary hex (equivalent to negative values)
			if (temperature > 4000) temperature = temperature - 4096;
			return {local_temperature_calibration: temperature};
		case tuya.dataPoints.moesHold: // state is inverted, preset_mode is deprecated
			return {preset_mode: value ? 'program' : 'hold', preset: value ? 'program' : 'hold'};
		case tuya.dataPoints.moesScheduleEnable: // state is inverted, preset_mode is deprecated
			return {preset_mode: value ? 'hold' : 'program', preset: value ? 'hold' : 'program'};
		case tuya.dataPoints.moesValve:
			return {heat: value ? 'OFF' : 'ON'};
		case tuya.dataPoints.moesSensor:
			switch (value) {
			case 0:
				return {sensor: 'IN'};
			case 1:
				return {sensor: 'AL'};
			case 2:
				return {sensor: 'OU'};
			default:
				return {sensor: 'not_supported'};
			}
		default: // DataPoint 17 is unknown
			meta.logger.warn(`zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #${
				dp} with data ${JSON.stringify(msg.data)}`);
		}
	},
};

const tz_state = {
	key: ['state'],
	convertSet: async (entity, key, value, meta) => {
		if(value == "OFF") {
			await tuya.sendDataPointBool(entity, 1, false);
		} else if(value == "ON") {
			await tuya.sendDataPointBool(entity, 1, true);
		}
	},
};

const tz_system_mode = {
	key: ['system_mode'],
	convertSet: async (entity, key, value, meta) => {
		if(value == "cool") {
			await tuya.sendDataPointEnum(entity, 2, 0);
		} else if(value == "heat") {
			await tuya.sendDataPointEnum(entity, 2, 1);
		} else if(value == "fan_only") {
			await tuya.sendDataPointEnum(entity, 2, 2);
		}
	},
};

const tz_temperature_setpoint = {
	key: ['occupied_heating_setpoint'],
	convertSet: async (entity, key, value, meta) => {
		await tuya.sendDataPointValue(entity, 16, value);
	},
};

const tz_fan_mode = {
	key: ['fan_mode'],
	convertSet: async (entity, key, value, meta) => {
		if(value == "low") {
			await tuya.sendDataPointEnum(entity, 28, 0);
		} else if(value == "medium") {
			await tuya.sendDataPointEnum(entity, 28, 1);
		} else if(value == "high") {
			await tuya.sendDataPointEnum(entity, 28, 2);			
		} else if(value == "auto") {
			await tuya.sendDataPointEnum(entity, 28, 3);			
		}
	},
};

module.exports = [
    {
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_dzuqwsyg'}],
        model: 'BAC-002-ALZB',
        vendor: 'HKGK',
        description: 'BAC series thermostat',
        fromZigbee: [fz_moes_thermostat_custom],
		onEvent: tuya.onEventSetLocalTime,
        toZigbee: [tz_state, tz_system_mode, tz_fan_mode, tz_temperature_setpoint, legacy.tz.moes_thermostat_child_lock, legacy.tz.moes_thermostat_current_heating_setpoint, legacy.tz.moes_thermostat_mode,
            legacy.tz.hgkg_thermostat_standby, legacy.tz.moes_thermostat_sensor, legacy.tz.moes_thermostat_calibration,
            legacy.tz.moes_thermostat_deadzone_temperature, legacy.tz.moes_thermostat_max_temperature_limit],
			exposes: [e.child_lock(), e.deadzone_temperature(), e.max_temperature_limit(),
			exposes.climate()
			.withSetpoint('occupied_heating_setpoint', 16, 35, 0.1, ea.STATE_SET)
			.withLocalTemperature(ea.STATE)
			.withLocalTemperatureCalibration(ea.STATE_SET)
			.withSystemMode(['cool', 'heat', 'fan_only'], ea.STATE_SET)
			.withFanMode(['low', 'medium', 'high', 'auto', 'off'], ea.STATE_SET)
			.withRunningState(['idle', 'heat', 'cool'], ea.STATE)
			.withPreset(['hold', 'program']),
			e.temperature_sensor_select(['IN', 'AL', 'OU']),
			
			exposes.composite('programming_mode', 'programming_mode', ea.STATE)
                .withDescription(
                    'Schedule MODE ⏱ - In this mode, the device executes a preset week programming temperature time and temperature.',
                )
                .withFeature(e.week())
                .withFeature(exposes.text('workdays_schedule', ea.STATE_SET))
                .withFeature(exposes.text('holidays_schedule', ea.STATE_SET)),
			
			exposes.switch()
				.withState('state', false, 'OFF/ON state'),
	],
    },
];
```

closed 2023-10-25 by github-actions[bot]

pippocuce, 2023-11-06:
Please add thermostat support for "_TZE204_dzuqwsyg"

rwarasaurus, 2023-12-30:
+1 for "_TZE204_dzuqwsyg" support! :pray: 

https://www.aliexpress.com/item/1005004174932182.html

[image]

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE204_dzuqwsyg",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

saltandvinegarcrisps, 2024-01-12:
Most controls working using this: https://github.com/Koenkk/zigbee2mqtt/issues/17521#issuecomment-1890058986