Koenkk/zigbee2mqtt #18991: Paul Neuhaus Q-Remote enhanced support

Diaoul, 2023-09-16:
### What happened?

The remote commands group that I did not choose.
See https://github.com/Koenkk/zigbee-herdsman-converters/issues/3103 for an illustrations of the different commands possible.

### What did you expect to happen?

1. I can choose the group mapping when configuring the remote
2. Selecting the all groups does not act on ALL lights of the network

### How to reproduce it (minimal and precise)

### 1. Single group selected
- Select group 1 on the remote
- Press the ON button
- Notice that the configured group 1 turns on

### 2. All groups selected
- Select all groups with the button `1-3`
- Press the ON button
- Notice that **all** lights of the network turn on

### Zigbee2MQTT version

1.33.0

### Adapter firmware version

20230507

### Adapter

tube_zb_gw_cc2652p2_poe

### Debug log

Single group
info 2023-09-16 21:45:34MQTT publish: topic 'zigbee2mqtt/Ground Floor Remote', payload '{"action":"on","action_group":1,"linkquality":83}'
info 2023-09-16 21:45:34MQTT publish: topic 'zigbee2mqtt/Ground Floor Remote/action', payload 'on'

All groups
info 2023-09-16 21:43:43MQTT publish: topic 'zigbee2mqtt/Ground Floor Remote', payload '{"action":"on","linkquality":87}'
info 2023-09-16 21:43:43MQTT publish: topic 'zigbee2mqtt/Ground Floor Remote/action', payload 'on'

Diaoul, 2023-09-16:
I am not sure if there is another way to solve this but I think it can be solved if it was possible to implement a per-device configuration based mapping at [addActionGroup](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/lib/utils.ts#L117-L122) such that:
- On remote A
- Given a mapping configuration `{2: 102}`
- When the remote sends the ON command with groupID 2
- Then zigbee2MQTT interprets this as group 102
- Then lights on group 102 turn ON and nothing happens for the lights on group 2

Koenkk, 2023-10-17:
You can probably achieve this via binding: https://www.zigbee2mqtt.io/guide/usage/binding.html

> Then zigbee2MQTT interprets this as group 102
Then lights on group 102 turn ON and nothing happens for the lights on group 2

Note this isn't possible, when a remote sends a message to a group, all devices in it will react, z2m cannot influence this.

Diaoul, 2023-10-17:
I have 3 such remotes 😬 I'd love to be able to make use of different groups on each remote for different purposes.
If I understand correctly, that this is not possible?

Koenkk, 2023-10-17:
You can try binding it to a different group (https://www.zigbee2mqtt.io/guide/usage/binding.html)

Diaoul, 2023-10-17:
Ah OK I'll try that. I have used binding before but not with groups, I'll dig into that 🙏

Diaoul, 2023-11-05:
> when a remote sends a message to a group, all devices in it will react, z2m cannot influence this

* I think I'm in this case and the remote always sends group 1, 2 and 3.
* When pressing the button 1-3 (all groups), the remote does not send the `action_group`

Let's call the group sent by the remote the "button group"

**What I did:**
* I bound the remote to a group with the lights I want to control
* I've banned the use of groups 1, 2 and 3 so they don't conflict
* :+1: This makes the remote behaves well when no group is selected (1-3 button, all groups)
* :-1: However, it makes it such that I can only bind to one group and I cannot use the "button group" buttons of the remote as they "override" the group that the remote has been bound to
  * Weirdly though: this is only the case for white level control and brightness, the on-off buttons seem to work with the bound group regardless of the button group I chose :confused: 

:point_right: Do you know why?

If this is again a limitation linked to Zigbee let's leave it at that. I have something not perfect but usable and I can bind many different remotes to many different lights, I only wished I could re-use the "button group" feature of the remote to control different lights with a single remote.

closed 2024-06-03 by github-actions[bot]