Koenkk/zigbee2mqtt #18992: ZMS-102 Tuya status always on occupied

deef88, 2023-09-16:
### What happened?

I installed 2 ZMS-102 PIR sensors, but noticed that the occupied status is always on status TRUE.

### What did you expect to happen?

After xxx seconds the status will switch to false (in case there isn't any motion.

### How to reproduce it (minimal and precise)

Tried a few times and days always the same issue.

### Zigbee2MQTT version

1.33.0

### Adapter firmware version

20211217

### Adapter

zStack3x0

Anashost, 2023-09-25:
Kinda same issue with this sensor, seems unstable sometimes it won't clear and sometimes it stuck at occupied, and sometimes there's long delay.

- Model
ZMS-102
- Vendor
TuYa
- Device type
EndDevice
- Zigbee Model
TS0202
- Zigbee Manufacturer
_TZ3000_msl6wxk9
- Description
Motion sensor
- Support status
Supported

This sensor looks the same as other tuya pir sensors i have, both working perfectly:
- TuYa IH012-RT01
- TuYa 809WZT

MoLiNaSa, 2023-10-23:
same issue here
PIR sensor model ZMS-102 or manufacturer _TZ3000_msl6wxk9 was working well with some version < 1.28 under OpenHab (not sure which version)
upgrade tool for zigbee2mqtt in OpenHab failed then PIR sensor didn't work anymore, after pairing occupancy remained TRUE
so I installed manual extra Zigbee2MQTT server which has latest 1.33.1, behaving did not change
device can be paired successfully, but then occupancy remains TRUE
removing the device without force it starts to blink, connection seems to work, and re-pairs 
even following the manual for troubleshooting issue with similar model did not help:
https://www.zigbee2mqtt.io/devices/IH012-RT01.html

Anashost, 2023-10-23:
@MoLiNaSa I've got mine to work better with changing the placement of my coordinator (sonoff model - P), i put it closer, higher and i had a usb 3.0 cable close to it, so it may caused some interference. It seems like this sensors has very bad signal which may cause this issue, 100% speculation here tho.

>https://www.zigbee2mqtt.io/devices/IH012-RT01.html

The other model you mentioned here is one of a sensors i have (looks the same with blue lighting), which i consider to be one of the best small PIR tuya has, it works very very well.

jonathh21, 2024-02-06:
I've got a few of these and seem to have issues with them from time to time.

I currently have 2 of them in boxes and one report's ghosts the other doesn't.

I noticed that the battery was low (both in terms of voltage and percentage) on ghoater so I have switched it out for a Duracell one, we'll see if it helps.

I am particularly aware of this following well documented voltage issues with IKEA VALLHORN pirs.

jonathh21, 2024-02-11:
Well it helped or rather the device was more reliable until today.

Today it has been stuck on 'occupied' for hours on 2 separate occasions.

Also, after I put in a brand new Duracell battery, the voltage and battery percentage never updated.

I am phasing in aqara sensors which seem more reliable..

Shame, I've got loads of these and the dome pirs - but what use is an unreliable PIR.

xelemorf, 2024-03-25:
> has been stuck on 'occupied' for hours on 2 separate occasions.
> 
> Also, after I put in a brand new Duracell battery, the voltage and battery percentage never updated.
> 
> I am phasing in aqara sensors wh

You might regret Aqara/Xiaomi (they're all Lumi actually) sensors on the long run (as do I), those devices are the most stubborn I have ever found straight up refusing to pair with anything except aqara routers. Even refusing to pair with their own smart plugs, accepting just smart wall switches in my case, won't pair to sonoff dongle-e routers or sonoff dongle-p coordinator, not to mention also refusing to pair any TuYa device (or in some case the'll pair if battery was removed but will drop off very quickly). And for the last my favorite, when pairing they tend to dissapear from Z2M inventory never joining back so you'll need to keep a tight record of your stuff, as they vanish. 

However if you go that route, pairing can take multiple tries including battery removal and reinsert.

dCypherNx, 2024-04-21:
Anyone know if it's possible to clear status sending a command to mqtt topic? In my case I know that if the sensor say it's occupied for more than 10 minutes it's a false positive, and I may clear it manually

xelemorf, 2024-04-21:
Recently just stumbled upon a page for a very similar device, the described method helped me to fix occupancy stuck issue, added a note ito it aswell:

```
Troubleshooting: Occupancy stays on (or device "last seen" does not update): 

If the occupancy state stays on despite a lack of movement and regularly sent messages, it needs to be repaired. This can happen, e.g., when the power is interrupted by a battery change. 

1.    Remove from Zigbee2MQTT (without force). Device should start blinking upon successful removal. 
2.    Remove battery, wait a few seconds, reinsert. 
3.    Execute normal pairing procedure. 

Repairing without removal or power cut does not fix this issue. 
XelNote: If the above fails, pair it with a TuYa plug (like: corridor-ac-plug).

https://www.zigbee2mqtt.io/devices/IH012-RT01
```

closed 2024-11-18 by github-actions[bot]

jonathh21, 2025-09-17:
I've just found this thread.

Have a similar ish problem.

This didn't resolve it, nothing so far has.

Device was connected and operating for a year/18 months.

Suddenly it froze, repaired several times in several locations. No dice.

I have had problems with this model before, but I have 2 others right now that are operating fine.

Id write this unit off and put it in the bin were it not for the fact it pairs and works with Tuya.

So I can still use if I go via Tuya but it is far from ideal