Koenkk/zigbee2mqtt #18993: Tunable white fails: Zigbee 3.0 DALI CONV LI dimmer for DALI-based luminaires (only one device)

thorstenhornung1, 2023-09-16:
### What happened?

According to Osram the decide Supportseite tunable white. However when hooking it up to a tunable white led the light turns to a very cold color temperature. Zigbee to Mqtt does not offer color Temperatur. 

using the developer console I get this  

2023-09-16 22:06:16Read result of 'lightingColorCtrl': {"startUpColorTemperature":6535,"coupleColorTempToLevelMin":160,"colorTempPhysicalMax":65279,"colorTempPhysicalMin":0,"colorCapabilities":16,"numPrimaries":255,"colorTemperature":250}

Color temperature seems to be read only 

### What did you expect to happen?

Being able to adjust color temperature 

### How to reproduce it (minimal and precise)

Use device with tunable white led

### Zigbee2MQTT version

1.33.0

### Adapter firmware version

20220302

### Adapter

zStack3x0

thorstenhornung1, 2023-10-17:
I had to remove the box from the Dali system as the white level and brightness were no longer controllable.

los7702, 2023-11-02:
I also have the same issue
I need some guidance to make necessary updates on the device instructions. but are willing to take the time necessary.
Where do I start?

los7702, 2023-11-06:
I have made a pullrequest fixing this issue.
https://github.com/Koenkk/zigbee-herdsman-converters/pull/6410

los7702, 2023-11-06:
@thorstenhornung1  I guess you could use the Converter  after updating to a newer version than today the issue is fixed in the reposetory for devices, remember to close rthe isue afterwards

closed 2024-06-04 by github-actions[bot]