Koenkk/zigbee2mqtt #18996: [New device support]: New IKEA TRÅDFRI lightbulb LED2103GS

yaskad, 2023-09-17:
### Link

https://www.ikea.com/se/sv/p/tradfri-led-ljuskaella-e27-806-lumen-smart-tradloes-dimbar-varmvit-klot-60541496/

### Database entry

iam not sure where to get this?

### Comments

Hello

Sorry for the missing some information, this is my first time doing this. I bought new IKEA lightbulbs and to my surprises there were not supported yet, I think these are the new generation or something. Regardless, I have read the guide and created my external converter that works (at least I can control them now, with on/off and brightness level :) )

I would like to help others who might face the same issue, also, I appreciate the feedback if there is anything wrong with the external converter.

PS, I am using the Home Assistant Add-on setup!
Thanks in advance!

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TRADFRI bulb E27 WW globe 806lm'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'LED2103GS', // Vendor model number, look on the device for a model number
    vendor: 'IKEA', // Vendor of the device (only used for documentation and startup logging)
    description: 'TRADFRI LED bulb E27 806 lumen, wireless dimmable warm white', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.on_off, fz.brightness, fz.level_config, fz.ignore_basic_report], // We will add this later
    toZigbee: [tz.light_onoff_brightness, tz.ignore_rate, tz.level_config], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.light_brightness()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
};

module.exports = definition;
```

### Supported color modes

None, only warm white with brightness

Koenkk, 2023-09-17:
added, thanks!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-09-17 by Koenkk

muchul, 2023-10-06:
Hello,

the link https://www.ikea.com/se/sv/p/tradfri-led-ljuskaella-e27-806-lumen-smart-tradloes-dimbar-varmvit-klot-60541496/ above refers to a LED with the designation LED2103G (**5**) number five  and not to LED2103G (**S**) letter S.

I have this bulb and my iobroker tells me that it is a LED2103G5
If this is a typo with the S, it would be nice if this could be corrected.