Koenkk/zigbee2mqtt #19000: Tuya smart curtain/new manufacture name

LirikToscanini, 2023-09-17:
### Link

https://www.zigbee2mqtt.io/devices/TS0601_cover_1.html

### Database entry

{"id":36,"type":"Router","ieeeAddr":"0xa4c138c66659b971","nwkAddr":53736,"manufId":4417,"manufName":"_TZE204_1fuxihti","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"���,g���,g���,f���,\u0012���,g���,g֐�,g","65506":56,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE204_1fuxihti","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1694939938012,"defaultSendRequestWhen":"immediate"}

### Comments

I tried to connect another Tuya Curtain motor via z2m, but z2m identified it with a new manufacturer name '_TZE204_1fuxihti' , which is not in the list of official Tuya converters. I tried to make an external converter, but failed. device is not supported. please help. the previous 2 Tuya Curtain motors were identified without problems.

### External converter

```shell
const fz = require( 'zigbee-herdsman-converters/converters/fromZigbee' );
const tz = require( 'zigbee-herdsman-converters/converters/toZigbee' );
const exposes = require( 'zigbee-herdsman-converters/lib/exposes' );
const reporting = require( 'zigbee-herdsman-converters/lib/reporting' );
const extend = require( 'zigbee-herdsman-converters/lib/extend' );
const tuya = require( 'zigbee-herdsman-converters/lib/tuya' );
const e = exposes.presets;
const ea = exposes.access;

const definition = {
        zigbeeModel: [
            'owvfni3\u0000', 'owvfni3', 'u1rkty3', 'aabybja', // Curtain motors
            'mcdj3aq', 'mcdj3aq\u0000', // Tubular motors
            ],
        fingerprint: [
            { modelID: 'TS0601', manufacturerName: '_TZE204_1fuxihti' },
            ],
        model: 'TS0601',
        vendor: 'TuYa',
        description: 'Curtain motor/roller blind motor/window pusher/tubular motor',
        whiteLabel: [
            {vendor: 'Tuya', model: 'M515EGZT'},
            ],
        fromZigbee: [
            fz.tuya_cover,
            fz.ignore_basic_report,
            ],
        toZigbee: [
            tz.tuya_cover_control,
            tz.tuya_cover_options,
            ],
        exposes: [
            e.cover_position().setAccess('position', ea.STATE_SET),
            ],
    };

module.exports = definition;
```

Koenkk, 2023-09-17:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/72079b9919518c071406c9a5921de809

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

wildchai, 2023-10-19:
> Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/72079b9919518c071406c9a5921de809
>
> * save this as file next to `configuration.yaml` as [...]

 Thanks for this. I had the same issue. With this it was resolved.

LirikToscanini, 2023-10-20:
Thanks a lot,
 the issue is fixed with the external converter.

Koenkk, 2023-10-20:
Integrated the fix, you can remove the ext converter after updating to the next release.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-10-20 by Koenkk

gbrest, 2025-06-30:
Hey everyone! 👋

Just wanted to share a bit of my experience with the **MOES Zigbee Curtain Motor** (`_TZE204_1fuxihti`, model `TS0601`) and how I finally got it working in **Home Assistant with ZHA**, after a lot of digging and frustration.

---

📆 **As of mid-2025, the issue is still ongoing**

I bought a couple of these curtain motors in 2025 and paired them to Home Assistant using ZHA. To get them into pairing mode, I:

* Pressed the reset button **three times** quickly
* Then held it down until the LED started flashing

The device paired and showed up in the ZHA device list, but — **no entities appeared.**

---

❗**Why? Because it’s not really Zigbee-native**

This is another case of a Tuya TS0601 device. It communicates using Tuya’s proprietary MCU protocol over the `0xEF00` cluster, which is **not natively understood by ZHA**.

That’s why the device shows up, but doesn’t expose any entities or usable functionality unless you provide a **custom quirk**.

I found an old discussion where user **Prozpect** had posted a working quirk, but the link is gone. I had to do a deep dive across GitHub, Reddit and the Wayback Machine to find pieces of it and put it all back together.

---

🧩 **So here’s the full working solution, in one place, for rookies like me**

This is everything I did to get the curtain working properly with ZHA, including open/close/stop and position slider.

---

 ✅ **Step 1: Create the quirks folder**

Inside your `/config` directory (where `configuration.yaml` lives), create a new folder called:

**quirks**

So the full path is:

/config/quirks/

---

 ✅ **Step 2: Create an empty `__init__.py`**

Inside the `quirks` folder, create a new file called:

__init__.py

It can be completely empty, but it **must exist**.

---

 ✅ **Step 3: Create the custom quirk file**

Still inside the `quirks` folder, create a file called:

moes_curtain.py

And paste the following code into it (this is the full working quirk):

```

"""Tuya MCU based cover and blinds."""

from typing import Dict, Optional, Union
from zigpy.profiles import zha
import zigpy.types as t
[... 205 lines cut ...]
✅ **Step 4: Tell ZHA to load custom quirks**

In your `configuration.yaml`, add or update the `zha:` section like this:

```
zha:
  custom_quirks_path: quirks
```

> **Do not** include `/config/quirks` — just `quirks`

---

 ✅ **Step 5: Restart and pair again**

1. Restart Home Assistant completely (not just ZHA).
2. Remove any previous pairing of the curtain motor.
3. Put the curtain back into pairing mode (3 presses, then long press).
4. Pair it again from **ZHA > Add Device**.

You should now see a new **`cover.xxxxxx` entity**, and you’ll be able to open/close/stop and use the position slider.

---

Let me know if this works for you or if you get stuck — happy to help!