Koenkk/zigbee2mqtt #19002: When pairing a SONOFF ZBMINIL2 an error occurs

FWeinb, 2023-09-17:
### What happened?

I have multiple ZBMINIL2 in my network all paired before `1.33` and they are working fine. Today I wanted to add another and got the following error, the device is working for a few minutes and then will appear offline:

```
Failed to configure '0x9035eafffe02b0a2', attempt 2 (Error: Unbind 0x9035eafffe02b0a2/1 genPollCtrl from '0x00212effff095933/1' failed (Error: status: 136) at DeconzAdapter.unbind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:818:19) at Endpoint.unbind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:673:13) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/sonoff.ts:63:13) at Configure.configure (/app/lib/extension/configure.ts:117:13))
```

### What did you expect to happen?

There should not be an error during interviewing and the device should stay online. 

### How to reproduce it (minimal and precise)

1. Pair ZBMINIL2
2. See error above

### Zigbee2MQTT version

1.33.0

### Adapter firmware version

0x26780700

### Adapter

ConBee2/RaspBee2

francisp2, 2023-09-18:
Did you try 'reconfigure' ?

julianfs, 2023-09-19:
The same is true for the previous version of this hardware, the ZBMINI-L.

As far as I can tell it doesn't cause a problem, you just get a lot of 'junk' in your log file.

closed 2023-10-27 by github-actions[bot]

Kitof, 2024-03-18:
Is there a way to suppress this error ? I have it on each of my switch on every action, it's very annoying...

viniciusgava, 2024-05-10:
Has anybody found a solution for this?