Koenkk/zigbee2mqtt #19005: Problem to change CC2531 to SonOff ZB Dongle 3.0 USB CC2652P + CP2102N

diazovitch, 2023-09-17:
### What happened?

I do this with FLASH-PROGRAMMER-2

_Plug in your adapter and put it in BSL mode, consult your adapters manual on how to put it in BSL mode
Select your adapter, go to "MAC address"
Fill the old coordinator ieee address into "Secondary Address" -> "IEEE 802.15.4 MAC address" (first 0x can be skipped)
Press "Write"
Reflash the firmware on your stick - in the right corner under Secondary MAC check "Retain secondary IEEE" (this is important, otherwise the coordinator will not use the new ieee address)_

But when i plug it on my RPI 3B+ with domoticz, zigbee devices doesn't work.
I restart domoticz and i restart rpi too.
I don't know what I should do ? Shall I wait ?
Could you help me please ?

### Zigbee2MQTT version

1.33.0

### Adapter firmware version

20221226

### Adapter

SonOff ZB Dongle 3.0 USB CC2652P + CP2102N

francisp2, 2023-09-18:
You can check the logs in the Zigbee2MQTT front-end

diazovitch, 2023-09-20:
Hi,
Now it works but some devices are very slow to respond.
Do you think I should buy a powered usb hub for this new adapter ?
And could you tell me how do you see the log in front-end with domoticz ?

Here 
[image]

francisp2, 2023-09-21:
The Zigbee2MQTT front-end is a separate web-server, normally running on port 8080 on the IP Zigbee2MQTT is configured
[image]

diazovitch, 2023-09-21:
What is the default address to access ?

francisp2, 2023-09-21:
I don't use Domoticz myself, but according to the Domoticz documentation
https://github.com/Koenkk/zigbee2mqtt/issues/19005
```
Add Zigbee2MQTT frontend to Domoticz Custom menu

Optional you can add the [frontend User interface of Zigbee2MQTT](https://www.zigbee2mqtt.io/guide/configuration/frontend.html) as a custom menu in Domoticz.

Make a file Zigbee2MQTT.html in domoticz/www/templates

Contents

<IFRAME SRC="http://ipaddress:8090/" height="800" width="100%"></IFRAME> 

Rename IPAddress to the real IP address of your domoticz server. Also change the port of the frontend ifchanged in setup.

Restart Domoticz and you will have another Zigbee2MQTT entry in custom menu.

See also wiki page [Custom Menu](https://www.domoticz.com/wiki/Custom_menu)
```

closed 2023-10-30 by github-actions[bot]