Koenkk/zigbee2mqtt #19007: AwoX - set off does not switch off the LED but sets the brightness to 1

87insane, 2023-09-18:
### What happened?

See comments here: https://github.com/Koenkk/zigbee2mqtt/discussions/18366#discussioncomment-6595176

### Zigbee2MQTT version

1.33.0 commit: f224ef7

### Adapter firmware version

0x26580700

### Adapter

ConBee2/RaspBee2

julianfs, 2023-09-19:
You may need to try an external converter to include the line 'meta: {turnsOffAtBrightness1: true},', as used in this definition for an INNR bulb:

    {
        zigbeeModel: ['RCL 240 T'],
        model: 'RCL 240 T',
        vendor: 'Innr',
        description: 'Smart round ceiling lamp comfort',
        extend: extend.light_onoff_brightness_colortemp({colorTempRange: [200, 454]}),
        meta: {turnsOffAtBrightness1: true},
        ota: ota.zigbeeOTA,
    },

87insane, 2023-09-20:
can somebody here, try that? I think thats the way

Hypfer, 2023-09-27:
> can somebody here, try that? I think thats the way

What about you as the author of the issue?

87insane, 2023-09-29:
iam on holliday without my hardware...

closed 2023-11-06 by github-actions[bot]

87insane, 2023-11-10:
Hey can somebody test this? I cant...

Frans-Willem, 2023-11-15:
Did some research on this, it appears Awox did not properly implement the "Move to Level with On/Off" command, as in it doesn't properly turn off when asked to go to level 0.
Sending the "Off" command of the "On/Off" cluster does work, and could be a workaround, although I'm not sure how to implement this properly.

Can I un-stale this issue ?

87insane, 2023-11-17:
Hey - I think this can resolve our problem:
https://github.com/Koenkk/zigbee2mqtt/issues/19007#issuecomment-1725871948

But i cant test actually.

kaigame, 2023-11-25:
I'll happily test this, but i'm no external converter expert, the config will need to be modified to support the Awox light, and there should be more lines of code than this no?

Please forgive my ignorance :)

kaigame, 2023-11-27:
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['EGLO_ZM_RGB_TW'],
    model: '900024/12253',
    vendor: 'EGLO',
    description: 'SALITERAS-Z ceiling light',
    extend: extend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 370], supportsHueAndSaturation: true}),
    meta: {turnsOffAtBrightness1: true},
};

module.exports = definition;

This seems to work for me, description is a bit off so might change that.

Can confirm that lights turn off with this added and the lights re-paired.

Thanks!!

MarcR2000, 2023-12-20:
I have a similar issue... Will there be a fix or do I need to workaround?

What happened?
When you turn the brightness of AwoX22955 to zero, it does not switch off but stays on lowest brightness level. Device only switches off when on/off toggle is pressed.

This is would it looks like after tuning to 0 (no on/off toggle pressed)

topic 'zigbee2mqtt/Eglo_Panel_1', payload '{"brightness":0,"color_mode":"color_temp","color_temp":256.68,"color_temp_startup":65535,"device":{"applicationVersion":2,"friendlyName":"Eglo_Panel_1","hardwareVersion":60,"ieeeAddr":"0xa4c13823a657eeb0","manufacturerID":4417,"manufacturerName":"AwoX","model":"33955","networkAddress":33719,"powerSource":"Mains (single phase)","softwareBuildID":"2.3.12_250","stackVersion":1,"type":"Router","zclVersion":3},"linkquality":94,"power_on_behavior":"previous","state":"OFF"}'

So it does "think" it's off, but actually it isn't.

After toggling ON/OFF it looks like this...

topic 'zigbee2mqtt/Eglo_Panel_1', payload '{"brightness":254,"color_mode":"color_temp","color_temp":256.68,"color_temp_startup":65535,"device":{"applicationVersion":2,"friendlyName":"Eglo_Panel_1","hardwareVersion":60,"ieeeAddr":"0xa4c13823a657eeb0","manufacturerID":4417,"manufacturerName":"AwoX","model":"33955","networkAddress":33719,"powerSource":"Mains (single phase)","softwareBuildID":"2.3.12_250","stackVersion":1,"type":"Router","zclVersion":3},"linkquality":91,"power_on_behavior":"previous","state":"OFF"}'

... and is actually switched off

What did you expect to happen?
Switch off on brightness 0

How to reproduce it (minimal and precise)
Happens always when tuning to 0 brightness instead of ON/OFF toggle

Zigbee2MQTT version
1.34.0

Adapter firmware version
20220507

Adapter
USB Stick CC2652P Sonoff

mbbert, 2024-01-28:
I got the same issue after upgrading from firmware version 2.2.10 to 2.7.3 (HW version 4.60). Before the upgrade, turning off the lamp was possible. I did the update due to the [warning](https://www.zigbee2mqtt.io/devices/33957.html#warning-degrades-network-performance) about degraded network performance, and I noticed that the device was reporting it's status every couple of seconds.

I tried to add `meta: {turnsOffAtBrightness1: true},`  to a custom definition after creating it as described [here](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-creating-the-external-definition).
But this didn't work for me, or I did something wrong. Here is my file:
```
const {light} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TLSR82xx'],
    model: 'TLSR82xx',
    vendor: 'AwoX',
    description: 'Automatically generated definition',
    extend: [light({"colorTemp":{"range":[153,370]}})],
    meta: {turnsOffAtBrightness1: true},
};

module.exports = definition;
```

But if you add the device to a group, and send `{"state": "OFF"}` to the group, the device actually turns off. This might be a work-around for this issue.

35gh, 2025-04-08:
I manage to work/around my on/off issue with my EGLO-ZM-RGB-TW, version 2.7.6, hardware 4.60 by following Diaoul's comment in linked issue:

> :tada: **I've managed to fix this issue (it seems) by emptying the `Transition` in the `Setting (specific)` tab.**
> 
> It was previously at 1 as I wanted transitions on every change. I even tried to set it back to 0 which is supposed to be the default but it did not work either. I imagine `0` and `empty` have different meanings :shrug: 
> 
> Edit: I think it's still actually a bit broken when trying to use transitions.
> 
> _Originally posted by @Diaoul in https://github.com/Koenkk/zigbee2mqtt/discussions/18366#discussioncomment-6960681_

I am now able to switch it on and off using toggle switch in HA.

With non-empty transitions I was also getting errors such as if that helps to identify and fix the issue:
`error 2025-04-02 13:52:23z2m: Publish 'set' 'brightness' to 'Lamp' failed: 'Error: ZCL command 0xIEEEADDRESS/1 genLevelCtrl.moveToLevelWithOnOff({"level":199,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":64473,"apsFrame":{"profileId":260,"clusterId":8,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":224},"zclSequence":217,"commandIdentifier":11} timed out after 10000ms)'`

Ruud-cb, 2025-07-15:
Just adding my 2 cents here, I have the same issue with a set of spotlights from Eglo Crosslink Z (Awox manufacturer). Bedsides that I also have a lot of irresponsive issues when changing brightness or turning the light on and off quickly. Full topic: https://community.home-assistant.io/t/eglo-crosslink-z-spots-irresponsible-lag-after-some-commands/911274

Is there an update available or expected from Zigbee2Mqtt to address this or am I better off returning the lights..