Koenkk/zigbee2mqtt #19010: external converter breaks down z2m

schluese, 2023-09-18:
### What happened?

external converter breaks down z2m ,z2m works well after removing it ,but I need it to use my device
this converter is not compatible with newest z2m version ,but functions with older version

this is the external converter file:

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const tuyaLocal = {
  dataPoints: {
  sh4Mode: 2,
  sh4HeatingSetpoint: 16,
  sh4LocalTemp: 24,
  sh4ChildLock: 30,
  sh4Battery: 34,
  sh4FaultCode: 45,
  sh4ComfortTemp: 101,
  sh4EcoTemp: 102,
  sh4VacationPeriod: 103,
  sh4TempCalibration: 104,
  sh4ScheduleTempOverride: 105,
  sh4RapidHeating: 106,
  sh4WindowStatus: 107,
  sh4Hibernate: 108,
  sh4ScheduleMon: 109,
  sh4ScheduleTue: 110,
  sh4ScheduleWed: 111,
  sh4ScheduleThu: 112,
  sh4ScheduleFri: 113,
  sh4ScheduleSat: 114,
  sh4ScheduleSun: 115,
  sh4OpenWindowTemp: 116,
  sh4OpenWindowTime: 117,
  sh4RapidHeatCntdownTimer: 118,
  sh4TempControl: 119,
  sh4RequestUpdate: 120,
  },
};
const fzLocal = {
  sh4_thermostat: {
    cluster: 'manuSpecificTuya',
    type: ['commandDataResponse', 'commandDataReport'],
    convert: (model, msg, publish, options, meta) => {
      //const dp = msg.data.dp;
      //const value = tuya.getDataValue(msg.data.datatype, msg.data.data);
      const dpValue = tuya.firstDpValue(msg, meta, 'sh4_thermostat');
      const dp = dpValue.dp;
      const value = tuya.getDataValue(dpValue);
      function weeklySchedule(day, value) {
        // byte 0 - Day of Week (0~7 = Mon ~ Sun) <- redundant?
        // byte 1 - 1st period Temperature (1~59 = 0.5~29.5°C (0.5 step))
        // byte 2 - 1st period end time (1~96 = 0:15 ~ 24:00 (15 min increment, i.e. 2 = 0:30, 3 = 0:45, ...))
        // byte 3 - 2nd period Temperature
        // byte 4 - 2nd period end time
        // ...
        // byte 16 - 8th period end time
        // byte 17 - 9th period Temperature
        const weekDays=['mon', 'tue', 'wed', 'thu', 'fri', 'sat', 'sun'];
        // we get supplied in value only a weekday schedule, so we must add it to
        // the weekly schedule from meta.state, if it exists
        const weeklySchedule= meta.state.hasOwnProperty('weekly_schedule') ? meta.state.weekly_schedule : {};
        meta.logger.info(JSON.stringify({'received day': day, 'received values': value}));
[... 291 lines cut ...]
    tzLocal.sh4_thermostat_get_data,
    tz.tuya_data_point_test,
  ],
  onEvent: tuya.onEventSetLocalTime,
  exposes: [
    e.battery(), e.battery_low(), e.child_lock(),
    exposes.climate().withSetpoint('current_heating_setpoint', 0.5, 29.5, 0.5)
                     .withLocalTemperature()
                     .withSystemMode(['auto','heat','off'])
  ],
};

module.exports = device;

### Zigbee2MQTT version

 1.33.0

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0

schluese, 2023-09-22:
Information database.db:

{"id":32,"type":"EndDevice","ieeeAddr":"0x84ba20fffe2dd7c5","nwkAddr":24409,"manufId":4098,"manufName":"_TZE200_i48qyn9s","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,10,4,5,61184],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65503":"�l,\u0011","65506":27,"modelId":"TS0601","manufacturerName":"_TZE200_i48qyn9s","powerSource":3,"zclVersion":3,"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1695359977827,"defaultSendRequestWhen":"immediate"}

What did you expect to happen?

help with modifying the external converter file or add device support in z2m directly

lux73, 2023-09-22:
you had to post/push your external Converter after you got it working with Versions before

I'm sure @Koenkk has it implemented soon. And not you haven't no Problem after upgrade

schluese, 2023-09-25:
yes, I should have done that :-(

Koenkk, 2023-09-25:
Could you refactor the ext converter to the new format? I think you can use many things from existing TuYa thermostats, examples:

- https://github.com/Koenkk/zigbee-herdsman-converters/blob/a60d4f6e3b709160a045e836200910a1ab8660dc/src/devices/tuya.ts#L2616
- https://github.com/Koenkk/zigbee-herdsman-converters/blob/a60d4f6e3b709160a045e836200910a1ab8660dc/src/devices/tuya.ts#L2664

schluese, 2023-09-27:
Hallo, Ich habe den Konverter jetzt so weit, dass z2mqtt nicht mehr abstürzt :-) aber obwohl ich die Datenpunkte vom alten Konverter genommen habe, scheint es nicht zu funktionieren :-( Nur die lokale Temperatur ist korrekt.  Hier sind die Informationen aus dem Protokoll, leider verstehe ich sie nicht:

Debug 2023-09-27 12:58:00Received Zigbee message from 'Regina Fenster Links', type 'commandMcuSyncTime', cluster 'manuSpecificTuya', data '{"payloadSize":65180}' from endpoint 1 with groupID 0
Debug 2023-09-27 12:58:01Received MQTT message on 'zigbee2mqtt/Regina Fenster Links/set' with data '{"system_mode":"heat"}'
Debug 2023-09-27 12:58:01Publishing 'set' 'system_mode' to 'Regina Fenster Links'
Info 2023-09-27 12:58:11MQTT publish: topic 'zigbee2mqtt/Regina Fenster Links', payload '{"away_data":{"away_hours":0,"day":1,"hour":0,"minute":0,"month":1,"temperature":"17.0","year":2019},"away_mode":"OFF","battery":"74.8","battery_low":false,"boost_timeset_countdown":null,"child_lock":"UNLOCKED","comfort_temp_preset":"21.0","comfort_temperature":null,"current_heating_setpoint":8.5,"eco_temp_preset":"17.0","eco_temperature":null,"error_status":null,"frost_protection":null,"heating_stop":null,"holiday_start_stop":null,"holiday_temperature":null,"linkquality":65,"local_temperature":"20.0","local_temperature_calibration":"0.0","manual_heating_setpoint":"23.5","online":null,"open_window":"OFF","open_window_temperature":null,"schedule_friday":null,"schedule_heating_setpoint_override":"23.5","schedule_monday":null,"schedule_saturday":null,"schedule_sunday":null,"schedule_thursday":null,"schedule_tuesday":null,"schedule_wednesday":null,"system_mode":"heat","weekly-schedule":{"fri":[{"hour":6,"minute":0,"temperature":17},{"hour":9,"minute":0,"temperature":21},{"hour":17,"minute":0,"temperature":17},{"hour":23,"minute":0,"temperature":21},{"hour":24,"minute":0,"temperature":17},{"hour":24,"minute":0,"temperature":21},{"hour":24,"minute":0,"temperature":17},{"hour":24,"minute":0,"temperature":21},{"hour":null,"minute":null,"temperature":17}],"mon":[{"hour":6,"minute":0,"temperature":17},{"hour":9,"minute":0,"temperature":21},{"hour":17,"minute":0,"temperature":17},{"hour":23,"minute":0,"temperature":21},{"hour":24,"minute":0,"temperature":17},{"hour":24,"minute":0,"temperature":21},{"hour":24,"minute":0,"temperature":17},{"hour":24,"minute":0,"temperature":21},{"hour":null,"minute":null,"temperature":17}],"sat":[{"hour":6,"minute":0,"temperature":17},{"hour":9,"minute":0,"temperature":21},{"hour":17,"minute":0,"temperature":17},{"hour":23,"minute":0,"temperature":21},{"hour":24,"minute":0,"temperature":17},{"hour":24,"minute":0,"temperature":21},{"hour":24,"minute":0,"temperature":17},{"hour":24,"minute":0,"temperature":21},{"hour":null,"minute":null,"temperature":17}],"sun":[{"hour":6,"minute":0,"temperature":17},{"hour":9,"minute":0,"temperature":21},{"hour":17,"minute":0,"temperature":17},{"hour":23,"minute":0,"temperature":21},{"hour":24,"minute":0,"temperature":17},{"hour":24,"minute":0,"temperature":21},{"hour":24,"minute":0,"temperature":17},{"hour":24,"minute":0,"temperature":21},{"hour":null,"minute":null,"temperature":17}],"thu":[{"hour":6,"minute":0,"temperature":17},{"hour":9,"minute":0,"temperature":21},{"hour":17,"minute":0,"temperature":17},{"hour":23,"minute":0,"temperature":21},{"hour":24,"minute":0,"temperature":17},{"hour":24,"minute":0,"temperature":21},{"hour":24,"minute":0,"temperature":17},{"hour":24,"minute":0,"temperature":21},{"hour":null,"minute":null,"temperature":17}],"tue":[{"hour":6,"minute":0,"temperature":17},{"hour":9,"minute":0,"temperature":21},{"hour":17,"minute":0,"temperature":17},{"hour":23,"minute":0,"temperature":21},{"hour":24,"minute":0,"temperature":17},{"hour":24,"minute":0,"temperature":21},{"hour":24,"minute":0,"temperature":17},{"hour":24,"minute":0,"temperature":21},{"hour":null,"minute":null,"temperature":17}],"wed":[{"hour":6,"minute":0,"temperature":17},{"hour":9,"minute":0,"temperature":21},{"hour":17,"minute":0,"temperature":17},{"hour":23,"minute":0,"temperature":21},{"hour":24,"minute":0,"temperature":17},{"hour":24,"minute":0,"temperature":21},{"hour":24,"minute":0,"temperature":17},{"hour":24,"minute":0,"temperature":21},{"hour":null,"minute":null,"temperature":17}]},"working_day":null}'

here is my converter, I tried both of your tips:

const legacy = require('zigbee-herdsman-converters/lib/legacy');
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = {...require('zigbee-herdsman-converters/converters/toZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').toZigbee};
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
[... cut ...]
                [111, 'schedule_wednesday', tuya.valueConverter.thermostatScheduleDaySingleDP],
                [112, 'schedule_thursday', tuya.valueConverter.thermostatScheduleDaySingleDP],
                [113, 'schedule_friday', tuya.valueConverter.thermostatScheduleDaySingleDP],
                [114, 'schedule_saturday', tuya.valueConverter.thermostatScheduleDaySingleDP],
                [115, 'schedule_sunday', tuya.valueConverter.thermostatScheduleDaySingleDP],
            ],
    },
};

module.exports = definition;

closed 2023-11-04 by github-actions[bot]