Koenkk/zigbee2mqtt #19011: [New device support]: MOES Carbon Monoxide Alarm

luchtgitaar, 2023-09-18:
### Link

https://aliexpress.com/item/1005005992369494.html

### Database entry

{"id":20,"type":"EndDevice","ieeeAddr":"0xa4c138aab8ad70a2","nwkAddr":24798,"manufId":4417,"manufName":"_TZE200_rjxqso4a","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"UV�,fVV�,\u0012�V�,e�V�,\u0019�V�,\u0007\u0000\u0000\u0000\u0000\u0011","65506":56,"65508":1,"stackVersion":0,"dateCode":"","appVersion":72}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1695055512365,"defaultSendRequestWhen":"immediate"}

### Comments

I tried to follow the supporting new device page but got stuck at adding Tuya datapoints.

follow-suit, 2023-09-21:
I played around with a custom converter for this device, but I'm unsure since I am a noob...Still, maybe my findings will help someone.

Pretty confident with these datapoints:
`meta: {
	tuyaDatapoints: [
		[9, 'self_test_result', tuya.valueConverter.selfTestResult],
		[15, 'battery', tuya.valueConverter.raw]
	],
}`

What I tested:
1. Pressed the physical button, thereby triggering a test alarm
2. Increased the CO concentration until the CO alarm was triggered
3. Decreased the CO concentration
4. Silenced the alarm multiple times, since it kept going off for a few minutes despite being removed from the concentrated CO.

My guesses:
- DP 2 seems to be CO level, but I have no idea if or how the raw value needs to converted. 
- DP1 seems to be the alarm indicator (boolean), but the value was always 0 (I would expect any non-zero value here).
- DP 16 might be the "hush" button, i.e. when i silenced the alarm via the physical button.

Received messages (log):

> debug 2023-09-16 14:34:12: Received Zigbee message from 'CO_SENSOR', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":9}],"seq":7936}' from endpoint 1 with groupID 0
debug 2023-09-16 14:41:25: Received Zigbee message from 'CO_SENSOR', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":9}],"seq":8192}' from endpoint 1 with groupID 0
debug 2023-09-16 14:44:33: Received Zigbee message from 'CO_SENSOR', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":1}],"seq":8448}' from endpoint 1 with groupID 0
debug 2023-09-16 14:44:34: Received Zigbee message from 'CO_SENSOR', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,2,78],"type":"Buffer"},"datatype":2,"dp":2}],"seq":8704}' from endpoint 1 with groupID 0
debug 2023-09-16 14:44:54: Received Zigbee message from 'CO_SENSOR', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":1}],"seq":8960}' from endpoint 1 with groupID 0
debug 2023-09-16 14:44:54: Received Zigbee message from 'CO_SENSOR', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,2,70],"type":"Buffer"},"datatype":2,"dp":2}],"seq":9216}' from endpoint 1 with groupID 0
debug 2023-09-16 14:46:45: Received Zigbee message from 'CO_SENSOR', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":1}],"seq":9472}' from endpoint 1 with groupID 0
debug 2023-09-16 14:46:46: Received Zigbee message from 'CO_SENSOR', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,3,231],"type":"Buffer"},"datatype":2,"dp":2}],"seq":9728}' from endpoint 1 with groupID 0
debug 2023-09-16 14:48:31: Received Zigbee message from 'CO_SENSOR', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":1}],"seq":9984}' from endpoint 1 with groupID 0
debug 2023-09-16 14:48:32: Received Zigbee message from 'CO_SENSOR', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":1}],"seq":9984}' from endpoint 1 with groupID 0
debug 2023-09-16 14:48:32: Received Zigbee message from 'CO_SENSOR', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":1}],"seq":9984}' from endpoint 1 with groupID 0
debug 2023-09-16 14:48:33: Received Zigbee message from 'CO_SENSOR', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":1}],"seq":9984}' from endpoint 1 with groupID 0
debug 2023-09-16 14:48:33: Received Zigbee message from 'CO_SENSOR', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":1}],"seq":9984}' from endpoint 1 with groupID 0
debug 2023-09-16 14:48:34: Received Zigbee message from 'CO_SENSOR', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":16}],"seq":10496}' from endpoint 1 with groupID 0
debug 2023-09-16 14:48:35: Received Zigbee message from 'CO_SENSOR', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,104],"type":"Buffer"},"datatype":2,"dp":2}],"seq":10240}' from endpoint 1 with groupID 0
debug 2023-09-16 14:50:23: Received Zigbee message from 'CO_SENSOR', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":1}],"seq":10752}' from endpoint 1 with groupID 0
debug 2023-09-16 14:50:23: Received Zigbee message from 'CO_SENSOR', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,20],"type":"Buffer"},"datatype":2,"dp":2}],"seq":11008}' from endpoint 1 with groupID 0
debug 2023-09-16 14:50:48: Received Zigbee message from 'CO_SENSOR', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":1}],"seq":11264}' from endpoint 1 with groupID 0
debug 2023-09-16 14:50:49: Received Zigbee message from 'CO_SENSOR', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":16}],"seq":11776}' from endpoint 1 with groupID 0
debug 2023-09-16 14:50:50: Received Zigbee message from 'CO_SENSOR', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":2}],"seq":11520}' from endpoint 1 with groupID 0

Koenkk, 2023-09-21:
@follow-suit I've improved the logging a bit to make reverse engineering easier.

- First update to the latest dev in 2 hours from now
- Don't provide any datapoints `meta: { tuyaDatapoints: [ ]}`
- Make sure you use `fromZigbee: [tuya.fz.datapoints],`

Once the device reports a datapoint, you will now see `Datapoint XX not defined for 'XXX' with value XX` in the debug log

follow-suit, 2023-09-22:
@Koenkk Just tried that and I'm getting a similar log message for the test alarm: 

> Datapoint 9 not defined for '_TZE200_rjxqso4a' with data {"dp":9,"datatype":4,"data":{"type":"Buffer","data":[1]}}

However, I don't see any additional information compared to the lines I already collected:

> Received Zigbee message from 'CO_SENSOR', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":9}],"seq":21504}' from endpoint 1 with groupID 0

Did I make a mistake?

Koenkk, 2023-09-23:
Did you update to the latest dev branch? (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

follow-suit, 2023-09-23:
@Koenkk Yes.

[image]

Koenkk, 2023-09-23:
How are you running z2m?

Koenkk, 2023-09-23:
did you execute a 'npm ci' after updating?

NickJLange, 2023-09-24:
+1 As I'm also trying to integrate these devices. They are not very chatty devices, and I've been unable to get DP2 to change values ( I suppose that is a good thing...)

follow-suit, 2023-09-24:
@Koenkk 
> did you execute a 'npm ci' after updating?

Apparently, I haven't when I should have ;)

> Datapoint 9 not defined for '_TZE200_rjxqso4a' with value 1

So I guess what this now does is converting the 4 bytes into a decimal like `tuya.valueConverter.raw`? 
My problem with interpreting DP2 values was rather if the raw (decimal) value needs to be corrected, e.g. [divided by 100](https://github.com/Koenkk/zigbee-herdsman-converters/blob/9f31dbb90fe000db155be54b8db8600af42193f2/src/lib/legacy.ts#L4026C98-L4026C98) as implemented for the [TS0601_smart_CO_air_box](https://github.com/Koenkk/zigbee-herdsman-converters/blob/9f31dbb90fe000db155be54b8db8600af42193f2/src/devices/tuya.ts#L720). I did already manually calculate the decimal values, but I am still unable to know if these values directly correspond to ppm CO or if they are e.g. a multiple of the actual ppm...

follow-suit, 2023-09-24:
> +1 As I'm also trying to integrate these devices. They are not very chatty devices, and I've been unable to get DP2 to change values ( I suppose that is a good thing...)

This really depends on what you did to try and change DP2 values. I put my sensor into a sealed box (glass tupperware) with a smoldering piece of wood (a [joss stick](https://www.amazon.com/dp/B0CD55RNQZ/) might also work). It actually took quite a while before the alarm was triggered, though i didn't track the DP2 messages in real time, so the rising CO level might have been reported earlier. Of course, self-protection must be considered when experimenting with fire and/or carbon monoxide.

Koenkk, 2023-09-24:
> So I guess what this now does is converting the 4 bytes into a decimal like tuya.valueConverter.raw?

Yes this is what it does (it shows a more readable value)

> My problem with interpreting DP2 values was rather if the raw (decimal) value needs to be corrected,

What you could do is pair it to the TuYa gateway and see what values it reports there.

follow-suit, 2023-09-24:
@Koenkk  
> What you could do is pair it to the TuYa gateway and see what values it reports there.

I figured that's what it woudl take, but I would be happy to contribute. Slight problem is that I don't have one of these.
Any cheap recommendations? Or will [this one](https://www.aliexpress.com/item/1005005238326171.html) do the job?

Koenkk, 2023-09-24:
@follow-suit yes, that's the one I also have

follow-suit, 2023-10-03:
Quick update: I was able to confirm my guess regarding the Data Points via TuYa IoT Platform:

> {"1":"CO state","2":"CO value","9":"Auto-Detect Result","15":"Battery level","16":"Silence"}

In the TuYa app, I can also see the CO ppm (luckily 0 currently :smirk:). Since it is quite a hassle to raise the CO concentration, I want to make sure I have a plan on how to figure out the calibration formula.
Is there a way to read the raw value reported for DP 2 as it would be in Z2M? It seems that only the resulting ppm is logged:

[image]

follow-suit, 2023-10-04:
Here's my implementation suggestion and a few remarks:

- Not sure where the model name _**ZSS-HM-CO**_ is coming from (didn't see it in the manual), but I just copied it from this issue's title.
- PPM values are not calibrated for now, maybe some knowledge will appear as to whether this needs to be adjusted.
- The sensor does not seem to report any initial state values, so CO detection, CO level, and Self test result are **_N/A_** until a specific action initializes them. Battery % is reported continuously every few hours. TuYa app initially either reads or maybe just assumes default values (see my [previous post](https://github.com/Koenkk/zigbee2mqtt/issues/19011#issuecomment-1744552553)). @Koenkk Can something similar be done in Z2M?
- I did not get around to testing the converter (at least battery % is working), I will report back as soon as I could do so. Maybe someone else owning this sensor can test in parallel.

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE200_rjxqso4a']),
    model: 'ZSS-HM-CO',
    vendor: 'Moes',
    description: 'Carbon Monoxide Alarm',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [
        /* DP  1 */ e.carbon_monoxide(),
        /* DP  2 */ e.co(),
        /* DP  9 */ tuya.exposes.selfTestResult(),
        /* DP 15 */ e.battery(),
        /* DP 16 */ tuya.exposes.silence()
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'carbon_monoxide', tuya.valueConverter.trueFalse0],
            [2, 'co', tuya.valueConverter.raw],
            [9, 'self_test_result', tuya.valueConverter.selfTestResult],
            [15, 'battery', tuya.valueConverter.raw],
            [16, 'silence', tuya.valueConverter.raw]
        ]
    }
};

module.exports = definition;
```

follow-suit, 2023-10-04:
Test was mostly successful :smiley:
Again, I put the sensor into a sealed box with a smoldering piece of wood. Nothing happened for a few minutes, then suddenly the first CO level message was 721 ppm and simultaneously, "Carbon monoxide" switched to true (and of course the device started beeping).

However, I wasn't able to silence the alarm by means of the exposed switch. After enabling the switch, it gets reset as soon as the device decides to beep again. So maybe it makes sense to set DP 16 to read-only, but I have no idea how to do that.

[image]

closed 2023-10-21 by Koenkk

masp31, 2023-11-15:
HI. 
First of all I have to say that i am not a very skil user, and probably this is not the way to resolve my problem, nevetheless  I have pair a Carbon Monoxide Sensor Zigbee Model TS0601 Zigbee Manufacturer _TZE200_rjxqso4a

The zigbee2mqtt is the Current version: 1.33.2-1 but the  sensor not exposes: 

        / e.carbon_monoxide(), -  N/A
        / e.co(),  -   N/A

and report the other 

        /exposes.selfTestResult(), - Success
        / e.battery(), - 100%
        /silence() -  True/False
       / linkquality() -  114 lqi

I realy apreciate your sugestions to identify the problem of the two parameters N/A 

Thank you Miguel Pereira

MqlBql, 2025-01-18:
> HI. First of all I have to say that i am not a very skil user, and probably this is not the way to resolve my problem, nevetheless I have pair a Carbon Monoxide Sensor Zigbee Model TS0601 Zigbee Manufacturer _TZE200_rjxqso4a
> 
> The zigbee2mqtt is the Current version: 1.33.2-1 but the sensor not exposes:
> 
> ```
>     / e.carbon_monoxide(), -  N/A
>     / e.co(),  -   N/A
> ```
> 
> and report the other
> 
> ```
>     /exposes.selfTestResult(), - Success
>     / e.battery(), - 100%
>     /silence() -  True/False
>    / linkquality() -  114 lqi
> ```
> 
> I realy apreciate your sugestions to identify the problem of the two parameters N/A
> 
> Thank you Miguel Pereira
> 
> PS: I have now trie with the external_converters: and the tow parameters are now (Null)

Were you able to fix this issue ? I have the same issue with same sensor. Got only battery and link state.

masp31, 2025-01-20:
Hi MqlBql,

I did not fix the issue, i still have the same problem and nonetheless i give up.