Koenkk/zigbee2mqtt #19012: SNZB-02D: Failed to interview - Error: Interview failed because can not get node descriptor

flxtr, 2023-09-18:
### What happened?

I use Zigbee2MQTT as home assistant addon. (Home Assistant OS on RasbPi 4)
I tried to pair a Sonoff SNZB-02D Smart Temperature Humidity Sensor.

It failed to pair and the device is listed as not supported.
[image]

Another SNZB-02D (the same delivery) paired successfully.

I tried several things:
* Tried to delete the device after fail (which fails too)
* Forcefully deleted the device before the next try
* Connect the adapter with extension cable to the pi
* Hold the Sensor very close to the adapter
* Reset the Sensor by removing the battery several times
* Tried a fresh battery from my stock
* Tried die keep the Sensor alive by tapping the pairing button every 2 to 3 seconds
* Tried to pair via a Philips Hue bulb in another room
* Restarted Home Assistent and the whole pi
* Un- and re-pugged the adapter

### What did you expect to happen?

I did the device expect to pair successfully since it's supported.

### How to reproduce it (minimal and precise)

* Start pairing in z2m frontend
* Hold pair button on the sensor for about 5 seconds until the pairing symbol starts to blink
* Wait for the error message

### Zigbee2MQTT version

1.33.0

### Adapter firmware version

20210708

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle_Plus / zStack3x0

### Debug log

<details>
  <summary><b>Whole herdsman debug log [click to expand]</b></summary>
  <p><pre>
Zigbee2MQTT:debug 2023-09-18 22:25:45: Received MQTT message on 'zigbee2mqtt/bridge/request/device/remove' with data '{"block":false,"force":true,"id":"0x94deb8fffe360fce","transaction":"jhruq-9"}'
Zigbee2MQTT:info  2023-09-18 22:25:45: Removing device '0x94deb8fffe360fce' (block: false, force: true)
2023-09-18T20:25:45.233Z zigbee-herdsman:controller:database:log Writing database to '/config/zigbee2mqtt/database.db'
Zigbee2MQTT:debug 2023-09-18 22:25:45: Clearing Home Assistant discovery topic for '0x94deb8fffe360fce'
Zigbee2MQTT:info  2023-09-18 22:25:45: MQTT publish: topic 'zigbee2mqtt/0x94deb8fffe360fce', payload ''
Zigbee2MQTT:info  2023-09-18 22:25:45: Successfully removed device '0x94deb8fffe360fce' (block: false, force: true)
Zigbee2MQTT:info  2023-09-18 22:25:45: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/remove', payload '{"data":{"block":false,"force":true,"id":"0x94deb8fffe360fce"},"status":"ok","transaction":"jhruq-9"}'
2023-09-18T20:25:51.652Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,32,68,129,0,0,25,0,157,1,1,13,0,141,0,181,226,3,0,0,12,17,106,1,0,134,18,5,8,9,16,0,0,157,1,29,194]
2023-09-18T20:25:51.653Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,32,68,129,0,0,25,0,157,1,1,13,0,141,0,181,226,3,0,0,12,17,106,1,0,134,18,5,8,9,16,0,0,157,1,29,194]
2023-09-18T20:25:51.653Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 32 - 2 - 4 - 129 - [0,0,25,0,157,1,1,13,0,141,0,181,226,3,0,0,12,17,106,1,0,134,18,5,8,9,16,0,0,157,1,29] - 194
2023-09-18T20:25:51.653Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":25,"srcaddr":413,"srcendpoint":1,"dstendpoint":13,"wasbroadcast":0,"linkquality":141,"securityuse":0,"timestamp":254645,"transseqnumber":0,"len":12,"data":{"type":"Buffer","data":[17,106,1,0,134,18,5,8,9,16,0,0]}}
2023-09-18T20:25:51.656Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":106,"manufacturerCode":null,"commandIdentifier":1},"Payload":{"fieldControl":0,"manufacturerCode":4742,"imageType":2053,"fileVersion":4105},"Command":{"ID":1,"response":2,"parameters":[{"name":"fieldControl","type":32},{"name":"manufacturerCode","type":33},{"name":"imageType","type":33},{"name":"fileVersion","type":35}],"name":"queryNextImageRequest"}},"address":413,"endpoint":1,"linkquality":141,"groupID":0,"wasBroadcast":false,"destinationEndpoint":13}'
Zigbee2MQTT:debug 2023-09-18 22:25:51: Received Zigbee message from 'Klimasensor Wohnzimmer', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":4105,"imageType":2053,"manufacturerCode":4742}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2023-09-18 22:25:51: Device 'Klimasensor Wohnzimmer' requested OTA
2023-09-18T20:25:51.666Z zigbee-herdsman:controller:endpoint CommandResponse 0x94deb8fffe360e29/1 genOta.queryNextImageResponse({"status":152}, {"sendWhen":"active","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2023-09-18T20:25:51.666Z zigbee-herdsman:controller:endpoint Request Queue (0x94deb8fffe360e29/1): send request
2023-09-18T20:25:51.667Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x94deb8fffe360e29:413/1 (0,0,1)
2023-09-18T20:25:51.667Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":413,"destendpoint":1,"srcendpoint":1,"clusterid":25,"transid":248,"options":0,"radius":30,"len":4,"data":{"type":"Buffer","data":[25,156,2,152]}}
2023-09-18T20:25:51.668Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,14,36,1,157,1,1,1,25,0,248,0,30,4,25,156,2,152,83]
2023-09-18T20:25:51.669Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-09-18T20:25:51.672Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
[... 264 lines cut ...]
2023-09-18T20:28:28.499Z zigbee-herdsman:controller:device:log Interview - quirks did not match
2023-09-18T20:28:28.501Z zigbee-herdsman:controller:device:log Interview - failed for device '0x94deb8fffe360fce' with error 'Error: Interview failed because can not get node descriptor ('0x94deb8fffe360fce')
    at Device.interviewInternal (/app/node_modules/zigbee-herdsman/src/controller/model/device.ts:598:19)
    at Device.interview (/app/node_modules/zigbee-herdsman/src/controller/model/device.ts:488:13)
    at Controller.onDeviceJoined (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:579:17)'
2023-09-18T20:28:28.501Z zigbee-herdsman:controller:database:log Writing database to '/config/zigbee2mqtt/database.db'
2023-09-18T20:28:28.510Z zigbee-herdsman:controller:error Interview failed for '0x94deb8fffe360fce with error 'Error: Interview failed because can not get node descriptor ('0x94deb8fffe360fce')'
Zigbee2MQTT:error 2023-09-18 22:28:28: Failed to interview '0x94deb8fffe360fce', device has not successfully been paired
Zigbee2MQTT:info  2023-09-18 22:28:28: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x94deb8fffe360fce","ieee_address":"0x94deb8fffe360fce","status":"failed"},"type":"device_interview"}'
</pre></p>
</details>

The last lines:
```
2023-09-18T20:28:28.498Z zigbee-herdsman:controller:device:log Interview - quirks check for 'undefined'-'undefined'-'Unknown'
2023-09-18T20:28:28.499Z zigbee-herdsman:controller:device:log Interview - quirks did not match
2023-09-18T20:28:28.501Z zigbee-herdsman:controller:device:log Interview - failed for device '0x94deb8fffe360fce' with error 'Error: Interview failed because can not get node descriptor ('0x94deb8fffe360fce')
    at Device.interviewInternal (/app/node_modules/zigbee-herdsman/src/controller/model/device.ts:598:19)
    at Device.interview (/app/node_modules/zigbee-herdsman/src/controller/model/device.ts:488:13)
    at Controller.onDeviceJoined (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:579:17)'
2023-09-18T20:28:28.501Z zigbee-herdsman:controller:database:log Writing database to '/config/zigbee2mqtt/database.db'
2023-09-18T20:28:28.510Z zigbee-herdsman:controller:error Interview failed for '0x94deb8fffe360fce with error 'Error: Interview failed because can not get node descriptor ('0x94deb8fffe360fce')'
Zigbee2MQTT:error 2023-09-18 22:28:28: Failed to interview '0x94deb8fffe360fce', device has not successfully been paired
Zigbee2MQTT:info  2023-09-18 22:28:28: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x94deb8fffe360fce","ieee_address":"0x94deb8fffe360fce","status":"failed"},"type":"device_interview"}'
```

flxtr, 2023-09-22:
Is there anything else I can try? Or have I got a defective device?

thebeetleuk, 2023-09-25:
I have similar issues with all the Sonoff devices. Its really hit and miss. sometimes it joins as unknown and other times it just doesn't join at all. No idea what's going on :(

lcaparros, 2024-05-13:
Exactly the same error here with SNZB-02, the one without display. Logs are simple:
```
Info 2024-05-13 13:36:08Interview for '0x00124b002a6cce6b' started
Info 2024-05-13 13:36:08Device '0x00124b002a6cce6b' joined
Info 2024-05-13 13:36:08Starting interview of '0x00124b002a6cce6b'
Error 2024-05-13 13:42:11Interview failed for '0x00124b002a6cce6b with error 'Error: Interview failed because can not get node descriptor ('0x00124b002a6cce6b')'
Error 2024-05-13 13:42:11Failed to interview '0x00124b002a6cce6b', device has not successfully been paired
```

dboring, 2024-07-21:
I was having this same issue and tried several times to pair a SNZB-02D sensor. I ended up replacing the battery with a brand new Panasonic CR2450 that I had in my drawer (even though the battery indicator showed it being 100%, but these are lithium batteries and they don't report accurately), and pressed the button for 5 seconds. **It paired and the interview completed successfully!** After the interview was complete, I put the no-name factory battery back in, and it seems to keep working.

This is a trick I learned for Aqara devices, and it appears that it may also be needed for Sonoff devices.

idefix38, 2024-12-14:
have 2 SONOFF SNZB-02 sensors, two of them work perfectly, and the other two have the issue described above. All 4 sensors worked very well with a ZIGBee dconz stick, but after I changed the stick to a SONOFF_Zigbee_3.0_USB_Dongle_Plus-E, it's impossible to pair some SONOFF sensors.
 
 `[2024-12-14 21:59:38] error: 	zh:controller: Interview failed for '0x00124b00226943f2 with error 'Error: Interview failed because can not get node descriptor ('0x00124b00226943f2')'
[2024-12-14 21:59:38] error: 	z2m: Failed to interview '0x00124b00226943f2', device has not successfully been paired`

otto2704, 2025-01-10:
> I was having this same issue and tried several times to pair a SNZB-02D sensor. I ended up replacing the battery with a brand new Panasonic CR2450 that I had in my drawer (even though the battery [...]

thanks a LOT for this! was already desperating, but with fresh batteries my sonoff trv is also working now!

hugochmel, 2025-01-25:
I have same problem, the senzor is fresh unoacked and battery had the paper contact protection.. ill try replace battery

gvlekke, 2025-02-25:
I think our issue is related: https://github.com/Koenkk/zigbee2mqtt/issues/25438

mikolajpiecuch, 2025-03-23:
Had the same issue with Tuya temperature sensor. I replaced batteries to new ones and interview completed with success :)

closed 2025-05-30 by github-actions[bot]

igittigitt, 2025-06-05:
I have the same problem, even with outlet powered devices and devices that where in the network before. Does Z2M somehow "remember" that it has seen an node before? Yes, i deleted it before trying to repair.
And what does "Interview failed because can not get node descriptor" mean at all?

Lampie01, 2025-06-14:
(coppied from a discussion on this github because i tink it is related)
I myself bought two new Frient SMSZB-120 and wanted to add them to the 2 existing smoke alarms from Frient (same type) this week. I also get the "Interview failed because can not get node descriptor" with both new detectors. Also when I remove one of the two detectors that was paired for several months and try to pair it again, I also get the message "Interview failed because can not get node descriptor". I see several other messages with all kinds of solutions here on Github such as replacing the battery, rebooting, etc. but nothing helped. What can I contribute to possibly solve this or am I hijacking the wrong topic?

GameWatchProjects, 2025-09-19:
Hello, same problem here with 3 x IKEA E1746, can not pair with SONOFF Zigbee 3.0 USB Dongle Plus (TI CC2652P + CP2102N)over z2m:

```
[2025-09-19 10:30:02] error: 	zh:controller: Interview failed for '0x5c0272fffe5b80da with error 'Error: Interview failed because can not get node descriptor ('0x5c0272fffe5b80da')'
[2025-09-19 10:30:02] error: 	z2m: Failed to interview '0x5c0272fffe5b80da', device has not successfully been paired
```

Supreme-Bob, 2025-10-25:
I'm getting similar issues with TS011F _TZ3210_cehuw1lw (Smart plug (with power monitoring)). I have 20 of them added them all a few months ago then removed(didn't block them) the 6 I didn't end up using from Z2M. I can't rejoin the any of the ones i removed now. Not sure whats going on, clearly they work because I've joined them before.

Koenkk, 2026-01-04:
@engels0n try permitting joining only via the coordinator and pair them close to it.

engels0n, 2026-01-06:
Ok, before I start sniffing, I tried it again today and this is what I get from the logs all the time:

```
`[6.1.2026, 08:45:04] z2m: Starting interview of '0x70b3d52b60167794'

[6.1.2026, 08:45:10] zh:ember: Delivery of BROADCAST failed for '65533'.

[6.1.2026, 08:45:17] z2m: Device '0x70b3d52b60167794' left the network

[6.1.2026, 08:45:17] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x70b3d52b60167794","ieee_address":"0x70b3d52b60167794"},"type":"device_leave"}'

[6.1.2026, 08:45:33] zh:controller: Interview failed for '0x70b3d52b60167794 with error 'Error: DatabaseEntry with ID '143' does not exist'`
```

Could the Broadcast error be the problem?

Koenkk, 2026-01-07:
That could be indeed, not sure why it would fail (we should see that in the sniff)