Koenkk/zigbee2mqtt #19018: [New device support]:  Viessmann Climate Sensor

Kasper027, 2023-09-19:
### Link

https://www.loebbeshop.de/viessmann-vicare-klimasensor-zk03839/

### Database entry

{"id":7,"type":"EndDevice","ieeeAddr":"0x5cc7c1fffea229a1","nwkAddr":42830,"manufId":4641,"manufName":"Viessmann","powerSource":"Battery","modelId":"7637435","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,1,3,10,32,1026,1029,2821],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"7637435","manufacturerName":"Viessmann","powerSource":3,"zclVersion":3,"appVersion":0,"stackVersion":12,"hwVersion":2,"dateCode":"20000101","swBuildId":"02.07.0000"}},"genPollCtrl":{"attributes":{"checkinInterval":1200}},"msTemperatureMeasurement":{"attributes":{"measuredValue":2143}},"msRelativeHumidity":{"attributes":{"measuredValue":6540}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":170,"batteryVoltage":28}},"hvacUserInterfaceCfg":{"attributes":{}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2f40fb","endpointID":1},{"cluster":1029,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2f40fb","endpointID":1},{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2f40fb","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2f40fb","endpointID":1},{"cluster":3,"type":"endpoint","deviceIeeeAddress":"0x5cc7c1fffea229a1","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":12,"hwVersion":2,"dateCode":"20000101","swBuildId":"02.07.0000","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1695097663502,"defaultSendRequestWhen":"fastpoll","checkinInterval":300}

### Comments

It Works with the external Converter

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['7637435'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'ZK03839', // Vendor model number, look on the device for a model number
    vendor: 'Viessmann', // Vendor of the device (only used for documentation and startup logging)
    description: 'ViCare Climasensor', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.temperature, fz.humidity, fz.battery], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.battery(), e.temperature(), e.humidity()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
};

module.exports = definition;
```

Koenkk, 2023-09-19:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-09-19 by Koenkk

tuxiano, 2024-03-04:
@Kasper027 Hi, is the device working for you? I can pair it, but I don't see any measurements. Thank you

Kasper027, 2024-03-05:
Yes, it works fine. 
[image]

tuxiano, 2024-03-17:
@Kasper027 Please :)

Kasper027, 2024-03-19:
I cant help you. I don't have any programming skills. @Koenkk can you help?

tuxiano, 2024-03-19:
@Kasper027  Sure, I only would like to hear your feedback on the pairing process, as described in #21671 .

Kasper027, 2024-03-19:
@tuxiano I cant say it. I connected directly with Home-Assistant. I never connect it to ViCare