Koenkk/zigbee2mqtt #19026: GiEX - QT06_2 (_TZE200_a7sghmms) - State not define anymore

cloudbr34k84, 2023-09-20:
### What happened?

For some reason the State for this device is not defined anymore

[image]

### Zigbee2MQTT version

 1.33.0-dev commit: 0a69bb5

### Adapter firmware version

20230716

### Adapter

Zigbee ESP32GW (Zigstar)

Koenkk, 2023-09-20:
Could it be that you are using any external converter for this device?

cloudbr34k84, 2023-09-20:
> Could it be that you are using an external converter for this device?

No external converter, I only have 2,  the solar siren and the pool tester you are helping me on

Koenkk, 2023-09-21:
Pushed a fix!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-09-21 by Koenkk

cloudbr34k84, 2023-09-21:
> Pushed a fix!
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Oh nice!

What happened??

Koenkk, 2023-09-23:
Bug in the frontend fixed by https://github.com/nurikk/zigbee2mqtt-frontend/commit/912b1b0d0f2c0a2f252ddbeffdc2bdf909bbb042