Koenkk/zigbee2mqtt #19029: OTA AQARA H1 switches

Jens-Wymeersch, 2023-09-20:
### What happened?

Not all switches are visible on the OTA dashboard. Xiaomi WS-EUK04 isn't showing up (although that it says otherwise on the documentation).
Secondly, for the QBKG32LM, the OTA isn't working. Basically, it updates from null till null. 

### What did you expect to happen?

That all switches are visible in the OTA dashboard and can be updated when applicable.

### Zigbee2MQTT version

1.33.0 

### Adapter firmware version

20230507

### Adapter

SLZB-06

Jens-Wymeersch, 2023-09-30:
@Koenkk Besides the problems listed above related to OTA, I've noticed that this is valid for all aqara related devices. Anyway around this besides using the aqara hub itself?

erasmus83, 2023-10-08:
I have an WS-EUK01 & WS-EUK02 - also have the same issue that they are not visible in OTA dashboard

closed 2023-11-16 by github-actions[bot]

sozoalvin, 2023-11-16:
have the same problem - QBKG32LM 
tried OTA update but fails with this reason code in z2m logs. 

been trying to debug for days - can't seem to find out the reason why 

Update of '0x24df44100068d456' failed (Update failed with reason: 'undefined')

ahma, 2023-11-17:
Still an issue.

sozoalvin, 2023-11-18:
> Still an issue.

are you getting the same logs as I am?

ahma, 2023-11-18:
@sozoalvin , yeah the same error:
`Error 2023-11-18 22:25:28Update of '0x54ef4410005d6de6' failed (Update failed with reason: 'undefined')`
Furthermore, until I tried to update the switch, it worked fine. But since the update, it hasn't been responsive at all. :(

Jens-Wymeersch, 2023-11-19:
@Koenkk you have created an amazing platform, and I'm extremely grateful for your efforts. OTA (where possible) as you know is extremely important as the devices need updates. For example, I see one of the H1 switches is acting out and this most probably will be sorted with a firmware update. Can you please have a look at what needs to happen as it will benefit everyone who uses your platform

Koenkk, 2023-11-19:
We don't have an OTA for this device, to get it follow https://www.zigbee2mqtt.io/advanced/more/tuya_xiaomi_ota_url.html