Koenkk/zigbee2mqtt #19033: Transmit power doesn't work

jonathan-gatard, 2023-09-20:
### What happened?

Hi,

When I change transmit power, it has no effect on my device
[image]

The 1st red circle : 5dBm
The 2nd red circle : 20dBm
The 3rd red circle : 0dBm

My LQI is still arround 65

### What did you expect to happen?

I expected to gain some LQI by changing transmit_power

### How to reproduce it (minimal and precise)

Use this env var:
ZIGBEE2MQTT_CONFIG_ADVANCED_TRANSMIT_POWER=5
ZIGBEE2MQTT_CONFIG_ADVANCED_TRANSMIT_POWER=0
ZIGBEE2MQTT_CONFIG_ADVANCED_TRANSMIT_POWER=20

Then update container : docker-compose up -d zigbee2mqtt

### Zigbee2MQTT version

1.32.2

### Adapter firmware version

20230507

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus (zStack3x0) (ZBDongle-P)

### Debug log

2023-09-20 21:07:53: Set transmit power to '5'
2023-09-20 21:12:02: Set transmit power to '20'
2023-09-20 21:17:35: Set transmit power to '0'

francisp2, 2023-09-21:
Transmit power regulates how load your coordinator yells, but has no effect on how well it listens

Shred99, 2023-09-26:
Raising the Tx power can be really useful to enhance communication between multiple devices that support it.  I have a 'backbone' of coordinator + 2 x router, all of which use a slightly higher transmit power to ensure reliable communication between them.

It does cause problems for some battery powered sensors though - the higher transmit power sometimes fools them in to associating with a non-optimal router.

closed 2023-09-26 by jonathan-gatard