Koenkk/zigbee2mqtt #19042: [New device support]: EMIZB-141 Power Meter

lSh4dowl, 2023-09-21:
### Link

https://www.amazon.de/Electricity-Zählerauslesung-Energieverbrauchsüberwachung-LED-Impulsen-funktioniert/dp/B0CGVB6LGC

### Database entry

{"id":54,"type":"EndDevice","ieeeAddr":"0x0015bc001b100a56","nwkAddr":51420,"manufId":4117,"manufName":"frient A/S","powerSource":"Battery","modelId":"EMIZB-141","epList":[1,2],"endpoints":{"1":{"profId":49353,"epId":1,"devId":1,"inClusterList":[5,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":83,"inClusterList":[0,1,3,32,1794,2817,2821],"outClusterList":[3,10,25],"clusters":{"genBasic":{"attributes":{}},"genPollCtrl":{"attributes":{"checkinInterval":14400}},"seMetering":{"attributes":{"currentSummDelivered":[0],"instantaneousDemand":354,"multiplier":1,"divisor":1000,"develcoPulseConfiguration":10000,"develcoInterfaceMode":0}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c2ad1c","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c2ad1c","endpointID":1}],"configuredReportings":[{"cluster":1794,"attrId":1024,"minRepIntval":5,"maxRepIntval":3600,"repChange":1}],"meta":{}}},"dateCode":"2023-09-06 11:54","zclVersion":7,"interviewCompleted":true,"meta":{"configured":-570313272},"lastSeen":1695324832904,"defaultSendRequestWhen":"active","checkinInterval":3600}

### Comments

It works reasonable, but shows no battery for example.
I basically just copied an old Version of the previous version (https://www.zigbee2mqtt.io/devices/ZHEMI101.html)
I only linked a German Amazon Page, because the Vendor doesn't have a site for the product yet (this is the old version: https://frient.com/products/electricity-meter-interface/)

### External converter

```shell
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const constants = require('zigbee-herdsman-converters/lib/constants');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const utils = require('zigbee-herdsman-converters/lib/utils');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;

// develco specific cosntants
const manufacturerOptions = {manufacturerCode: 0x1015};

/* MOSZB-1xx - ledControl - bitmap8 - r/w
 * 0x00 Disable LED when movement is detected.
 * 0x01 Enables periodic fault flashes. These flashes are used to indicate e.g. low battery level.
 * 0x02 Enables green application defined LED. This is e.g. used to indicate motion detection.
 * Default value 0xFF ( seems to be fault + motion)
 */
const develcoLedControlMap = {
    0x00: 'off',
    0x01: 'fault_only',
    0x02: 'motion_only',
    0xFF: 'both',
};

// develco specific convertors
const develco = {
    configure: {
        read_sw_hw_version: async (device, logger) => {
            for (const ep of device.endpoints) {
                if (ep.supportsInputCluster('genBasic')) {
                    try {
                        const data = await ep.read('genBasic', ['develcoPrimarySwVersion', 'develcoPrimaryHwVersion'],
                            manufacturerOptions);

                        if (data.hasOwnProperty('develcoPrimarySwVersion')) {
                            device.softwareBuildID = data.develcoPrimarySwVersion.join('.');
                        }

                        if (data.hasOwnProperty('develcoPrimaryHwVersion')) {
                            device.hardwareVersion = data.develcoPrimaryHwVersion.join('.');
                        }
                    } catch (error) {/* catch timeouts of sleeping devices */}
                    break;
                }
            }
        },
    },
    fz: {
        // Some Develco devices report strange values sometimes
        // https://github.com/Koenkk/zigbee2mqtt/issues/13329
[... 243 lines cut ...]
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(2);
            await reporting.bind(endpoint, coordinatorEndpoint, ['seMetering']);
            await reporting.instantaneousDemand(endpoint);
            await reporting.readMeteringMultiplierDivisor(endpoint);
        },
        exposes: [
            e.power(),
            e.energy(),
            e.battery_low(),
            exposes.numeric('pulse_configuration', ea.ALL).withValueMin(0).withValueMax(65535)
                .withDescription('Pulses per kwh. Default 1000 imp/kWh. Range 0 to 65535'),
            exposes.enum('interface_mode', ea.ALL,
                ['electricity', 'gas', 'water', 'kamstrup-kmp', 'linky', 'IEC62056-21', 'DSMR-2.3', 'DSMR-4.0'])
                .withDescription('Operating mode/probe'),
            exposes.numeric('current_summation', ea.SET)
                .withDescription('Current summation value sent to the display. e.g. 570 = 0,570 kWh').withValueMin(0)
                .withValueMax(268435455),
            exposes.binary('check_meter', ea.STATE, true, false)
                .withDescription('Is true if communication problem with meter is experienced'),
        ],
};

module.exports = definition;
```

Koenkk, 2023-09-23:
Could you make a PR to add support for this device?

lSh4dowl, 2023-09-25:
I am still testing and experiencing some weirdness, where I see less usage than in reality, however, I don’t know if it is due to my power meter or the device, since I have no other reader device to compare to.

lSh4dowl, 2023-10-01:
I could need some help with this :)

Readings are definitely not accurate when set to the right frequency, also another problem I have noticed is that I now have a small solar system and it seems to add the generated overflow energy that goes out to the total.

j91321, 2023-10-01:
I've encountered the same problem, it's mentioned even on develco website that this is not compatible with PV systems, it's doesn't happen to 100% of electrometers, but some of them are configured to give impulse also on energy export and you have no way of distinguishing import from export.

lSh4dowl, 2023-10-01:
my system seems to have 2 IR Lights, so I wonder if one is dedicated for export and one for import, might test this out by just covering one up.

j91321, 2023-10-01:
Many systems have multiple lights, my also has 2, but one shows imp./kWh the other imp./kvarh. Best to find and consult manual.

lSh4dowl, 2023-10-01:
That would be a bummer, since to my knowledge this is the only zigbee power meter, that’s why I bought it and this makes it practically useless :/

j91321, 2023-10-01:
There are other power meters like [TuYa TS0601](https://www.zigbee2mqtt.io/devices/TS0601_3_phase_clamp_meter.html) although the installation is definitely more complex than the LED one.

KaytheDays, 2023-10-02:
Will this device be officially supported ? I am considering it, as it seems the device with the most simple installation out there … 
https://www.develcoproducts.com/de/produkte/zaehlermodule/modul-fuer-stromzaehler-2-led/#EMI2LEDAngaben

petosiso, 2023-10-17:
I had same problem yesterday, this worked for me:

> Via the web terminal, I could create my converter file per the Z2M guide and place it in /config/zigbee2mqtt/<filename>.js (that's what the guide means by "This file has to be created next to the configuration.yaml", because the configuration.yaml file for Z2M is in /config/zigbee2mqtt, at least in HassOS).

> Then, to tell Z2M to use the converter file, go to the Z2M UI, then click "Settings" > "External Converters". Enter the filename of the converter file you just created ("SPP02GIP.js" in my case), and click "Submit", then "Restart".

> The external converter file should now be in use by Z2M, which you can verify in the logs per the guide linked in the OP.

AlmightyCZ, 2023-11-01:
@petosiso Can you confirm, that the device is working correctly when you don't have a solar system?

penroseg, 2023-11-02:
> @petosiso Can you confirm, that the device is working correctly when you don't have a solar system?

Hi @petosiso. I can confirm it's working well on a non-solar system meter.  Have had it running for about 3 weeks (using the custom converter shared by @lSh4dowl. Including showing up in Energy Dashboard.

There are issues with some fields (Battery mainly) and the Summation resets to Zero no matter what is entered.  I was hoping to use that latter field to input my starting Meter number and have it up from there but from reading back it appears that was an issue with the v1 meter also.  Maybe when 'proper' support comes to Z2M it will work but for the moment it is more than enough for me.

I've also set up Utility meters for Hour, Day & Month so I can see a visual.  I used a version of this article here to set up my Day, Night & Peak rates. https://community.home-assistant.io/t/stuck-setting-up-peak-off-peak-electricity-tariff/419894/5

[image]
 
[image]

[image]

petosiso, 2023-11-03:
@AlmightyCZ yes, I can also confirm that it works. Exactly as @penroseg described,
[image]
[image]

jllarraz, 2023-11-23:
Does anyone knows when is this to be added to the list of supported devices?

AlmightyCZ, 2023-12-02:
It's now supported in 1.34.0(https://github.com/Koenkk/zigbee-herdsman-converters/pull/6582 + https://www.zigbee2mqtt.io/devices/EMIZB-141.html)

ywaf, 2023-12-03:
> Great news that it is now supported in 1.34.0, but for some reason I don't get an energy reading. Does anybody face the same issue or have a fix? I tried reconfiguring. <img alt="Skærmbillede 2023-12-02 kl 11 55 09" width="446" src="https://private-user-images.githubusercontent.com/32738355/287443104-32479b86-fbe9-407b-97e1-e176caea2190.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDE2MDE3MTUsIm5iZiI6MTcwMTYwMTQxNSwicGF0aCI6Ii8zMjczODM1NS8yODc0NDMxMDQtMzI0NzliODYtZmJlOS00MDdiLTk3ZTEtZTE3NmNhZWEyMTkwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjAzVDExMDMzNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWEyYzE4ZGEzMDc2YjQ4NGJiNjA4M2NhZWNiNjA4ZTQ2Y2Y3YjdlZGY0ZTVlYTJlYTM0MDMxZTFiZWZlNmU0MzQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.t5djjBO2TILPtKmNO-kxHeA-SoM-kD1AjJljMxLpTho">

you have to set the pulse setting which isnt there in the official build, im currently working on a pr, or use the custom external converter

alerich-x, 2023-12-06:
> I've encountered the same problem, it's mentioned even on develco website that this is not compatible with PV systems, it doesn't happen to 100% of electrometers, but some of them are configured to give impulse also on energy export and you have no way of distinguishing import from export (if you have on-grid setup for solar)

So this is a hardware issue and the device support in Z2MQTT doesn't make any difference?

akostamo, 2023-12-11:
This custom converter works so much better than what is currently in the release version, so eagerly waiting this to be the officially released!

dawiinci, 2023-12-13:
> > Great news that it is now supported in 1.34.0, but for some reason I don't get an energy reading. Does anybody face the same issue or have a fix? I tried reconfiguring.
> 
> you have to set the pulse setting which isnt there in the official build, im currently working on a pr, or use the custom external converter

That is good news. What are your plans on the PR? I am considering using the external converter to make energy work, but this might interfere with the PR, right?

ywaf, 2023-12-13:
> That is good news. What are your plans on the PR? I am considering using the external converter to make energy work, but this might interfere with the PR, right?

nah just remove the custom converter if u wanna go back to normal converter

dawiinci, 2023-12-22:
In the external converter I am missing the battery indicator in %. Is there a way to add it?

AlmightyCZ, 2023-12-30:
> In the external converter I am missing the battery indicator in %. Is there a way to add it?

The converter in release is reporting battery percentage, but it's probably not working. Despite showing 100% consistently, my meter stopped updating (after 60 days of operation), indicating battery depletion. After replacing the batteries, it started working correctly again.

AlmightyCZ, 2023-12-30:
Also the energy reporting has stoped working (as someone has already mentioned here) so I'm going back to the original external converter for now.

fumantsu, 2024-01-04:
I will agree with the use of external converter. It is much more complete. I managed to set the current summation but to be fair not sure what should be. Is it the "current value" in kWh that I see in the power box? (for the Czech people I have PRE).

nivek1612, 2024-01-09:
Used the external converter mentioned above. Running for 5 days and the tracking of my usage matches the electricity meter perfectly.

jllarraz, 2024-01-25:
I have used the external converter but the values (energy and power) seems a bit odd to me, can someone point me in the right direction
`{
    "battery": 60,
    "battery_low": false,
    "energy": 4295032.83,
    "interface_mode": 65534,
    "linkquality": 255,
    "power": -8388608,
    "pulse_configuration": 1000,
    "voltage": 2700,
    "check_meter": null,
    "current_summation": null
} `

akostamo, 2024-01-25:
> I have used the external converter but the values (energy and power) seems a bit odd to me, can someone point me in the right direction `{ "battery": 60, "battery_low": false, "energy": 4295032.83, [...]

Check from your energy meter, there should be pulse rate mentioned near the pulse led. Default is 1000 per kwh like in my example picture attached. Just adjust the pulse configuration.

[image]

jllarraz, 2024-01-26:
> > I have used the external converter but the values (energy and power) seems a bit odd to me, can someone point me in the right direction `{ "battery": 60, "battery_low": false, "energy": 4295032.83, "interface_mode": 65534, "linkquality": 255, "power": -8388608, "pulse_configuration": 1000, "voltage": 2700, "check_meter": null, "current_summation": null } `
> 
> Check from your energy meter, there should be pulse rate mentioned near the pulse led. Default is 1000 per kwh like in my example picture attached. Just adjust the pulse configuration.
> 
> ![Screenshot_2024-01-25-19-48-50-19_ab7988c7b00b15bc78ec5a428c58236f](https://private-user-images.githubusercontent.com/100792730/299761116-845c6311-bae5-4772-8c6a-cbf880a7ebbf.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDYyNjE0MDMsIm5iZiI6MTcwNjI2MTEwMywicGF0aCI6Ii8xMDA3OTI3MzAvMjk5NzYxMTE2LTg0NWM2MzExLWJhZTUtNDc3Mi04YzZhLWNiZjg4MGE3ZWJiZi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEyNlQwOTI1MDNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wNGQxYWRlZDRjNDAxZTQ2MDRmN2QyNjg1ZmYxMzczOGIwZWQ0OGI0NzRmMTdkMDdlYjFmNzYxYzVlYzExM2RjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.wo2m_fHHxluK6aw9dVRzpxKk-QJWbG0RKJByn1F5Qr8)

I checked and is 1000 imp/kwh so wondering if I have to do something else

nivek1612, 2024-01-26:
Your power reading is negative which is impossible unless you are exporting energy to the grid. Even if you are thats a huge number. 

This is what mine looks like and its tracking the usage very accurately. 

{
    "battery": 100,
    "battery_low": false,
    "energy": 660.83,
    "interface_mode": "electricity",
    "linkquality": 48,
    "power": 3234,
    "pulse_configuration": 1000,
    "voltage": 3100,
    "check_meter": null,
    "current_summation": null
} 

maybe do a reset of the frient and check you put the sensor very accurately over the pulse LED.

reset procedure is here 
https://help.frient.com/en/frient/electricity-meter-interface-2-led/t/cllmabm0o17vmkv0unrk3y34j

jllarraz, 2024-01-26:
> Your power reading is negative which is impossible unless you are exporting energy to the grid. Even if you are thats a huge number.
>
> This is what mine looks like and its tracking the usage very [...]

thanks, reset it to factory seems to have worked
{
    "battery": 60,
    "battery_low": null,
    "current_summation": null,
    "energy": 65.54,
    "interface_mode": "electricity",
    "linkquality": 255,
    "power": 286,
    "pulse_configuration": 1000,
    "voltage": 2700,
    "check_meter": null
}

PaJaSoft, 2024-02-01:
Guys, is there a plan to create a new PR for this external converter to be a standard part of z2m converters?

In the default released version of z2m the EMIZB-141 is supported (recognized) but I'm lacking almost every setting (LED pulses EE meter) including resetting kWh summary and setting pulses value per kWh (in my case I need 500 pulses/kWh, factory default is 1000).

I've been using this external converted for awhile and it seems reliable and accurate as far as I can compare the measurement and stability.

So I'm asking is there something known what is blocking the release now? If necessary I would help with the PR.

akostamo, 2024-02-01:
I think It's the lack of battery indication is this to be pulled into the official one, but.. no-one has time to get it right. :(

jllarraz, 2024-02-02:
If the only thing missing is the battery level, I modified the converter to add the battery level. Here you have the new version

`
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const constants = require('zigbee-herdsman-converters/lib/constants');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const utils = require('zigbee-herdsman-converters/lib/utils');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;

// develco specific cosntants
const manufacturerOptions = {manufacturerCode: 0x1015};

/* MOSZB-1xx - ledControl - bitmap8 - r/w
 * 0x00 Disable LED when movement is detected.
 * 0x01 Enables periodic fault flashes. These flashes are used to indicate e.g. low battery level.
 * 0x02 Enables green application defined LED. This is e.g. used to indicate motion detection.
 * Default value 0xFF ( seems to be fault + motion)
 */
const develcoLedControlMap = {
    0x00: 'off',
    0x01: 'fault_only',
    0x02: 'motion_only',
    0xFF: 'both',
};

// develco specific convertors
const develco = {
    configure: {
        read_sw_hw_version: async (device, logger) => {
            for (const ep of device.endpoints) {
                if (ep.supportsInputCluster('genBasic')) {
                    try {
                        const data = await ep.read('genBasic', ['develcoPrimarySwVersion', 'develcoPrimaryHwVersion'],
                            manufacturerOptions);

                        if (data.hasOwnProperty('develcoPrimarySwVersion')) {
                            device.softwareBuildID = data.develcoPrimarySwVersion.join('.');
                        }

                        if (data.hasOwnProperty('develcoPrimaryHwVersion')) {
                            device.hardwareVersion = data.develcoPrimaryHwVersion.join('.');
                        }
                    } catch (error) {/* catch timeouts of sleeping devices */}
                    break;
                }
            }
        },
    },
    fz: {
        // Some Develco devices report strange values sometimes
        // https://github.com/Koenkk/zigbee2mqtt/issues/13329
        electrical_measurement: {
            ...fz.electrical_measurement,
            convert: (model, msg, publish, options, meta) => {
                if (msg.data.rmsVoltage !== 0xFFFF && msg.data.rmsCurrent !== 0xFFFF && msg.data.activePower !== -0x8000) {
                    return fz.electrical_measurement.convert(model, msg, publish, options, meta);
                }
            },
        },
        device_temperature: {
            ...fz.device_temperature,
            convert: (model, msg, publish, options, meta) => {
                if (msg.data.currentTemperature !== -0x8000) {
                    return fz.device_temperature.convert(model, msg, publish, options, meta);
                }
[... 230 lines cut ...]
            const endpoint = device.getEndpoint(2);
            await reporting.bind(endpoint, coordinatorEndpoint, ['seMetering']);
            await reporting.instantaneousDemand(endpoint);
            await reporting.readMeteringMultiplierDivisor(endpoint);
        },
        exposes: [
            e.power(),
            e.energy(),
            e.battery_low(),
            e.battery(),
            exposes.numeric('pulse_configuration', ea.ALL).withValueMin(0).withValueMax(65535)
                .withDescription('Pulses per kwh. Default 1000 imp/kWh. Range 0 to 65535'),
            exposes.enum('interface_mode', ea.ALL,
                ['electricity', 'gas', 'water', 'kamstrup-kmp', 'linky', 'IEC62056-21', 'DSMR-2.3', 'DSMR-4.0'])
                .withDescription('Operating mode/probe'),
            exposes.numeric('current_summation', ea.SET)
                .withDescription('Current summation value sent to the display. e.g. 570 = 0,570 kWh').withValueMin(0)
                .withValueMax(268435455),
            exposes.binary('check_meter', ea.STATE, true, false)
                .withDescription('Is true if communication problem with meter is experienced'),
        ],
};

module.exports = definition;
`

skonteron, 2024-02-09:
Maybe someone here can help. I have created the file emizb_141.js in the directory /homeassistant/zigbee2mqtt. Then I integrated the emizb_141.js file using Zigbee2MQTT -> Settings -> external converter. 

[image]
[image]

Unfortunately, I still do not get any energy displayed. 
After inserting the json file I reset the frient.

[image]
[image]

What am I doing wrong?

jllarraz, 2024-02-09:
Did you click on submit?

PaJaSoft, 2024-02-09:
Did you restart z2m instance?

skonteron, 2024-02-09:
I have restarted both z2m and HA. After the restart my js file was still there

nivek1612, 2024-02-09:
Try the original convertor at the start of this thread

skonteron, 2024-02-09:
Perfect :) with the orginal convertor it's working. I get this now.
I think thats it. Thank you so much for your help

> "battery": 100,
    "battery_low": false,
    "energy": 3.18,
    "linkquality": 15,
    "power": 452,
    "voltage": 3000,
    "check_meter": null,
    "current_summation": 138329565,
    "interface_mode": "electricity",
    "pulse_configuration": null

PaJaSoft, 2024-02-09:
@skonteron Check if your elmeter uses 1000 impulses/1kWh (`"pulse_configuration": null`) or set it up correctly...

skonteron, 2024-02-09:
Thx for the advice. It should be 10.000. I will try

penroseg, 2024-02-10:
@jllarraz that new version with the Battery didn't work for me unfortunately. I _did_ get Battery level displayed but anything after that in the convertor file is not longer being made available. I have been using the original convertor since October and additional fields were available so I think there _may_ be a bug in the code somewhere but I'm not able to correct it. I did have a quick try at it but got an invalid convertor message in Z2M so gave up before breaking my integration totally.

I can see those are in your latest version too but for some reason are not now being integrated.

Using the current connector the shorter list is exposed.
[image]

Using the original connector the longer list is available, including the Pulse Configuration entity.
[image]

jllarraz, 2024-02-10:
Apologies for that I left a bit of code on that adapter that shouldnt be there.

this is the screenshoot of my Adapter 
[image]

The modification is as simple as changing this 

` exposes: [
            e.power(),
            e.energy(),
            e.battery_low(),
            exposes.numeric('pulse_configuration', ea.ALL).withValueMin(0).withValueMax(65535)
                .withDescription('Pulses per kwh. Default 1000 imp/kWh. Range 0 to 65535'),
            exposes.enum('interface_mode', ea.ALL,
                ['electricity', 'gas', 'water', 'kamstrup-kmp', 'linky', 'IEC62056-21', 'DSMR-2.3', 'DSMR-4.0'])
                .withDescription('Operating mode/probe'),
            exposes.numeric('current_summation', ea.SET)
                .withDescription('Current summation value sent to the display. e.g. 570 = 0,570 kWh').withValueMin(0)
                .withValueMax(268435455),
            exposes.binary('check_meter', ea.STATE, true, false)
                .withDescription('Is true if communication problem with meter is experienced'),
        ]
`
to this (just add e.battery())
` exposes: [
            e.power(),
            e.energy(),
            e.battery_low(),
            e.battery(),
            exposes.numeric('pulse_configuration', ea.ALL).withValueMin(0).withValueMax(65535)
                .withDescription('Pulses per kwh. Default 1000 imp/kWh. Range 0 to 65535'),
            exposes.enum('interface_mode', ea.ALL,
                ['electricity', 'gas', 'water', 'kamstrup-kmp', 'linky', 'IEC62056-21', 'DSMR-2.3', 'DSMR-4.0'])
                .withDescription('Operating mode/probe'),
            exposes.numeric('current_summation', ea.SET)
                .withDescription('Current summation value sent to the display. e.g. 570 = 0,570 kWh').withValueMin(0)
                .withValueMax(268435455),
            exposes.binary('check_meter', ea.STATE, true, false)
                .withDescription('Is true if communication problem with meter is experienced'),
        ]`

And this is the file (hopefully this time is right for everyone) 
[emizb_141.txt](https://github.com/Koenkk/zigbee2mqtt/files/14230438/emizb_141.txt)

Again just change "txt" to "js"

PaJaSoft, 2024-02-12:
@jllarraz Your last version is working fine however I still can't see a value of reported battery in percents. There is still NULL although the uninterrupted run of the whole system (-> Frient ZED, Coordinator and z2m) is 24h+ long for sure. I'm not sure if the issue is on Frient side (it didn't report this value yet) or in the implementation of messages handling.

jllarraz, 2024-02-12:
@PaJaSoft you should be able to see the logs of the communication and filter by "Frient"

in my case it looks like
`Info 2024-02-12 12:54:51MQTT publish: topic 'zigbee2mqtt/Energy Metter Frient', payload '{"battery":60,"battery_low":false,"check_meter":null,"current_summation":3176500,"energy":308.92,"interface_mode":"electricity","linkquality":255,"power":1715,"pulse_configuration":1000,"voltage":2600}'`

PaJaSoft, 2024-02-12:
@jllarraz Yea, I know how to search for this however I would rather looking how looks the LOG message when ZED reports a battery level to coordinator. I mean sth similar like for Energy:
`
debug 2024-02-12 14:17:12: Received Zigbee message from 'XXX', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,224482],"instantaneousDemand":804}' from endpoint 2 with groupID 0
`

penroseg, 2024-02-12:
@jllarraz Thats newest version is working perfectly. Many thanks for your help.  Am seeing battery percentage straight away and all the original fields are still there.  Much appreciated. !!!

nivek1612, 2024-02-12:
just switched to the newer convertor. Seems to work ok. But my battery is still reading 100% but the device is new so maybe that's accurate but I would have expected it to read around 98% or similar after 8 weeks. 

`{
    "battery": 100,
    "battery_low": false,
    "energy": 1462.53,
    "interface_mode": "electricity",
    "linkquality": 51,
    "power": 1553,
    "pulse_configuration": 1000,
    "voltage": 3100,
    "check_meter": null,
    "current_summation": null
}`

penroseg, 2024-02-12:
> just switched to the newer convertor. Seems to work ok. But my battery is still reading 100% but the device is new so maybe that's accurate but I would have expected it to read around 98% or similar [...]

Mine is showing 95% after 4+ months so I think it could be okay.  Frient do say 2 years but that's at room temperature so likely less with winter weather but at least it's an indication.  In general, I find most battery indicators on Zigbee devices are pretty rough and have seen them go from 60%+ to dead, but that's mainly on coin cells.  It's a function of how they calculate based on the voltage but with that being so low anyway a tiny change can be a huge % drop.  I think AA batteries are much better at that so I am hoping that 12-15 months may be achievable.

jeppester, 2024-02-13:
I set up the latest configuration posted and it works almost flawlessly (only issue is that the energy resets to 0 when I attempt to update it).

Thank you all for working this out!

As I'm an experienced programmer I might have a go at the resetting issue. I can't promise much though as this type of programming is not something I'm intimately familiar with.

nivek1612, 2024-02-14:
Why do you want to update the energy ? The value to me is irrelevant I just want to know how much it has changed per day per hour etc 

just curious

jeppester, 2024-02-19:
> Why do you want to update the energy ? The value to me is irrelevant I just want to know how much it has changed per day per hour etc
>
> just curious

One might want to sync it with the display on the device it gets connected to.

I'm not personally going to need that, I mostly just don't like like that there's an input field that doesn't work. If resetting is all we can do, then the control should be button with a "reset" label instead.

nivek1612, 2024-02-19:
> One might want to sync it with the display on the device it gets connected to.
>
> I'm not personally going to need that, I mostly just don't like like that there's an input field that doesn't work. If [...]

Fair although my meter has different rates so there are two readings so not much use being able to synch it for me.

Datortarp5, 2024-04-16:
Hello team.

With all external converter, latest coordinator firmware and latest Z2M version I still have issues with this power meter. Looks like all working except Energy KwH field.
Also what I discovered, all my powered zigbee devices fall offline if external converter is activated.

[image]

Bronti90, 2024-05-01:
Hi, sorry for picking up this old topic again. I purchased a EMIZB-141 a while ago and worked around the limited functionality myself, yesterday upgraded to lSh4dowl external converter and could fix wrong pulse config and not working energy variable by that - so first of all thanks for that help!

Nevertheless I am still facing the issue I initially had, after implementation of solar power plant, I do have negative power / supply to grid power in peak solar phases that are detected by EMIZB-141 as positive. So my energy dashboard looks weird and my overall electricity parameter in HomeAssistant needs to be corrected by supply grid.

Am I doing something wrong or is it simply not feasible to read negative values? I was reading about Tasmota IR readers which are capable to read supply power to grid and was assuming EMIZB-141 could do same. There is unfortunately no reasonable way to get WiFi to the electricity meter, so I am limited to Zigbee devices.

Apologies upfront, my coding skill are very basic to not existing. I tried to read and understand the code, but wasn't able to read the information I am looking for. Any help would be greatly appreciated.

---------------------------------------------------------------------------

My MQTT says following:
{
    "battery": 100,
    "battery_low": false,
    "energy": 878.12,
    "interface_mode": "electricity",
    "linkquality": 63,
    "power": 216,
    "pulse_configuration": 5000,
    "voltage": 3000,
    "check_meter": null,
    "current_summation": null
}

-------------------------------------------------------------------------------

I am using HomeAssistant as below with latest MQTT integration.
Core 2024.4.4
Supervisor 2024.04.4
Operating System 12.2
Frontend 20240404.2

mnaved37, 2024-05-23:
I got it working by changing the power sensor to energy sensor and then using that energy sensor to add to my energy grid. I used the helper - Integration - Riemann sum surgical - I tried both left and trapezoid - slight difference but not significant. I am now using the left integration sensor as my default. 
[image]
 Then add anime and then use the power sensor to covert it to energy sensor and kWh.
[image]

Now it is all working
[image]

How that helps

nilsbyte, 2024-05-26:
I wanted to add my experience after purchasing a EMIZB-141. 

Home Assistant Core 2024.5.5
Zigbee2MQTT 1.37.1-1

- Using the external converter
- Pulse set to 10.000 for my power meter

Results:
- Power in W is showing correctly, identical values to the power meters display
- Energy is not showing at all
- Battery Percentage not showing

Firmware for the devices is available at https://github.com/Develco-Products/frient_upgrade_images
The firmware file for the EMIZB-141 is [EMI2LED_3.1.2.zigbee](https://github.com/Develco-Products/frient_upgrade_images/blob/main/EMI2LED_3.1.2.zigbee). Maybe OTA is possible @lSh4dowl ? On the supported Hubs (Homey, Smart Things, Smart Home by Hornbach) it should be supported, stated by the frient website.

Bronti90, 2024-05-26:
> I got it working by changing the power sensor to energy sensor and then using that energy sensor to add to my energy grid. I used the helper - Integration - Riemann sum surgical - I tried both left and trapezoid - slight difference but not significant. I am now using the left integration sensor as my default. ![HA](https://private-user-images.githubusercontent.com/143832655/333143540-ab8fa4dc-ad0d-4686-8a20-6ddc6667a729.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY3Mjg2ODIsIm5iZiI6MTcxNjcyODM4MiwicGF0aCI6Ii8xNDM4MzI2NTUvMzMzMTQzNTQwLWFiOGZhNGRjLWFkMGQtNDY4Ni04YTIwLTZkZGM2NjY3YTcyOS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUyNlQxMjU5NDJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00YTRiYTc1OWQ5NzUxMDRmMGJhZDgwYTY4MDgxNzY0N2QxYzM5OTViZTM0M2FiZmRhOWQyZDU3ZDM1MmI5YjJiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.VmsmAbBnDjukqu38yE3oCFyhuH1XA8wGV-42DrzJo2Q) Then add anime and then use the power sensor to covert it to energy sensor and kWh. ![HA1](https://private-user-images.githubusercontent.com/143832655/333143759-738d4dfc-65db-499e-8c89-358b8035f41e.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY3Mjg2ODIsIm5iZiI6MTcxNjcyODM4MiwicGF0aCI6Ii8xNDM4MzI2NTUvMzMzMTQzNzU5LTczOGQ0ZGZjLTY1ZGItNDk5ZS04Yzg5LTM1OGI4MDM1ZjQxZS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUyNlQxMjU5NDJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iNDFkYTRiN2I1MDg1N2NiNWVlYjBkMDE0YTgwYTA3NDczMWI1MmEzOGI0NDU0ZTQ0YWE3YzgyNjcwMDdkN2RjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.8AAwEEhNMyPUvMZo-G2k2_YStDJx-roszTlofF5BepU)
> 
> Now it is all working ![HA2](https://private-user-images.githubusercontent.com/143832655/333144249-07cebc26-ccd8-4337-8fe8-6641427a75e3.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY3Mjg2ODIsIm5iZiI6MTcxNjcyODM4MiwicGF0aCI6Ii8xNDM4MzI2NTUvMzMzMTQ0MjQ5LTA3Y2ViYzI2LWNjZDgtNDMzNy04ZmU4LTY2NDE0MjdhNzVlMy5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUyNlQxMjU5NDJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wYjY0ZmU3NzRhZWMzZGVkZTZiODdkMjk1MTcxZWNjZmU2ZDBhNmRhMDRjZmU3MmY1ZjA5ZDYwNDFkNmUyYjQwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.9uP3cuxl8vejo419qfcmA5w4Gh2s8C-wrni96NESAS4)
> 
> How that helps

But from what I see you do not feed energy right? So this is still an issue for me. 
I have correct values for power and energy both available for the device, but the power value is always plus - when my solar panel is supplying more than I use, it shows a positive value instead of an negative and by that destroys my energy evaluation. 
I have an work around which works ok. I monitor power of main devices by a script and calculate my current power estimation, compare this to the power value of the sensor and my solar power feed and depending on the conditions offset the sensor value by *-1. This approach works with a accurency of +/- 0.1 kWh a day and is good enough for the moment.

[image]

ebbeknudsen, 2024-06-04:
> Apologies for that @penroseg I left a bit of code on that adapter that shouldnt be there.
> 
> this is the screenshoot of my Adapter <img alt="Screenshot 2024-02-10 at 20 30 32" width="1048" src="https://private-user-images.githubusercontent.com/4919039/303863562-ed4a3d99-98dd-4923-a67f-e2c0b6458699.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc1NDMzNDcsIm5iZiI6MTcxNzU0MzA0NywicGF0aCI6Ii80OTE5MDM5LzMwMzg2MzU2Mi1lZDRhM2Q5OS05OGRkLTQ5MjMtYTY3Zi1lMmMwYjY0NTg2OTkucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDYwNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA2MDRUMjMxNzI3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NTU3MmZjNGJiZTRlMjkwMWE1MWQxN2EyZGI2OTljZDM2NDQ2NmFhYThhYWJmOWQxZTgyNzhmZmJjMTdjN2RiZSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.qna2JxFjL8qX8NVQAA54WSzP88l8iTXIrtET0G2_7ho">
> 
> The modification of the original adapter (the first one on this thread) is as simple as changing this
> 
> `exposes: [ e.power(), e.energy(), e.battery_low(), exposes.numeric('pulse_configuration', ea.ALL).withValueMin(0).withValueMax(65535) .withDescription('Pulses per kwh. Default 1000 imp/kWh. Range 0 to 65535'), exposes.enum('interface_mode', ea.ALL, ['electricity', 'gas', 'water', 'kamstrup-kmp', 'linky', 'IEC62056-21', 'DSMR-2.3', 'DSMR-4.0']) .withDescription('Operating mode/probe'), exposes.numeric('current_summation', ea.SET) .withDescription('Current summation value sent to the display. e.g. 570 = 0,570 kWh').withValueMin(0) .withValueMax(268435455), exposes.binary('check_meter', ea.STATE, true, false) .withDescription('Is true if communication problem with meter is experienced'), ]` to this (just add e.battery()) ` exposes: [ e.power(), e.energy(), e.battery_low(), e.battery(), exposes.numeric('pulse_configuration', ea.ALL).withValueMin(0).withValueMax(65535) .withDescription('Pulses per kwh. Default 1000 imp/kWh. Range 0 to 65535'), exposes.enum('interface_mode', ea.ALL, ['electricity', 'gas', 'water', 'kamstrup-kmp', 'linky', 'IEC62056-21', 'DSMR-2.3', 'DSMR-4.0']) .withDescription('Operating mode/probe'), exposes.numeric('current_summation', ea.SET) .withDescription('Current summation value sent to the display. e.g. 570 = 0,570 kWh').withValueMin(0) .withValueMax(268435455), exposes.binary('check_meter', ea.STATE, true, false) .withDescription('Is true if communication problem with meter is experienced'), ]`
> 
> And this is the file (hopefully this time is right for everyone) [emizb_141.txt](https://github.com/Koenkk/zigbee2mqtt/files/14230438/emizb_141.txt)
> 
> Again just change "txt" to "js"

This latest external converter by @jllarraz seems to work great for me, thanks!

[image]

nilsbyte, 2024-06-05:
@jllarraz I can set the current summation but the energy value stays at "Null" kWh. I am thinking the energy value is calculated by the current power and not read by the sensor?

[51 more replies]

closed 2024-10-16 by Koenkk