Koenkk/zigbee2mqtt #19044: Reported illuminace is too low on ZY-M100-S_1 (_TZE204_sxm7l9xa)

seblu, 2023-09-21:
### What happened?

The `illuminance` reported by the presence sensor is too low and not reacting to ceiling light switching in the room. Looks like the value is percentage lower than illuminance reported by an Aqara Illuminance sensor and an Aqara Motion sensor.

This is happening on both Tuya wall and ceiling sensors version.

Here is a graph of the measured values on few days:
[image]

### What did you expect to happen?

Have the correct lux reported. Probably a 10 factor is missing in the reading.

### How to reproduce it (minimal and precise)

- Setup the sensor
- No light in the room 4lx. Ceiling light on 4lx.

### Zigbee2MQTT version

1.33.0 

### Adapter firmware version

20221226

### Adapter

Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001

lux73, 2023-09-22:
pls Keep in mind that all cheap Sensors sonetimes acting weird measuring Values

My Aqara Motion Sensor with integrated Lux measured crap  compared with my desired Aqara Lux Sensor ^^

Even my Room is completely Dark the Motion Sensor measure 18lx

So i dont give a shit 🙃

seblu, 2023-09-23:
Yes, I'm aware these sensors do not measure a real lux value, but more something which is proportional to the quantity of light the device is exposed to.

I have 5 different model of illuminance sensors, none is as bad as 18lx when no light. Sorry to hear that.

What is problematic here:
- the measured values is proportional but not precise enough to be used in the lower range
- the value min value is 4 instead of 0 or 1.
- the max value is 25lx when other sensors measure 175lx.

closed 2023-10-31 by github-actions[bot]