Koenkk/zigbee2mqtt #19046: TS110E 2 gang 1 physical switch not working after turning off from front-end or voice command

indrekdt, 2023-09-22:
### What happened?

Having issues with Tuya TS110E 2 gang dimmer, if i turn lights off from anywhere but physical switch, the physical switch stops working. If lights are turned on, everything works as it should, i can turn it off and on again.

### What did you expect to happen?

lights turn on from physical switch after turning off with voice command or frontend

### How to reproduce it (minimal and precise)

turn light off from front end or voice command

### Zigbee2MQTT version

1.33.0

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff Dongle E

closed 2023-10-30 by github-actions[bot]

lippertmarkus, 2024-01-13:
@indrekdt did you find a solution to the problem?

indrekdt, 2024-01-13:
Nope, can't recall how far i got with this when searching for a solution. I remember from another thread somewhere that the device actually has two separate states for light on/off, but one of them is not exposed in zigbee2mqtt. Dunno if this is the case tho.

For now i'm using a workaround, light on channel 1 and physical switch wired on channel two, since i'm using a motion sensor in that area i only occasionally need to turn the light on. So another automation is used, if channel 2 turned on - turn on channel one.

Problem was with an all off automation, i have a small button at the front door, so when i'm leaving home i press it once and it turns on a script to turn off all the lights, tv etc. Sometimes if i need to go get something that i forgot when leaving i want to press the physical switch for the light at the front door and it does nothing. State changes but brightness remains 0, although it wasn't 0 before i pressed the all off button.

FPSUsername, 2025-05-09:
Not bumping this for getting a fix, merely for anyone with similar issues.

I too have a 2 gang model, _TZ3210_3mpwqzuu, and it's simply unreliable.

What I noticed (after half a year of usage):
- L1 often reports an incorrect state. It randomly reports off when it's on
- refreshing the state of the brightness (in the exposes tab) often shows 4242 instead of anything between 0 and 254
- Light type is often randomly set to halogen. Not that it really does anything (I have Calex dimmable bulbs).
- with an automation, it often doesn't like turning on the light and immediately changing the brightness with the same command. 100ms need to be in between commands.
- with an automation which turns on L1 and L2 with 1 second delay between switching, it sometimes doesn't turn on L1.
- sometimes it becomes unresponsive and disconnects. The lights blink (pairing mode).

It's unreliable but really the only device in existence. Sonoff doesn't make any dimmers like these, let alone a 2 gang.