Koenkk/zigbee2mqtt #19047: UNSH SMKG-1KNL-EU-Z Smart Circuit Breaker threshold and breaker setup soensn't work

sdalee, 2023-09-22:
### What happened?

I use z2m as a HA plugin (Ha runs on a Virtual machine). 
I joined this circuit breaker to z2m. At first, everything worked.  
But now the threshold settings and breaker switches don't work.

### What did you expect to happen?

I'd like to use the threshold settings and breaker switches (as this would be the task of the device)

### How to reproduce it (minimal and precise)

I joined this circuit breaker to z2m. At first, everything worked.  
After that, I changed the current, voltage and power precision from 0 to 0.0 at  HA entity setup of circuit breaker. After that the current, voltage, power measurement data did not come at Z2m, exposes window. I restore the original precision of the entity, but these data didn't come back, and the current, voltage, temperature, power threshold settings, on and off current, voltage, temperature, power breaker switches didn't work too.  Red error message appeared in the z2m control window: 
if I tried to change the threshold settings: 'Publish set' 'temperature_treshold' to 'Prb_Unsh_Kismegszakito_mero' failed: 'Error expected one of: on, off, got: undefined'  ('Prb_Unsh_Kismegszakito_mero' is the device name in z2m and HA)
if I tried to change on - off switches of current, voltage, temperature, power, temperature breaker: 'Publish set' 'temperature_breaker' to 'Prb_Unsh_Kismegszakito_mero' failed: 'TypeError Cannot read properties of undefined ('reading to string')  ('Prb_Unsh_Kismegszakito_mero' is the device name in z2m and HA)
At the end I restored an earlier VM backup. The current, voltage, power measures now are working, but the threshold settings and breaker switches haven't worded yet.
The other switches, sensors data work fine.

### Zigbee2MQTT version

1.33.0-1

### Adapter firmware version

20220219

### Adapter

Electrolama zzh

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/12698713/log.txt)

sdalee, 2023-09-25:
Well, I don't understand. It has been off for 2 days and today I turn on and everything is working now. All thresholds and breakers.

closed 2023-09-25 by sdalee

kartone, 2024-10-18:
Just installed, same issue here.