Koenkk/zigbee2mqtt #19062: v1.33.0 won't initialize `/app/data` from a clean slate

Paraphraser, 2023-09-24:
### What happened?

This issue was reported as [IOTstack #727](https://github.com/SensorsIot/IOTstack/issues/727).

It looks to me like a regression crept in between 1.32.0 and 1.33.0.

In the tests below I demonstrate that:

1. `latest` (aka `1.33.0`) will not initialise its persistent store. It goes into a restart loop reporting:

	```
	Using '/app/data' as data directory
	Creating configuration file...
	cp: can't stat '/app/configuration.example.yaml': No such file or directory
	```
 
2. Reverting to `1.32.0` will initialise its persistent store.
3. Letting `1.32.0` initialise its persistent store, and then upgrading to `latest` succeeds.

## Reference service definition

from `docker-compose.yml`:

``` yaml
  zigbee2mqtt:
    container_name: zigbee2mqtt
    # image: koenkk/zigbee2mqtt:latest
    image: koenkk/zigbee2mqtt:1.32.2
    restart: unless-stopped
    environment:
      - TZ=${TZ:-Etc/UTC}
      - ZIGBEE2MQTT_CONFIG_MQTT_SERVER=mqtt://mosquitto:1883
      - ZIGBEE2MQTT_CONFIG_FRONTEND=true
      - ZIGBEE2MQTT_CONFIG_ADVANCED_LOG_SYMLINK_CURRENT=true
    ports:
      - "8080:8080"
    volumes:
      - ./volumes/zigbee2mqtt/data:/app/data
    devices:
      - "/dev/Zigbee_Sonoff_P:/dev/ttyACM0"
    depends_on:
      - mosquitto
```

## Test 1 - `1.32.2` from clean slate

Starting position is above service definition, container not running:

1. Erase persistent store.
2. Instantiate container.
3. Show working (log and structure of persistent store).
4. Terminate container.

[image]

## Test 2 - `latest` from clean slate

Starting position is service definition changed to `latest` tag, container not running:

1. Erase persistent store.
2. Instantiate container.
3. Show working (log and structure of persistent store).
4. Terminate container.

[image]

## Test 3 (workaround)

Starting position is `1.32.2` tag is active, container not running:

1. Erase persistent store.
2. Instantiate container (as `1.32.2`).
3. Terminate container.
4. Show initialized state of persistent store.
4. Instantiate container (as `latest`).
5. Show working.

[image]

### What did you expect to happen?

I expected `latest` to initialize its persistent store from a clean slate.

### How to reproduce it (minimal and precise)

1. Assuming `latest`.
2. Down the container.
3. Erase persistent store.
4. Up the container.

### Zigbee2MQTT version

1.33.0

### Adapter firmware version

20230507

### Adapter

Sonoff ZBDongle-P

### Debug log

Included in screen captures.

> By the way, the Open button at "IMPORTANT: Check development branch changelog first!!" leads nowhere. Were you aware of that?

Koenkk, 2023-09-24:
This has been fixed in https://github.com/Koenkk/zigbee2mqtt/pull/18836 which will be available in the next release and is already available in the dev branch. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-09-24 by Koenkk

Paraphraser, 2023-09-24:
Thanks. Perhaps this graphic will help to better communicate the point I tried to make more gently earlier: that it would be slightly easier for newbies on this repo to check the development branch change log if the button could be fixed.

[image]

Koenkk, 2023-09-24:
Fixed the link, thanks for letting me know!