Koenkk/zigbee2mqtt #19065: Power-on behavior does not work on _TZ3000_mmkbptmx

DChe47, 2023-09-24:
### What happened?

I have 4-gang switch _TZ3000_mmkbptmx but somehow does not work "Power-on behavior" for all channels.

### What did you expect to happen?

I need the "Power-on behavior" function

### How to reproduce it (minimal and precise)

An error happens every time I try to change the state for "Power-on behavior" on any of the channels:
`Error 2023-09-24 15:08:07Publish 'set' 'power_on_behavior' to 'Контроль входа' failed: 'Error: Write 0xa4c1387644e6e388/1 manuSpecificTuya_3({"powerOnBehavior":2}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Error 2023-09-24 15:08:09Publish 'set' 'power_on_behavior' to 'Контроль входа' failed: 'Error: Write 0xa4c1387644e6e388/2 manuSpecificTuya_3({"powerOnBehavior":2}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Error 2023-09-24 15:08:10Publish 'set' 'power_on_behavior' to 'Контроль входа' failed: 'Error: Write 0xa4c1387644e6e388/3 manuSpecificTuya_3({"powerOnBehavior":2}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Error 2023-09-24 15:08:11Publish 'set' 'power_on_behavior' to 'Контроль входа' failed: 'Error: Write 0xa4c1387644e6e388/4 manuSpecificTuya_3({"powerOnBehavior":2}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'`

The state of this devise is:
`{
    "last_seen": "2023-09-24T12:08:53.490Z",
    "linkquality": 3,
    "state_l1": "OFF",
    "state_l2": "OFF",
    "state_l3": "ON",
    "state_l4": "OFF",
    "device": {
        "applicationVersion": 72,
        "dateCode": "",
        "friendlyName": "Контроль входа",
        "hardwareVersion": 1,
        "ieeeAddr": "0xa4c1387644e6e388",
        "manufacturerID": 4417,
        "manufacturerName": "_TZ3000_mmkbptmx",
        "model": "TS0004",
        "networkAddress": 19677,
        "powerSource": "Mains (single phase)",
        "stackVersion": 0,
        "type": "Router",
        "zclVersion": 3
    },
    "power_on_behavior_l1": null,
    "power_on_behavior_l2": null,
    "power_on_behavior_l3": null,
    "power_on_behavior_l4": null,
    "action": ""
}`

### Zigbee2MQTT version

1.33.0-1

### Adapter firmware version

Supervisor 2023.09.2

### Adapter

HamGeek USB

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/12708499/log.txt)

DChe47, 2023-10-03:
It's still relevant to me

bednim, 2023-10-22:
same here - @DChe47 have you got it fixed anyhow?

Koenkk, 2023-10-22:
Given the error, looks like this version dies not support it.

segaura, 2024-05-31:
> _TZ3000_mmkbptmx it is 100%, compatible with _TZ3000_ltt60asa Test it by myself. Power on behavior work as well.

This will probably also solve #20973 , the issue being the missing _Switch type_ configuration, which is indeed present in _TZ3000_ltt60asa.

Not completely sure what I am doing, but I tried to send the PR to the zigbee-herdsman-converters codebase

https://github.com/segaura/zigbee-herdsman-converters/pull/1

closed 2025-01-04 by github-actions[bot]

RealSpyKitty, 2026-10-16:
this issue stil persists in _TZ3000_pfc7i3kt