Koenkk/zigbee2mqtt #19066: [Wrong device]: Moes ZP-LZ-FR2U detected as Nue / 3A HGZB-01

duvalale, 2023-09-24:
### Link

https://www.zigbee2mqtt.io/devices/ZP-LZ-FR2U.html

### Model

https://www.zigbee2mqtt.io/devices/ZP-LZ-FR2U.html

### Description

Zigbee 3.0 dual USB wireless socket plug

### Vendor

Moes

### Picture (link)

https://www.zigbee2mqtt.io/images/devices/ZP-LZ-FR2U.jpg

### Database entry

{"id":24,"type":"Router","ieeeAddr":"0x086bd7fffe3f43ff","nwkAddr":115,"manufId":4478,"manufName":"3A Smart Home DE","powerSource":"Mains (single phase)","modelId":"LXN59-1S7LX1.0","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"hwVersion":0,"zclVersion":3}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0021b77fed","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}}},"appVersion":1,"stackVersion":0,"hwVersion":0,"dateCode":"20200519","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1695568195179,"defaultSendRequestWhen":"immediate"}

### Notes

My config is : 
Coordinator zzh! - zStack3x0
Firmware 20221226
Z2M version : 1.33

I tried force removing and re-pairing the device with no success, it keeps showing as Nue / 3A [HGZB-01](https://www.zigbee2mqtt.io/devices/HGZB-01.html#nue%2520%252F%25203a-hgzb-01).

Koenkk, 2023-09-24:
This is a whitelabel device which shares the same model ID (e.g. https://github.com/Koenkk/zigbee2mqtt/issues/4577), so unfortunately we cannot fix this (but note it is just a cosmetic issue)

duvalale, 2023-09-25:
Thanks for your answer.
If I understand correctly, there is no way to have the two endpoints work independently as stated on the [product's page](https://www.zigbee2mqtt.io/devices/ZP-LZ-FR2U.html) ?

Koenkk, 2023-09-25:
> there is no way to have the two endpoints work independently as stated on the [product's page](https://www.zigbee2mqtt.io/devices/ZP-LZ-FR2U.html) ?

This device has only 1 endpoint: `epList":[1]`

duvalale, 2023-09-26:
Ok, thanks for your answer and my bad :)

closed 2023-09-26 by duvalale