Koenkk/zigbee2mqtt #19070: Aqara Smart Radiator Thermostat E1: auto/manual mode mixed up

Ra72xx, 2023-09-25:
### What happened?

The device mentioned in the subject (firmware 0.0.0_0825, SRTS-A01, https://www.zigbee2mqtt.io/devices/SRTS-A01.html#xiaomi-srts-a01) mixes up the presets "automatic" and "manual" both in Homeassistant and in the Zigbee2MQTT web ui. 
If set to manual, the device is in automatic (internal schedule) mode and vice versa. This is the web ui when the device in fact works according to a schedule (the device display also shows a clock icon to signal this preset):

[image]

### What did you expect to happen?

Setting the thermostat to "auto": Heat according to internal schedule.
Setting the thermostat to "manual": Use manually fixed temperature.

### How to reproduce it (minimal and precise)

Install the thermostat, pair it with Z2M and use it.

### Zigbee2MQTT version

1.33.0 commit: 7ee207f

### Adapter firmware version

7.1.1.0 build 273

### Adapter

SkyConnect (EZSP v9)

Ra72xx, 2023-10-19:
I guess I misunderstood the meaning of "auto" for this thermostat -> invalid.

groenmarsmannetje, 2023-10-20:
I have the same expectation. Setting the mode to auto should enable the internal scheduler. Or there should be a different mode (eg "schedule") that you can select. That is how it works in my other Tuya Zigbee TRV's

Ra72xx, 2023-10-20:
> @Ra72xx so why did you close this ticket? I think you made a valid point.

I assume that it is not a Z2M problem, but a different logic this device is implementing:

- Switch "schedule": Turns on the (very rudimentary) scheduling the device provides
- Mode "automatic": Does something else - maybe using the (with the Aqara app) integrated presence- and window-open sensors, but that I don't know.

I assume both these settings are not related, as both "schedule" and "automatic mode" have their own, not mutually exclusive, symbols in the thermostat display. This wouldn't make sense if "automatic mode" would mean scheduling the heating.

groenmarsmannetje, 2023-10-20:
Actually in this comment https://github.com/Koenkk/zigbee-herdsman-converters/pull/5058#issuecomment-1334236460 the developer was asking for users feedback to improve this. So re-open this ticket and let this be the feedback that he was looking for.

Ra72xx, 2023-10-20:
OK, I'll reopen the ticket. However, I assume that it is rather a documentation issue to avoid misunderstanding this thermostat's operating modes.

groenmarsmannetje, 2023-10-20:
Just for reference. On my other Tuya TRV's I have the following presets (among others):

Schedule - This activates the internal schedular of the device.
Manual - The manually set temperature permanently overrules the internal schedular of the device.
Complex - The manually set temperature only applies until the next set point in time of the internal schedular is reached. After that the internal schedular is used.

closed 2024-05-18 by github-actions[bot]