Koenkk/zigbee2mqtt #19072: Aqara SRTS-A01 Error: Exception while calling fromZigbee converter: Expected one of: 0, 1, got: '2'}

DaN66O, 2023-09-25:
### What happened?

Aqara SRTS-A01 does not respond to an external temperature sensor

### What did you expect to happen?

Temperature would be updated (local_temperature)

### Zigbee2MQTT version

1.33.0 commit: 7ee207f

### Adapter firmware version

20230507

### Adapter

zigstar

### Debug log

`Error: Exception while calling fromZigbee converter: Expected one of: 0, 1, got: '2'}`

Koenkk, 2023-09-25:
Can you provide the debug log of this?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

DaN66O, 2023-09-26:
> Can you provide the debug log of this?
> 
> See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

I turned on log_level: debug

I have zigbe2mqtt running as a systemd service.
How do I enable Zigbee-herdsman debug logging in it?

Koenkk, 2023-09-26:
thats enough, herdsman debug log is not needed

DaN66O, 2023-09-26:
> thats enough, herdsman debug log is not needed

```
info  27-09-2023 00:28:05: MQTT publish: topic 'zigbee2mqtt2/radiator_bedroom', payload '{"away_preset_temperature":"15.0","battery":53,"calibrated":true,"child_lock":"LOCK","device_temperature":26,"internal_heating_setpoint":23.5,"last_seen":"2023-09-27T00:28:05+03:00","linkquality":131,"local_temperature":23.2,"occupied_heating_setpoint":23.5,"power_outage_count":0,"preset":"manual","schedule":"OFF","schedule_settings":"mon,sun|0:00,21.0|12:00,17.0|18:00,21.0|23:59,21.0","sensor":"external","setup":false,"system_mode":"heat","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"valve_alarm":false,"valve_detection":"OFF","voltage":3200,"window_detection":"OFF","window_open":false}'
debug 27-09-2023 00:28:05: Received Zigbee message from 'radiator_bedroom', type 'attributeReport', cluster 'aqaraOpple', data '{"638":1,"640":1}' from endpoint 1 with groupID 0
debug 27-09-2023 00:28:05: zigbee-herdsman-converters:aqara_trv: Unhandled key 640 = 1
info  27-09-2023 00:28:05: MQTT publish: topic 'zigbee2mqtt2/radiator_bedroom', payload '{"away_preset_temperature":"15.0","battery":53,"calibrated":true,"child_lock":"LOCK","device_temperature":26,"internal_heating_setpoint":23.5,"last_seen":"2023-09-27T00:28:05+03:00","linkquality":134,"local_temperature":23.2,"occupied_heating_setpoint":23.5,"power_outage_count":0,"preset":"manual","schedule":"OFF","schedule_settings":"mon,sun|0:00,21.0|12:00,17.0|18:00,21.0|23:59,21.0","sensor":"external","setup":false,"system_mode":"heat","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"valve_alarm":false,"valve_detection":"OFF","voltage":3200,"window_detection":"OFF","window_open":false}'
debug 27-09-2023 00:28:05: Received Zigbee message from 'radiator_bedroom', type 'attributeReport', cluster 'aqaraOpple', data '{"65522":{"data":[170,113,13,36,18,162,5,65,10,0,0,21,141,0,1,157,27,152,1],"type":"Buffer"}}' from endpoint 1 with groupID 0
debug 27-09-2023 00:28:05: zigbee-herdsman-converters:aqara_trv: Unhandled key 65522 = �q
$�A
info  27-09-2023 00:28:05: MQTT publish: topic 'zigbee2mqtt2/radiator_bedroom', payload '{"away_preset_temperature":"15.0","battery":53,"calibrated":true,"child_lock":"LOCK","device_temperature":26,"internal_heating_setpoint":23.5,"last_seen":"2023-09-27T00:28:05+03:00","linkquality":134,"local_temperature":23.2,"occupied_heating_setpoint":23.5,"power_outage_count":0,"preset":"manual","schedule":"OFF","schedule_settings":"mon,sun|0:00,21.0|12:00,17.0|18:00,21.0|23:59,21.0","sensor":"external","setup":false,"system_mode":"heat","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"valve_alarm":false,"valve_detection":"OFF","voltage":3200,"window_detection":"OFF","window_open":false}'
info  27-09-2023 00:28:05: MQTT publish: topic 'zigbee2mqtt2/radiator_children_right', payload '{"away_preset_temperature":"15.0","battery":79,"calibrated":true,"child_lock":"LOCK","device_temperature":25,"internal_heating_setpoint":23.5,"last_seen":"2023-09-27T00:28:05+03:00","linkquality":7,"local_temperature":24.2,"occupied_heating_setpoint":23.5,"power_outage_count":11,"preset":"manual","schedule":"OFF","schedule_settings":"mon,sun|0:00,21.0|12:00,17.0|18:00,21.0|23:59,21.0","sensor":"external","setup":false,"system_mode":"heat","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"valve_alarm":false,"valve_detection":"OFF","voltage":3300,"window_detection":"OFF","window_open":false}'
debug 27-09-2023 00:28:05: Received Zigbee message from 'radiator_children_right', type 'attributeReport', cluster 'aqaraOpple', data '{"638":2,"640":0}' from endpoint 1 with groupID 0
error 27-09-2023 00:28:05: Exception while calling fromZigbee converter: Expected one of: 0, 1, got: '2'}
debug 27-09-2023 00:28:05: Error: Expected one of: 0, 1, got: '2'
    at Object.getFromLookup (/opt/zigbee2mqtt2/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:526:37)
    at /opt/zigbee2mqtt2/node_modules/zigbee-herdsman-converters/src/devices/xiaomi.ts:111:46
    at Array.forEach (<anonymous>)
    at Object.convert (/opt/zigbee2mqtt2/node_modules/zigbee-herdsman-converters/src/devices/xiaomi.ts:85:38)
    at Receive.onDeviceMessage (/opt/zigbee2mqtt2/lib/extension/receive.ts:150:51)
    at EventEmitter.emit (node:events:526:35)
    at EventBus.emitDeviceMessage (/opt/zigbee2mqtt2/lib/eventBus.ts:102:22)
    at Controller.<anonymous> (/opt/zigbee2mqtt2/lib/zigbee.ts:108:27)
    at Controller.emit (node:events:514:28)
    at Controller.selfAndDeviceEmit (/opt/zigbee2mqtt2/node_modules/zigbee-herdsman/src/controller/controller.ts:527:14)
info  27-09-2023 00:28:05: MQTT publish: topic 'zigbee2mqtt2/radiator_children_right', payload '{"away_preset_temperature":"15.0","battery":79,"calibrated":true,"child_lock":"LOCK","device_temperature":25,"internal_heating_setpoint":23.5,"last_seen":"2023-09-27T00:28:05+03:00","linkquality":7,"local_temperature":24.2,"occupied_heating_setpoint":23.5,"power_outage_count":11,"preset":"manual","schedule":"OFF","schedule_settings":"mon,sun|0:00,21.0|12:00,17.0|18:00,21.0|23:59,21.0","sensor":"external","setup":false,"system_mode":"heat","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"valve_alarm":false,"valve_detection":"OFF","voltage":3300,"window_detection":"OFF","window_open":false}'
[... cut ...] ":"idle"},"update_available":null,"valve_alarm":false,"valve_detection":"OFF","voltage":3300,"window_detection":"OFF","window_open":false}'
debug 27-09-2023 00:28:25: Received Zigbee message from 'radiator_children_left', type 'attributeReport', cluster 'aqaraOpple', data '{"65522":{"data":[170,113,13,36,18,162,5,65,10,0,0,21,141,0,1,157,27,152,1],"type":"Buffer"}}' from endpoint 1 with groupID 0
debug 27-09-2023 00:28:25: zigbee-herdsman-converters:aqara_trv: Unhandled key 65522 = �q
$�A

```

Koenkk, 2023-09-27:
This point maps to the `sensor`, 0 = `internal`, 1 = `external` but the device sends 2, what kind of sensor is used in your case/

DaN66O, 2023-09-27:
> This point maps to the `sensor`, 0 = `internal`, 1 = `external` but the device sends 2, what kind of sensor is used in your case?

I use an external temperature sensor Xiaomi WSDCGQ11LM
I use it as described here :
https://github.com/Koenkk/zigbee2mqtt/discussions/14291#discussioncomment-4067310

Koenkk, 2023-09-27:
I think there is some misconfiguration then, but this exception is not the reason why the external sensor doesn't control the thermostat (you should try to fix that first)

DaN66O, 2023-09-27:
> I think there is some misconfiguration then, but this exception is not the reason why the external sensor doesn't control the thermostat (you should try to fix that first)

I don't understand what's wrong here and what I can fix
```
service: mqtt.publish
data:
  topic: zigbee2mqtt2/radiator_children_right/set/sensor_temp
  payload_template: "{{ (states('sensor.dt_children_temperature')|float(0)|round(1)) }}"service: mqtt.publish
data:
  topic: zigbee2mqtt2/radiator_children_right/set/sensor_temp
  payload_template: "{{ (states('sensor.dt_children_temperature')|float(0)|round(1)) }}"
```
And where do I fix it ?

Koenkk, 2023-09-27:
You can check the z2m debug logging to confirm whenever temp changes, z2m receives a messages on `zigbee2mqtt2/radiator_children_right/set/sensor_temp` with the correct temperature as the payload. As I don't have this device myself, for further help I suggest to ask here: https://github.com/Koenkk/zigbee2mqtt/discussions/14291#discussioncomment-4067310

DaN66O, 2023-09-27:
Got it, thanks. 
I'll look at the logs again

closed 2023-11-04 by github-actions[bot]