Koenkk/zigbee2mqtt #19073: [New device support]: LCU001 (Hue White and Color Ambiance E14)

ckref, 2023-09-25:
### Link

https://www.philips-hue.com/da-dk/p/hue-white-and-color-ambiance-krone---e14-pare---1-pak/8719514491229#overview

### Database entry

{"id":40,"type":"Router","ieeeAddr":"0x001788010d854970","nwkAddr":29989,"manufId":4107,"manufName":"Signify Netherlands B.V.","powerSource":"Mains (single phase)","modelId":"LCU001","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":269,"inClusterList":[0,3,4,5,6,8,4096,64515,768,64513],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"LCU001","manufacturerName":"Signify Netherlands B.V.","powerSource":1,"zclVersion":8,"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20230116","swBuildId":"1.101.10"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20230116","swBuildId":"1.101.10","zclVersion":8,"interviewCompleted":true,"meta":{},"lastSeen":1695630774902,"defaultSendRequestWhen":"immediate"}

### Comments

I tried to add this devise by extending the exiting philips light_onoff_brightness_colortemp_color using the following external converter.

zigbee2mqtt fails to start, because it does not recognice the philips extension.
I have downloaded the philips.js to the zigbee2mqtt root folder in Home Assistant.
What am I missing?

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['LCU001'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'LCU001', // Vendor model number, look on the device for a model number
    vendor: 'Signify Netherlands B.V.', // Vendor of the device (only used for documentation and startup logging)
    description: 'Hue White and Color Ambiance E14', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    extend: philips.extend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 500]}),
    endpoint: (device) => {
        return {'default': 11};
    },
};

module.exports = definition;
```

### Supported color modes

color temperature, color (rgb)

### Color temperature range

Assuming rage of 153-500

Koenkk, 2023-09-25:
Try adding `const philips = require('zigbee-herdsman-converters/lib/philips');` at the top of the file.

ckref, 2023-09-25:
Great, the add-on starts, however I'm getting the following error:
Zigbee2MQTT:error 2023-09-25 20:50:12: Failed to call 'ExternalExtension' 'start' (/app/data/extension/philips.js:1
import {ColorXY, ColorRGB} from './color';
^^^^^^
SyntaxError: Cannot use import statement outside a module
    at new Script (node:vm:100:7)
    at createScript (node:vm:265:10)
    at Object.runInNewContext (node:vm:306:10)
    at Object.loadModuleFromText (/app/lib/util/utils.ts:152:8)
    at /app/lib/extension/externalExtension.ts:98:42
    at Array.map (<anonymous>)
    at ExternalExtension.loadUserDefinedExtensions (/app/lib/extension/externalExtension.ts:98:14)
    at ExternalExtension.start (/app/lib/extension/externalExtension.ts:19:14)
    at Controller.callExtensions (/app/lib/controller.ts:319:40)
    at Controller.start (/app/lib/controller.ts:153:9))

ckref, 2023-09-25:
WIth the philips extention everything seems to work, eventough I get the error.
Using extend: extend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 500]}), I'm also able to control the lights as expected. 
Thank you very much @Koenkk

Koenkk, 2023-09-25:
@ckref added the device, you can remove the ext converter after updating to the next release.

closed 2023-09-25 by Koenkk