Koenkk/zigbee2mqtt #19076: icons NOT local saved?!?

HalliHalloSchatz, 2023-09-25:
### What happened?

Device ressource slsys.github.io down?!?

Why can not serve device pictures from own local server?
Can I change this in the settings and download and serve it self?

### What did you expect to happen?

device picture served by zigbee2mqtt self

### Zigbee2MQTT version

1.32.2

### Adapter firmware version

20230507

### Adapter

SONOFF

Hypfer, 2023-09-27:
This is not a bug. It's also not a feature request. It's nothing. Just raw emotions dumped into a form.

Also, duplicate of this issue closed by the stale bot which only needs to exist because of low-quality issues like these:
https://github.com/Koenkk/zigbee2mqtt/issues/11908

I get what you want to say though.
Home Assistant actually has the same thing going on with their brands icons and in that case, they're doing it like that, because the actual rendered brands assets are currently taking up 2.6 GB. You can't throw all those assets into the docker image. 

Therefore, this is actually a reasonable default. What you can do though is provide a way for advanced users to specify their own host/backend for those icons

For that, there's this issue in the frontend repo https://github.com/nurikk/zigbee2mqtt-frontend/issues/1477

WHERE YOU ALREADY COMMENTED. Fucking hell. I just noticed that while writing this response.
Could you at least put in the absolute minimum effort of linking the damn thing here?

Koenkk, 2023-09-27:
This is a duplicate indeed.

closed 2023-09-27 by Koenkk