Koenkk/zigbee2mqtt #19077: Neo NAS-AB06B2 not supported

chami8i, 2023-09-25:
### What happened?

Although a converter exists in the neo.ts database for  TS0601 _TZE200_nlrfgpny, my device is not supported. I tested it with Tuya and verify datapoints on tuyaIo platform.

### What did you expect to happen?

Help. I tried to clone herdsman converter as a new device. Error on e.enum not a function.

### Zigbee2MQTT version

1.30.4

### Adapter firmware version

na

### Adapter

 EZSP(EFR32/MGM)

### Debug log

[debugAB06B2.txt](https://github.com/Koenkk/zigbee2mqtt/files/12716290/debugAB06B2.txt)

chami8i, 2023-09-25:
Solved with Z2M v1.33.0

closed 2023-09-25 by chami8i

averter, 2024-09-29:
Hi,
I have the same problem and am on a newer version (below). Can you please 1) let me know how to produce a debug log to diagnose this, and 2) reopen the issue? THanks.

Zigbee2MQTT version
    [1.40.1](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.40.1) commit: [403d3c09](https://github.com/Koenkk/zigbee2mqtt/commit/403d3c09)

Coordinator type
    zStack3x0

Coordinator revision
    20230507

Coordinator IEEE Address
    0x00124b002c3a6118

Frontend version
    0.7.4

zigbee-herdsman-converters version
    20.12.1

zigbee-herdsman version
    0.57.3

peroanjo, 2024-10-03:
Hi @averter, I think we have a similar issue. I opened #24149. Could you check if the manufacturer id is the one I mentioned in that issue? I bought the solar siren a couple weeks ago.