Koenkk/zigbee2mqtt #19078: [device dont work correct]: LED-Trading 9133 Friends of Hue Module dont work correctly

Oggymator, 2023-09-25:
### Link

https://www.zigbee2mqtt.io/devices/9133.html

### Database entry

{"id":3,"type":"GreenPower","ieeeAddr":"0x000000000171ddb2","nwkAddr":56754,"manufId":null,"modelId":"GreenPower_2","epList":[242],"endpoints":{"242":{"epId":242,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":1695655658118,"defaultSendRequestWhen":"immediate"}

### Comments

Hi,

the product has the same functions like a Friends of Hue Module from Sunricher or Vimar, so like EnOcean PTM
https://www.zigbee2mqtt.io/devices/PTM_215Z.html

In Z2M it was found as a PTM 215Z, that is correct, but in Home Assistant I can only use in Automations ""release_4" action".
All other functions are missing.

Can anybody figuered out what could be the problem?

thanks and kind regards
Tobias

jophut, 2023-12-13:
I've figured out that the labeling on the case differs from the reported payload "action":
K1: gives "press_3" and "release_3"
K2: gives "press_4" and "release_4"
K3: gives "press_1" and "release_1"
K4: gives "press_2" and "release_2"

Thus it seems that inputs 1 is interchanged with 3 and 4 is interchanged with 2
Also simultaneously K1+K3 gives (correctly) "press_1_and_3" and "release_1_and_3".
and K2+K4 gives (correctly) "press_2_and_4" and "release_2_and_4" 

Unfortunately I've never received a long press or hold action...

`MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{
"definition":{"description":"Pushbutton transmitter module",
"exposes":[{"access":1,"description":"Triggered action (e.g. a button click)",
"label":"Action",
"name":"action",
"property":"action","type":"enum",
"values":["press_1","release_1","press_2","release_2","press_3","release_3","press_4","release_4",
"press_1_and_3","release_1_and_3","press_2_and_4","release_2_and_4","press_energy_bar"]},
{"access":1,"description":"Link quality (signal strength)",
"label":"Linkquality","name":"linkquality",
"property":"linkquality",
"type":"numeric","unit":"lqi","value_max":255,
"value_min":0}],
"model":"PTM 215Z","options":[],"supports_ota":false,"vendor":"EnOcean"},
"friendly_name":"0x00000000017184da",
"ieee_address":"0x00000000017184da","status":"successful","supported":true},
"type":"device_interview"}'`

The "action": "press_energy_bar" is also never received.

cromefire, 2024-02-03:
Can you elaborate on what you did to get it working? For me it never receives any action at all and the zig we clusters are just completely empty. It isn't even able to wether it's on wall power or not. The log is also not showing a single event when I press it.

Oggymator, 2024-02-04:
> I've figured out that the labeling on the case differs from the reported payload "action": K1: gives "press_3" and "release_3" K2: gives "press_4" and "release_4" K3: gives "press_1" and "release_1" K4: gives "press_2" and "release_2"

But if the labeling was not correct, why does it work with the Philips Hue Bridge without any problem.
If the labeling was not correct, the pairing with the Hue would not work...

lanwin, 2024-06-25:
I created a pullrequest to add this info to the docs https://github.com/Koenkk/zigbee2mqtt.io/pull/2844

Thank @jophut for figuring it out.

philippnbg, 2024-10-15:
Actually I had success pairing the pushbutton module by pressing K1 (for channel 11) for >10 seconds, wait 3 seconds for z2m to recognise the pairing request and then confirming the pairing request by again pushing K1 for >10 seconds.
Very handy if you use the modules in a DIN rail encasement and hard wired setup with single buttons in multiple rooms. This allowed me to do it on my own :)

A video of the process can be seen here (https://youtu.be/j8RFxQjE2Qc?t=140&si=lADK6xsyk7-XXAms , german)

closed 2025-01-12 by github-actions[bot]

Mojtra, 2025-09-09:
> Actually I had success pairing the pushbutton module by pressing K1 (for channel 11) for >10 seconds, wait 3 seconds for z2m to recognise the pairing request and then confirming the pairing request [...]

Thank you!
Press K1 > 10 sec. did not work for me. But the hint to press K1 short multiple times after the 10 sec. to change the channels worked perfect for me. I've paired 5 of these ***** with this method and it only took 4 hours...