Koenkk/zigbee2mqtt #19087: EMIZB-132: Missing unit_of_measurement for voltage_phase_c

svhelge, 2023-09-26:
### What happened?

I found that voltage_phase_c has no unit (missing V). 
[image]

### What did you expect to happen?

I expected that voltage_phase_c had the correct unit (it used to have, but I am not sure when the error was introduced).

### Zigbee2MQTT version

1.33.0

### Adapter firmware version

0x26780700

### Adapter

Conbee2

### Debug log

Info 2023-09-26 18:13:37MQTT publish: topic 'zigbee2mqtt/Strømmåler', payload '{"current":2.8,"current_firmware":"4.0.1","current_phase_b":6553.5,"current_phase_c":6.6,"energy":149655.27,"interface_mode":"norwegian_han","linkquality":255,"power":1634,"update":{"installed_version":262145,"latest_version":262145,"state":"idle"},"update_available":false,"voltage":238.6,"voltage_phase_b":240.5,"voltage_phase_c":238.5}'
Debug 2023-09-26 18:13:39Received Zigbee message from 'Strømmåler', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":32,"rmsCurrentPhB":65535,"rmsCurrentPhC":51,"rmsVoltage":2384,"rmsVoltagePhB":2403,"rmsVoltagePhC":2388}' from endpoint 2 with groupID null
Info 2023-09-26 18:13:39MQTT publish: topic 'zigbee2mqtt/Strømmåler', payload '{"current":3.2,"current_firmware":"4.0.1","current_phase_b":6553.5,"current_phase_c":5.1,"energy":149655.27,"interface_mode":"norwegian_han","linkquality":255,"power":1634,"update":{"installed_version":262145,"latest_version":262145,"state":"idle"},"update_available":false,"voltage":238.4,"voltage_phase_b":240.3,"voltage_phase_c":238.8}'
Debug 2023-09-26 18:13:42Received Zigbee message from 'Strømmåler', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,149655270],"currentSummReceived":[0,0],"instantaneousDemand":1639,"status":0}' from endpoint 2 with groupID null
Info 2023-09-26 18:13:42MQTT publish: topic 'zigbee2mqtt/Strømmåler', payload '{"current":3.2,"current_firmware":"4.0.1","current_phase_b":6553.5,"current_phase_c":5.1,"energy":149655.27,"interface_mode":"norwegian_han","linkquality":255,"power":1639,"update":{"installed_version":262145,"latest_version":262145,"state":"idle"},"update_available":false,"voltage":238.4,"voltage_phase_b":240.3,"voltage_phase_c":238.8}'
Debug 2023-09-26 18:13:47Received Zigbee message from 'Strømmåler', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,149655270],"currentSummReceived":[0,0],"instantaneousDemand":1525,"status":0}' from endpoint 2 with groupID null
Info 2023-09-26 18:13:47MQTT publish: topic 'zigbee2mqtt/Strømmåler', payload '{"current":3.2,"current_firmware":"4.0.1","current_phase_b":6553.5,"current_phase_c":5.1,"energy":149655.27,"interface_mode":"norwegian_han","linkquality":255,"power":1525,"update":{"installed_version":262145,"latest_version":262145,"state":"idle"},"update_available":false,"voltage":238.4,"voltage_phase_b":240.3,"voltage_phase_c":238.8}'
Debug 2023-09-26 18:13:49Received Zigbee message from 'Strømmåler', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":32,"rmsCurrentPhB":65535,"rmsCurrentPhC":51,"rmsVoltage":2388,"rmsVoltagePhB":2404,"rmsVoltagePhC":2391}' from endpoint 2 with groupID null
Info 2023-09-26 18:13:49MQTT publish: topic 'zigbee2mqtt/Strømmåler', payload '{"current":3.2,"current_firmware":"4.0.1","current_phase_b":6553.5,"current_phase_c":5.1,"energy":149655.27,"interface_mode":"norwegian_han","linkquality":255,"power":1525,"update":{"installed_version":262145,"latest_version":262145,"state":"idle"},"update_available":false,"voltage":238.8,"voltage_phase_b":240.4,"voltage_phase_c":239.1}'
Debug 2023-09-26 18:13:52Received Zigbee message from 'Strømmåler', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,149655270],"currentSummReceived":[0,0],"instantaneousDemand":1979,"status":0}' from endpoint 2 with groupID null
Info 2023-09-26 18:13:52MQTT publish: topic 'zigbee2mqtt/Strømmåler', payload '{"current":3.2,"current_firmware":"4.0.1","current_phase_b":6553.5,"current_phase_c":5.1,"energy":149655.27,"interface_mode":"norwegian_han","linkquality":255,"power":1979,"update":{"installed_version":262145,"latest_version":262145,"state":"idle"},"update_available":false,"voltage":238.8,"voltage_phase_b":240.4,"voltage_phase_c":239.1}'
Debug 2023-09-26 18:13:57Received Zigbee message from 'Strømmåler', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,149655270],"currentSummReceived":[0,0],"instantaneousDemand":2054,"status":0}' from endpoint 2 with groupID null
Info 2023-09-26 18:13:57MQTT publish: topic 'zigbee2mqtt/Strømmåler', payload '{"current":3.2,"current_firmware":"4.0.1","current_phase_b":6553.5,"current_phase_c":5.1,"energy":149655.27,"interface_mode":"norwegian_han","linkquality":255,"power":2054,"update":{"installed_version":262145,"latest_version":262145,"state":"idle"},"update_available":false,"voltage":238.8,"voltage_phase_b":240.4,"voltage_phase_c":239.1}'
Debug 2023-09-26 18:13:59Received Zigbee message from 'Strømmåler', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":36,"rmsCurrentPhB":65535,"rmsCurrentPhC":58,"rmsVoltage":2386,"rmsVoltagePhB":2403,"rmsVoltagePhC":2389}' from endpoint 2 with groupID null
Info 2023-09-26 18:13:59MQTT publish: topic 'zigbee2mqtt/Strømmåler', payload '{"current":3.6,"current_firmware":"4.0.1","current_phase_b":6553.5,"current_phase_c":5.8,"energy":149655.27,"interface_mode":"norwegian_han","linkquality":255,"power":2054,"update":{"installed_version":262145,"latest_version":262145,"state":"idle"},"update_available":false,"voltage":238.6,"voltage_phase_b":240.3,"voltage_phase_c":238.9}'
Debug 2023-09-26 18:14:03Received Zigbee message from 'Strømmåler', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,149655270],"currentSummReceived":[0,0],"instantaneousDemand":2004,"status":0}' from endpoint 2 with groupID null
Info 2023-09-26 18:14:03MQTT publish: topic 'zigbee2mqtt/Strømmåler', payload '{"current":3.6,"current_firmware":"4.0.1","current_phase_b":6553.5,"current_phase_c":5.8,"energy":149655.27,"interface_mode":"norwegian_han","linkquality":255,"power":2004,"update":{"installed_version":262145,"latest_version":262145,"state":"idle"},"update_available":false,"voltage":238.6,"voltage_phase_b":240.3,"voltage_phase_c":238.9}'

svhelge, 2023-09-26:
I had a look in the log after restart, I find unit_of_measurement for phase b, but not for phase c:

`info  2023-09-26 17:24:30: MQTT publish: topic 'homeassistant/sensor/0x0015bc001b0252b1/voltage_phase_b/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"configuration_url":"http://10.20.0.39:8080//#/device/0x0015bc001b0252b1/info","identifiers":["zigbee2mqtt_0x0015bc001b0252b1"],"manufacturer":"Develco","model":"Wattle AMS HAN power-meter sensor (EMIZB-132)","name":"Strømmåler","sw_version":"4.0.1"},"device_class":"voltage","enabled_by_default":false,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/Strømmåler","object_id":"strømmåler_voltage_phase_b","origin":{"name":"Zigbee2MQTT","sw":"1.33.0","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/Strømmåler","unique_id":"0x0015bc001b0252b1_voltage_phase_b_zigbee2mqtt",`
**"unit_of_measurement":"V"**,`
"value_template":"{{ value_json.voltage_phase_b }}"}'`

`info  2023-09-26 17:24:30: MQTT publish: topic 'homeassistant/sensor/0x0015bc001b0252b1/voltage_phase_c/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"configuration_url":"http://10.20.0.39:8080//#/device/0x0015bc001b0252b1/info","identifiers":["zigbee2mqtt_0x0015bc001b0252b1"],"manufacturer":"Develco","model":"Wattle AMS HAN power-meter sensor (EMIZB-132)","name":"Strømmåler","sw_version":"4.0.1"},"enabled_by_default":false,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/Strømmåler","name":"Voltage phase C","object_id":"strømmåler_voltage_phase_c","origin":{"name":"Zigbee2MQTT","sw":"1.33.0","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/Strømmåler","unique_id":"0x0015bc001b0252b1_voltage_phase_c_zigbee2mqtt","value_template":"{{ value_json.voltage_phase_c }}"}'`

Koenkk, 2023-09-26:
Fixed in https://github.com/Koenkk/zigbee-herdsman-converters/pull/6154 which will be included in the next release.

closed 2023-09-26 by Koenkk