Koenkk/zigbee2mqtt #19090: ZM25TQ Roller Motor wont report position after some time using it

rafareusch, 2023-09-26:
### What happened?

I've been encountering this behavior for more than a year.
I have about 30 ZM25 Roller Motors (both [0.8n](https://www.zemismart.com/products/zm25-zb) and [2n](https://www.zemismart.com/products/zm25tqzb-2-25)) across three HA instances, each one using different coordinators.

Seems that after some days using these devices (can be anything from 1 to 30 days) one or more motors stop reporting position, all commands works perfectly fine, position_set cover.open, close etc. But the position never changes, I've seen the position stuck in both open and closed states. (the up/down limits also work fine)

I tried many Z2M versions (incluing the current and dev) and all have this behavior, also tried using other coordinators (SLZB-06) and reparing it with no success.
The only way (I know) to make it report position again is to cut the power and restore.

Haven't tried to mess with the reporting tab in Z2M (Can I force a position report there?)

### What did you expect to happen?

To the cover always report the position correctly.

### How to reproduce it (minimal and precise)

I can't reproduce it using an specific step-by-step, seems that the issue happens by using the motor normally.

### Zigbee2MQTT version

1.33.0

### Adapter firmware version

20210708

### Adapter

Sonoff Dongle-P

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/12731775/log.txt)

**Log Context:**
First I set many positions for motor 192_1_8
The entity never updated it's position, but the motor physically moved to the desired position (issue example)
Then I set many positions for motor 177_1_6
The entity correctly updated the position. (expected behavior example)

Hypfer, 2023-09-27:
> The only way (I know) to make it report position again is to cut the power and restore.

Cut the power to what exactly? The motor? Because if so, then that would strongly imply that the issue is part of the motor firmware and thus nothing z2m could do anything about.

rafareusch, 2023-09-27:
> > The only way (I know) to make it report position again is to cut the power and restore.
> 
> 
> 
> Cut the power to what exactly? The motor? Because if so, then that would strongly imply that the issue is part of the motor firmware and thus nothing z2m could do anything about.

Yes! to the device.

Could I force it send an update using the reporting feature?

Tks

rafareusch, 2023-10-10:
I wanted to add that I've been having this issue in other 2 HA&Z2M instances which isues these zemismart motors:

https://www.zemismart.com/products/zm25tqzb-2-25 (2n version)
https://www.zemismart.com/products/zm35-zb (10n version)

Both are also recognized as ZM25TQ in Z2M

rafareusch, 2024-04-08:
The issue still persists

closed 2025-01-12 by github-actions[bot]