Koenkk/zigbee2mqtt #19091: [Wrong device]: BHT-002-GCLZB detected as BHT-006GBZB

Roman-Ber, 2023-09-27:
### Link

https://zigbee.blakadder.com/Moes_BHT-002.html

### Model

BHT-002-GCLZB

### Description

Water heater thermostat

### Vendor

MOES

### Picture (link)

https://zigbee.blakadder.com/assets/images/devices/Moes_BHT-002.webp

### Database entry

"0xa4c138812c7ea141": {             "date_code": "",             "definition": {                 "description": "Smart heating thermostat",                 "exposes": [                     {                         "features": [                             {                                 "access": 3,                                 "description": "Enables/disables physical input on the device",                                 "label": "State",                                 "name": "state",                                 "property": "child_lock",                                 "type": "binary",                                 "value_off": "UNLOCK",                                 "value_on": "LOCK"                             }                         ],                         "label": "Child lock",                         "type": "lock"                     },                     {                         "access": 3,                         "description": "The delta between local_temperature and current_heating_setpoint to trigger Heat",                         "label": "Deadzone temperature",                         "name": "deadzone_temperature",                         "property": "deadzone_temperature",                         "type": "numeric",                         "unit": "°C",                         "value_max": 5,                         "value_min": 0,                         "value_step": 1                     },                     {                         "access": 3,                         "description": "Maximum temperature limit. Cuts the thermostat out regardless of air temperature if the external floor sensor exceeds this temperature. Only used by the thermostat when in AL sensor mode.",                         "label": "Max temperature limit",                         "name": "max_temperature_limit",                         "property": "max_temperature_limit",                         "type": "numeric",                         "unit": "°C",                         "value_max": 35,                         "value_min": 0                     },                     {                         "access": 3,                         "description": "Minimum temperature limit for frost protection. Turns the thermostat on regardless of setpoint if the tempreature drops below this.",                         "label": "Min temperature limit",                         "name": "min_temperature_limit",                         "property": "min_temperature_limit",                         "type": "numeric",                         "unit": "°C",                         "value_max": 5,                         "value_min": 1                     },                     {                         "features": [                             {                                 "access": 3,                                 "description": "Temperature setpoint",                                 "label": "Current heating setpoint",                                 "name": "current_heating_setpoint",                                 "property": "current_heating_setpoint",                                 "type": "numeric",                                 "unit": "°C",                                 "value_max": 35,                                 "value_min": 5,                                 "value_step": 1                             },                             {                                 "access": 1,                                 "description": "Current temperature measured on the device",                                 "label": "Local temperature",                                 "name": "local_temperature",                                 "property": "local_temperature",                                 "type": "numeric",                                 "unit": "°C"                             },                             {                                 "access": 3,                                 "description": "Offset to be used in the local_temperature",                                 "label": "Local temperature calibration",                                 "name": "local_temperature_calibration",                                 "property": "local_temperature_calibration",                                 "type": "numeric",                                 "unit": "°C",                                 "value_max": 30,                                 "value_min": -30,                                 "value_step": 0.1                             },                             {                                 "access": 3,                                 "description": "Mode of this device",                                 "label": "System mode",                                 "name": "system_mode",                                 "property": "system_mode",                                 "type": "enum",                                 "values": [                                     "off",                                     "heat"                                 ]                             },                             {                                 "access": 1,                                 "description": "The current running state",                                 "label": "Running state",                                 "name": "running_state",                                 "property": "running_state",                                 "type": "enum",                                 "values": [                                     "idle",                                     "heat",                                     "cool"                                 ]                             },                             {                                 "access": 3,                                 "description": "Mode of this device (similar to system_mode)",                                 "label": "Preset",                                 "name": "preset",                                 "property": "preset",                                 "type": "enum",                                 "values": [                                     "hold",                                     "program"                                 ]                             }                         ],                         "type": "climate"                     },                     {                         "access": 3,                         "description": "Select temperature sensor to use",                         "label": "Sensor",                         "name": "sensor",                         "property": "sensor",                         "type": "enum",                         "values": [                             "IN",                             "AL",                             "OU"                         ]                     },                     {                         "access": 3,                         "description": "Time of day and setpoint to use when in program mode",                         "features": [                             {                                 "access": 3,                                 "label": "Weekdays p1 hour",                                 "name": "weekdays_p1_hour",                                 "property": "weekdays_p1_hour",                                 "type": "numeric",                                 "unit": "h",                                 "value_max": 23,                                 "value_min": 0                             },                             {                                 "access": 3,                                 "label": "Weekdays p1 minute",                                 "name": "weekdays_p1_minute",                                 "property": "weekdays_p1_minute",                                 "type": "numeric",                                 "unit": "m",                                 "value_max": 59,                                 "value_min": 0                             },                             {                                 "access": 3,                                 "label": "Weekdays p1 temperature",                                 "name": "weekdays_p1_temperature",        [... cut ...]   "manufacturer": "_TZE204_aoclfnxz",             "model_id": "TS0601",             "network_address": 18769,             "power_source": "Mains (single phase)",             "supported": true,             "type": "Router"         }

### Notes

Two of five new BHT-002-GCLZB thermostates are detected as BHT-006GBZB. They are all the same. I did checked this with the MOES support.

It still works like the correct discovered thermostates and can be controlled. Currently it is only a cosmetic defect.

Koenkk, 2023-09-27:
Can you provide a screenshot of the z2m frontend -> device -> about page?

Roman-Ber, 2023-09-27:
The second one. 
[image]

Koenkk, 2023-09-27:
@dankarization in https://github.com/Koenkk/zigbee2mqtt/issues/18097 you posted that the `_TZE204_aoclfnxz` is a BHT-006GBZB, here it is stated that it is a BHT-002-GCLZB , what is correct?

Roman-Ber, 2023-09-27:
The "BHT-002-GCLZB" (which would be correct) is detected as a "BHT-006GBZB" (not correct). On the screenshot is the wrong one:

[image]

dankarization, 2023-09-29:
@Koenkk i typed what I had on the box. This was a new design thermostat, not the old one

dankarization, 2023-09-29:
By the way, technically, both models mentioned in Z2M are wrong, since all 3 heating types (gas boiler, electric heating and thermal head) of each design is marked as one specific model in Z2M. It would be correct to remove "GCLZB" and "GBZB" alltogether and just say BHT-002 and BHT-006
Or divide each of these 2 models by 3 more, what has no point because all 3 heating modes are the same in terms of control, only wiring and relays inside is different
[image]

closed 2023-10-03 by Koenkk

jvcsw, 2023-12-18:
Hi,

I don't know if the following problem is due to the changes you made, but now, when I put a temperature with decimals (16.5 or 22.5) the target temperature of the thermostat is set to 0.

This has been happening for a few weeks. I thought it was my fault, but if you change the temperature from the Zigbee2MQTT UI it also happens.