Koenkk/zigbee2mqtt #19092: model TS0601_temperature_humidity_sensor_1

arteck, 2023-09-27:
### What happened?

sdfvdfv

### What did you expect to happen?

dfvdfv

### How to reproduce it (minimal and precise)

dfv

### Zigbee2MQTT version

1.22.22

### Adapter firmware version

2022022

### Adapter

1234123

### Debug log

234

Koenkk, 2023-09-27:
Looks like an old zigbee-herdsman-converters version is used? Model is `TS0601_temperature_humidity_sensor` in your screenshot but should be `TS0601_temperature_humidity_sensor_1`

arteck, 2023-10-05:
strange TZE200_zl1kmjqx works with TS0601_temperature_humidity_sensor_2

rocky753, 2023-10-06:
with new herdsman version is the same:
[image]

but when i change TZE200_zl1kmjqx from TS0601_temperature_humidity_sensor_1 to TS0601_temperature_humidity_sensor_2 its fine. Only battery state isnt shown.

[image]

closed 2023-11-14 by github-actions[bot]

Smo-RBR, 2023-12-06:
Have this issue with TZE200_bjawzodf too. Works as TuYa TS0601_temperature_humidity_sensor_2