Koenkk/zigbee2mqtt #19093: Device offline problem while in range

unf0rg0tt3n, 2023-09-27:
### What happened?

Devices go randomly offline while LQI is at least above 140.
Tried to follow the steps from z2m FAQ.
Happens primarily to the IKEA bulbs and Gledopto RGBW controller.

Wasn't such a problem before, devices are well in range (also in range of a router device)

### How to reproduce it (minimal and precise)

Reboot and let it sit for a few hours or minutes. Devices go offline

### Zigbee2MQTT version

1.33.0

### Adapter firmware version

6.10.3.0 

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus V2

### Debug log

`Failed to read state of 'Tuin schutting 1' after reconnect (Read 0x90395efffe731a21/1 genOnOff(["onOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 51007 - 1 - 15 - 6 - 1 after 10000ms))`

`2023-09-27 14:02:58Failed to read state of 'Tuin schutting 2' after reconnect (Read 0x90395efffe5461eb/1 genOnOff(["onOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 29952 - 1 - 30 - 6 - 1 after 10000ms))`

lux73, 2023-09-27:
try to remove from z2m - restart z2m & re-pair the failing Devices

unf0rg0tt3n, 2023-09-27:
> try to remove from z2m - restart z2m & re-pair the failing Devices

That doesn't work unfortunately.

unf0rg0tt3n, 2023-09-28:
[image]

LQI 124, device offline devices further away (back of my garden) are still available.
After reboot I can use the devices, takes about 30 minutes to 1 hour and they go offline again.
Sometimes they come back, most of the times they don't. The bulbs are brand new 2 weeks old.

unf0rg0tt3n, 2023-10-07:
Plugged the controller in an USB2.0 port. Also move the controller 20cm further away from my server. 

LQI lower at 70 - 80, but devices don't go offline randomly.

closed 2023-10-07 by unf0rg0tt3n

bobobado, 2023-10-08:
This unfortunately did not work for me. I tried repositioning my hub but the one bulb is still not responding. All my other devices, including battery powered devices, have been very stable. Please advise.

unf0rg0tt3n, 2023-10-08:
> This unfortunately did not work for me. I tried repositioning my hub but the one bulb is still not responding. All my other devices, including battery powered, have been very stable. Please advise.

Did you try to replace the bulb for another one to test? Maybe switch it from another place

bobobado, 2023-10-08:
Thank you for the reply. I tried that as well. No change in behavior

bobobado, 2023-10-12:
This appears to have been an issue with my config. I turned on this problem bulb on a whim and despite my zigbee network not being open for new devices to join and me having forcefully removed the bulb previously, this bulb appeared in my devices. I then factory reset the bulb and removed it without forcing. I then readded and the bulb is responding as expected. Thank you for those who provided input. I will report back if the problems arise again.

bobobado, 2023-10-12:
Sadly, I spoke to soon. The device has become completely unresponsive. I can reconfigure via the UI but cannot control it anymore.

unf0rg0tt3n, 2023-10-12:
> Sadly, I spoke to soon. The device has become completely unresponsive. I can reconfigure via the UI but cannot control it anymore.

Could you please create a separate issue regarding your problem? Its better for problem solving.