Koenkk/zigbee2mqtt #19099: [New device support]: Neo NAS-AB02B02 will not connect to Zigbee2MQTT in home assistant.

wilfriedbaars, 2023-09-28:
### Link

https://nl.aliexpress.com/item/1005003658351897.html?spm=a2g0o.order_list.order_list_main.22.21ef79d2HhaEBd&gatewayAdapt=glo2nld

### Database entry

none

### Comments

I try to pair the Neo NAS-AB02B02 but it will not pair. so no database entry or so. the Neo NAS-AB02B2 is supported.

There is only a reset button and description say reset device and connect to the ZigBee network. so short or long press on reset it will not join the zigbee2mqtt network in my home assistant.

please help with NAS-AB02B02 to connect to the zigbee2mqtt in home assistant.

Thanks in advance.

Wilfried

CodeWilliamson, 2023-09-30:
I'm trying to pair this device as well and no luck. When I press and hold the reset button, I see a log message that the device has left but it never joined successfully to begin with so I'm confused.

CodeWilliamson, 2023-10-31:
@wilfriedbaars have you had any luck getting this device to pair ?

wilfriedbaars, 2023-10-31:
Hello CodeWilliamson,

I'm sorry but until now not able to pair :( also not many suggestions of how to pair.

Maybe you can help?

Regards,

Wilfried

brjhaverkamp, 2024-03-22:
I added a  NAS-AB02B2 successfully to my network a few months ago. 
Then it ended up in a drawer while Io updated my home automation and moved to home assistant.
Initially on the new device I was not able to make it appear again. 
But then I also had problems pairing an aqara button. That had never happened before. And I have several.
 I added a USB cable between my raspberry and the dongle because I read some posts about possible interference. 
And behold. Bothd devices joined without a hitch. I can recommend trying this out. Maybe it works for you as well.

closed 2024-10-20 by github-actions[bot]