Koenkk/zigbee2mqtt #19103: [New device support]: new Clone Tuya Thermostat TS601 _TZE200_rtrmfadk

tomiss288, 2023-09-28:
### Link

https://allegro.pl/oferta/sterownik-glowica-termostatyczna-zigbee-tuya-3xaa-14159389244

### Database entry

{"id":6,"type":"EndDevice","ieeeAddr":"0xa4c13805ba3c5b7e","nwkAddr":25888,"manufId":4417,"manufName":"_TZE200_rtrmfadk","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0011\u0000\u0000\u0000\u0000\u0011\u0000\u0000\u0000\u0000\u0011\u0000\u0000\u0000\u0000\u0011.\u0017�,f/\u0017�,\u0012","65506":56,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZE200_rtrmfadk","zclVersion":3,"appVersion":72,"modelId":"TS0601","powerSource":3}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1440156621},"lastSeen":1695897569268,"defaultSendRequestWhen":"immediate"}

### Comments

Hello, 
I received on warranty new thermostat which is exactly same like previous (_TZE200_z1tyspqw) but with different manufacturerName: "_TZE200_rtrmfadk".
In my opinion, it seems to use exactly the same converter as the other TS0601_thermostat_1 already known.
Could you pls help me take configuration from "_TZE200_z1tyspqw" to create external converter to test it ?

Koenkk, 2023-09-28:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/3993885d51baf82ed334afe0781c7d55

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

tomiss288, 2023-09-28:
> Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/3993885d51baf82ed334afe0781c7d55
>
>     * save this as file next to `configuration.yaml` as [...]

Hello,
It works perfectly.
Many thx for your very very quickly help and support.

Koenkk, 2023-09-28:
Integrated the changes, you can remove the ext converter after updating to the next release.

closed 2023-09-28 by Koenkk

tomiss288, 2023-09-28:
Sorry OT, but I removed previous thermostat from my Z2M but there's still error displayed: "Entity '0xa4c138a7322068fe' is unknown". Where I should delete this device permanently ?

Mikescotland, 2023-10-06:
After software update to 1.02 for these TRVs:
{modelID: 'TS0601', manufacturerName: '_TZE200_z1tyspqw'},
{modelID: 'TS0601', manufacturerName: '_TZE200_rtrmfadk'},

the converter is not working correctly. Missing "eco" and "comfort" settings, mode MANUAL does not work - always go to AUTO mode.
I made a converter that works for both with eco/comfort:
https://github.com/Koenkk/zigbee2mqtt/issues/9265

MicSchi, 2023-12-29:
Hello, I have a Avatto TRV10 that has the same fingerprint, but some additional data points: I put it as a new device request: https://github.com/Koenkk/zigbee2mqtt/issues/20303