Koenkk/zigbee2mqtt #19106: [New device support]: TY0201 _TZ3000_zl1kmjqx TUYA device - temperature and humidity sensor

Maginius, 2023-09-28:
### Link

https://de.aliexpress.com/item/1005005454153400.html?spm=a2g0o.order_list.order_list_main.11.46775c5fB6q725&gatewayAdapt=glo2deu

### Database entry

{"id":12,"type":"EndDevice","ieeeAddr":"0xa4c13891862ef1db","nwkAddr":21488,"manufId":4417,"manufName":"_TZ3000_zl1kmjqx","powerSource":"Battery","modelId":"TY0201","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,1,3,1026,1029],"outClusterList":[25],"clusters":{"msTemperatureMeasurement":{"attributes":{"measuredValue":2566}},"msRelativeHumidity":{"attributes":{"measuredValue":5707}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}},"manuSpecificTuya_2":{"attributes":{"53256":5}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":211,"stackVersion":2,"hwVersion":1,"zclVersion":3,"interviewCompleted":false,"meta":{},"lastSeen":1695916492284,"defaultSendRequestWhen":"immediate"}

### Comments

I have the EDGE version installed and I did not find this device in the "new devices" section. I am a novice Home Assistant user and I can't manage to create a converter on my own. Please add this device. Thanks a lot

fsaris, 2023-10-01:
@Maginius does your device (package) also states `Product model: IH-K009`?

Have similar device. The next fingerprint needs to be added to `IH-K009` in https://github.com/Koenkk/zigbee-herdsman-converters/blob/edf7a49e00feebd98557e14610f31711ab3151c2/src/devices/tuya.ts#L1827

```
 fingerprint: [{modelID: 'TY0201', manufacturerName: '_TZ3000_zl1kmjqx'}],
```

Maginius, 2023-10-01:
Yes, it also state Product mode: IH-K009

Koenkk, 2023-10-03:
Thanks for the pr @fsaris ! Will be included in todays release.

closed 2023-10-03 by Koenkk

fsaris, 2023-10-14:
ok, there is still 1 reported value that's not converted. But aren't able yet to determine what the value stands for.

`Received Zigbee message from '0xa4c138713477571e', type 'attributeReport', cluster 'manuSpecificTuya_2', data '{"53256":5}' from endpoint 1 with groupID null`

panteLx, 2023-12-07:
I guess this will work for my issue (#20056) as well? It's the same sensor with the same manufacturer but I did not get a zigbee model.

djcrawleravp, 2024-04-09:
Anyone with updating issues?.. It is connected but after a couple of minutes stops updating

Gman1988, 2024-10-06:
+1 any updates ?