Koenkk/zigbee2mqtt #19108: wrong module detection

apages2, 2023-09-29:
### Link

https://www.zigbee2mqtt.io/devices/067776.html

### Model

067726A

### Description

Netatmo wired shutter switch

### Vendor

LEGRAND

### Picture (link)

https://www.zigbee2mqtt.io/devices/067776.html

### Database entry

"device": {"applicationVersion": 0,"dateCode": " \u000020221202\u0000\u0000\u0000\u0000\u0000","hardwareVersion": 1,"manufacturerID": 4129,"manufacturerName": " Legrand","model": "K4027C/L4027C/N4027C/NT4027C","networkAddress": 13499,"powerSource": "Mains (single phase)","softwareBuildID": "002c","stackVersion": 66,"type": "Router","zclVersion": 2},

### Notes

Hello

my Legrand 067726A module is detected as BTICINO K4027C/L4027C/N4027C/NT4027C module. It seems to work on basic up/down/stop actions, but the LED remains flashing green and the specific "support position" configuration does not work (no calibration)

Best regards

Diaoul, 2023-10-02:
Same issue here but the led stays red in my case and I can't seem to get the rolling shutter move.

Diaoul, 2023-10-02:
database.db entry
```
{"id":54,"type":"Router","ieeeAddr":"0x0004740000a434c1","nwkAddr":56064,"manufId":4129,"manufName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":"Mains (single phase)","modelId":" Shutter SW with level control\u0000","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":514,"inClusterList":[0,3,4,5,15,64513,258],"outClusterList":[0,64513,5,25,258],"clusters":{"genBasic":{"attributes":{"appVersion":0}}},"binds":[{"cluster":15,"type":"endpoint","deviceIeeeAddress":"0x00124b00237dfb5a","endpointID":1},{"cluster":258,"type":"endpoint","deviceIeeeAddress":"0x00124b00237dfb5a","endpointID":1},{"cluster":3,"type":"endpoint","deviceIeeeAddress":"0x00124b00237dfb5a","endpointID":1}],"configuredReportings":[{"cluster":258,"attrId":8,"minRepIntval":1,"maxRepIntval":62000,"repChange":1}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":66,"hwVersion":4,"dateCode":" \u000020230221\u0000\u0000\u0000\u0000\u0000","swBuildId":"002c","zclVersion":2,"interviewCompleted":true,"meta":{"configured":-904845555},"lastSeen":1696267835803,"defaultSendRequestWhen":"immediate"}
```

Diaoul, 2023-10-02:
Trying out things... (papertrail)

ext_converter.js

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: [' Shutter SW with level control\u0000'],
    model: '067726A',
    vendor: 'Legrand',
    description: 'Netatmo wired shutter switch with level',
    fromZigbee: [fz.identify, fz.ignore_basic_report, fz.legrand_binary_input_moving, fz.cover_position_tilt, fz.legrand_led_in_dark],
    toZigbee: [tz.cover_state, tz.cover_position_tilt, tz.legrand_identify, tz.legrand_settingEnableLedInDark],
    ota: ota.zigbeeOTA,
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBinaryInput', 'closuresWindowCovering', 'genIdentify']);
        await reporting.currentPositionLiftPercentage(endpoint);
    },
    exposes: [e.cover_position(),
        e.binary('led_in_dark', ea.ALL, 'ON', 'OFF').withDescription(`Enables the LED when the power socket is turned off,
        allowing to see it in the dark`)],
    },
};
```

Diaoul, 2023-10-02:
Well that's it, this is the max I can do :grimacing: 
I'm not sure it changed anything ant BTicino and Legrand seem to actually be in the same basked device-wise.

apages2, 2023-10-03:
[https://github.com/Koenkk/zigbee-herdsman-converters/pull/6227](url)

Diaoul, 2023-10-04:
@apages2 does this solve the led blinking issue even after pairing?

apages2, 2023-10-04:
yes, but the push request has a problem because it is duplicated with the BTICINO module. Under discussion with koenkk

FabianMangold, 2023-10-09:
Hi @apages2 and @Diaoul ,

Basic support for 067776A has been added to the DEV branch today.
Can you please try this, and close this issue in case the issue is solved ?

Regarding the pairing mode, you may need to:
- Try to power cycle it (that worked for some of my devices)
- Unfortunately "Factory reset" the device and (there is a small dip-switch underneath the down button, next to the LED) and subsequently re-pair it once more.

Diaoul, 2023-10-21:
Wanted to test today but I cannot without a docker image and there is no image for dev builds :cry:

Diaoul, 2023-10-21:
Nevermind! It seems it's just in [another repository](https://hub.docker.com/r/zigbee2mqtt/zigbee2mqtt-edge-amd64/)!

Diaoul, 2023-10-21:
Seems to be recognized correctly :+1:  A few comments:
- I still have the red led staying even after reset then pairing
- There is no way to trigger the "calibration" from zigbee2mqtt it seems
- Manually triggering the calibration with the buttons has no effect

All in all, it does not seem that there is a lot of improvements compared to the previous situation with the exception of the correct detection.

Diaoul, 2023-10-21:
I'll sniff the traffic on the Legrand gateway and report back with the logs (in a few days).

FabianMangold, 2023-10-31:
Hi @Diaoul,

The latest DEV build contains an update, that features switching the calibration mode on the switch.
The behaviour you've described is somewhat odd, I cannot reproduce this on any of my device.

You may want to give this build a try, and attempt to switch the calibration modes. Switching to a simple "Up/Down/Stop" may help to "reset" the device to a normal state.

Diaoul, 2023-11-01:
Testing now, I'll report back when I've played a bit with the calibrations.

Diaoul, 2023-11-02:
So the options work, it triggers the calibration. But then, asking for a position just does 0 or 100% :shrug: 
Regarding the led, it's still blinking green like it was not paired.
I've not yet had the time to sniff the network with the Legrand gateway to see what the difference is.

FabianMangold, 2023-11-02:
Hi @Diaoul,
Well, first of all, your device must be paired, otherwise it would not answer to the requests sent by Z2M.
In order to request a position, the switch must first be "calibrated", as the switch has no actual knowledge of the actual position a blind in the first place! (Please keep in mind that there is no feedback loop / sensor, the switch remains a switch after all)!
I'll reply to similar questions in PR#6333 shortly (you may want to read through this as well), but in a nutshell, you will have to go through the calibration learning prior requesting a position.

When switching modes, the switch automatically enters "calibration mode". Therefore, i suppose that your switch is currently locked there. If i were you, i would deliberately mess up the calibration, and start over.
To do so, simply physically press and stay on the down button until the blind is down and you hear the relay click, then press down shortly, and then press and stay on the up button until the blind is fully open. This process must be repeated twice. Once done, the switch is back is "normal" mode, but with a messed up calibration.
I would then start the process over.
Can you please try this ? This worked just fine for all of my devices.

Sniffing the traffic might be interesting, albeit unrelevant at this stage, as the Legrand GW uses the very same commands and attributes to handle the situation.
Please let us know the outcome.

Diaoul, 2023-11-02:
> Well, first of all, your device must be paired

They are, although the led's pattern does not change from before to after pairing.

> In order to request a position, the switch must first be "calibrated"

I triggered a calibration by switching to "specific_nllv" which did the same thing as pressing the up + down button for 5 seconds when connected to the Legrand Gateway: the led turns orange and the auto calibration starts.

> you will have to go through the calibration learning prior requesting a position

With the Legrand gateway, the above calibration worked and I could request a position.

> Sniffing the traffic might be interesting, albeit unrelevant at this stage

It's really weird that my module still has the LED as if it was not paired but is paired nonetheless. I do not have this behavior when connected to the Legrand Gateway. I suspect this causes the module to behave weirdly and that I could learn more by sniffing the traffic.
If you did that already, let me know and I will not waste my time 😅

FabianMangold, 2023-11-02:
> They are, although the led's pattern does not change from before to after pairing.

That's really odd, i really cannot reproduce this. To be honest, i remember one similar situation when i first bought the switches a couple of month back, and i may have gotten rid of this by re-forcing a firmware update. Probably not the finest way, but given the fact i wanted to update anyway....
Might be worth trying this on one of your devices.

> I triggered a calibration by switching to "specific_nllv" which did the same thing as pressing the up + down button for 5 seconds when connected to the Legrand Gateway: the led turns orange and the auto calibration starts.

Interesting. I've added some notes here:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/6333#issuecomment-1790548665
Would you mind checking / confirming or obviously commenting on those findings ? I have no Legrand GW and thus my findings are reverse-engineering and trial and errors.
I have to check my config. once back home, but on top of my head, i am using the classic mode on my NLLV switches, not the specific mode. Would you know the difference ?
I am interested in the BSO mode anyway, therefore i have not lost much time on the NLLV modes really, but i am happy to assist if i can.

Diaoul, 2023-11-02:
FYI my firmware version is 0036 (on the Zigbee2MQTT dashboard)

FabianMangold, 2023-11-02:
Which is the latest, and that’s okay. I just meant you could reforce the update in order to go through an OTA cycle and see what happens. That worked back then for me (albeit I had updated from an super old FW version)

Diaoul, 2023-11-02:
I cannot capture the gateway's network key because when pairing there is no device announcement message :flushed: 
@Koenkk is there something I should do to capture the traffic specifically on Legrand's Gateway? Maybe there is some other kind of security?

When monitoring I see an Association Request from the module followed by an Association Response from the gateway then the following messages are encrypted :confused:

Diaoul, 2023-11-02:
Well, it turns out it uses a link key over APS and it's actually the default global trust center link key `5A 69 67 42 65 65 41 6C 6C 69 61 6E 63 65 30 39` as described [here](https://research.kudelskisecurity.com/2017/11/08/zigbee-security-basics-part-2/).
With this I was able to retrieve the actual network key used by the Legrand gateway :tada:

FabianMangold, 2023-11-02:
Regarding the firmware updates, you may want to take a look at this page:
https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#local-ota-index-and-firmware-files

Apologies for the confusion regarding the NLLV modes.
You are right, and i've update my comment accordingly.
The "classic" mode indeed seems to be a basic Up/Down/Stop mode, and the "specific" mode is the one supporting the auto-calibration. Just verified this on one of the switches.

Diaoul, 2023-11-02:
There is a capture of the binding step: [legrand.pcapng.zip](https://github.com/Koenkk/zigbee2mqtt/files/13244758/legrand.pcapng.zip)

I don't think that's an issue with my firmware, it works well with the Legrand Gateway, it must do something that is not done by the Zigbee2MQTT implementation :shrug:

Diaoul, 2023-11-02:
On the Legrand app, there are 3 modes for "Control mode":
- Aperture level
- Orientable sun shades
- Up / Down / Stop

When picking "Aperture level", you can then choose automatic calibration or manual calibration.

I think this matches well what you did in the newest PR (with the exception of the names)

FabianMangold, 2023-11-02:
Hi @Diaoul ,
The names and the cluster attributes were extracted from the Legrand documentation, we didn't make them up.
Z2M doesn't do any magic there, all that gets done is changing to calibration_mode. (Attribute 0xf002 from Cluster 0x102).

You'll find the document attached hereafter for reference.
[NLLV-cluster-list.xlsx](https://github.com/Koenkk/zigbee2mqtt/files/13245347/NLLV-cluster-list.xlsx)

Diaoul, 2023-11-03:
> To recap.: All 3 modes work with the Legrand App right ?

Yes, everything works with the Legrand app.

My main focus is to solve this LED issue, it either keeps blinking green as it was before pairing OR still red as it was before pairing.
I imagine there must be something that differs right after the pairing is done.

For example [this device](https://www.zigbee2mqtt.io/devices/067774.html) does not considers itself "configured" until bound to a group.
> As part of its configuration, the remote expects to have a "group" binding configured for each of its button. Otherwise, it will consider itself not fully configured and send unecessary commands on the zigbee network resulting in noticeable delay when controlling your device(s).

Maybe something similar is happening here I have no idea :shrug: 

> What exactly differs between Z2M and the Legrand GW ?

That is my next step, I'll run the same sniffing with my own coordinator and compare in a few days.

> as i have identical ones (10 of them actually)

I'd like to have more but first testing with one first so I have no comparison point. Glad you could get them working! No reason I cannot 💪

Maybe my problem is a coordinator issue :shrug: 
Coordinator: `tube_zb_gw_cc2652p2_poe`
Coordinator type: `zStack3x0`
Coordinator revision: `20230507`

To recap my next steps:
1. Sniff network on my own coordinator and compare the results
2. Force update the firmware on the module
3. Buy an additional one to compare

apages2, 2023-11-05:
Hello

 

For me, the latest DEV build is OK.

 

The module is detected in 067776A. It works correctly and the calibration is ok.

 

Adding the module to the network is a little difficult but ok after a few tries (resetting the module, then successive presses on the toothed wheel for 4 or 5 times when zigbee2mqtt scans)

 

Best regards

Diaoul, 2023-11-06:
I did not find anything super relevant in the capture of the network sadly, join is successful, binding as well, no error shown. I don't really even know what to look for so it makes it a bit hard :cry:
 
There must be something though, or something related to my coordinator I don't know as it is working on the gateway :angry: 

I'm no trying to force update the firmware on the device but I have this error:
```
Update of '0x0004740000a434c1' failed (Timeout: device did not request any image blocks)
```

I've order 4 other modules, see if they work otherwise I'll send them back.

FabianMangold, 2023-11-06:
Bonsoir @Diaoul,

That's really odd indeed.
Not sure if that helps, but i am using a "cheap" Sonoff Dongle-E, absolutely happy with it, not a single glitch.

I am stating the obivous, and i am quite sure you've done all of that already, but better safe than sorry:
- Factory Reset
- Long power off (electrically disconnected)
- and for the OTA, attempt to re-launch an OTA straight after the failure of the first one 
The latter is a bit of a "known" issue with the last firmware, that was noticed in another thread when the OTA process was fixed..

Also, can you confirm that your modules are Legrand 067776As ? I've seen that the hardware cersion reported is 4, whereas all of mines are having HW version 1. That "could" explain some differences.

Cheers

Diaoul, 2023-11-06:
I've tried the Factory Reset many many times, several times in a row and even reset -> cut power -> reset again
As for the OTA, I tried that but I've made many attempts in various conditions (right after reset / power cycle).
Nonetheless, I will try the long power cycling and re-attempt OTA after failure specifically.

> Also, can you confirm that your modules are Legrand 067776As ? I've seen that the hardware cersion reported is 4, whereas all of mines are having HW version 1. That "could" explain some differences.

* My module is `067726A` **not** `067776A`
* `hwVersion` is `4`

FabianMangold, 2023-11-08:
Hi @Diaoul,
Are you aware of this ? https://github.com/Koenkk/zigbee2mqtt/issues/16948#issuecomment-1680592070
From variour reports, it looks like the issue seems to be linked to the Hardware ID indeed, as we suspected.

You may want to take a look at the procedure described over there.

Cheers,

Diaoul, 2023-11-08:
I was not aware of that, trying now.

Diaoul, 2023-11-08:
Thanks for finding this, this is the exact same issue.

It seems to be partially working at least:
* the led is now solid green
* the rolling shutter moved up right after the second pairing
* I also could trigger the auto-calibration by pressing the buttons

Now I'll wait, see how it goes from there and report back in a few hours.

Diaoul, 2023-11-08:
No need to wait a few hours, the auto-calibration finished and everything seems to be working fine now :+1:

FabianMangold, 2023-11-08:
Amazing, so it was really a weird association issue it seems right ?
Happy to hear it is fixed. Credits go to the users in the thread, and thanks to @syntesys87 for mentioning it in another thread.
Not sure how we can debug this behaviour best really... Should we reach out to Legrand and report that issue ?
Is it really strange that it affects certain HW Versions only..

FabianMangold, 2023-11-08:
@apages2: Is  everything fixed on your end too ? If so, can we close that issue ?

Diaoul, 2023-11-08:
> Amazing, so it was really a weird association issue it seems right ?

Yes, I tried resetting many times but not with Zigbee2MQTT down. It's worth noting that the whole time my coordinator was up though so I'm not sure an issue with Zigbee2MQTT is to exclude entirely.

> Should we reach out to Legrand and report that issue ?

I believe yes. One lead maybe is that the association to the Legrand Gateway seems different as it does not expose the Network Key the regular way but relies on APS Link security first. Maybe the second association without Zigbee2MQTT noticing that the device left the network triggered another round of packet exchanges that was missing the first time?
The good thing is that now we've tied it to the HW version, Legrand will be able to reproduce and hopefully fix it.

syntesys87, 2023-11-09:
So the green led blinking problem was related to the hardware id exchanged on interview phase?

apages2, 2023-11-12:
Helloyes everything is ok for me on the latest version
you can close this number
THANKSEnvoyé de mon iPhone

FabianMangold, 2023-11-12:
Hi @apages2,

Thanks for confirming. Could you please close it, since you've opened it ?
Thanks a lot.

closed 2023-11-12 by apages2