Koenkk/zigbee2mqtt #19118: [New device support]:

wickedmkpl, 2023-09-29:
### Link

https://www.ikea.com/pl/pl/p/tradfri-zarowka-led-e27-1055-lumenow-smart-bezprzewodowy-przyciemniany-biale-spektrum-kula-60545673/

### Database entry

{"id":19,"type":"Router","ieeeAddr":"0xd87a3bfffe03b4a6","nwkAddr":39067,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI bulb E27 WS globe 1055lm","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096,64599,64636],"outClusterList":[25],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":106,"hwVersion":1,"dateCode":"20230421","swBuildId":"2.4.5","zclVersion":8,"interviewCompleted":true,"meta":{},"lastSeen":1696014415969,"defaultSendRequestWhen":"immediate"}

### Comments

IKEA LED2201G8 bulb

theodiem, 2023-10-02:
Just came from ikea today to find this issue :'( ...
I can see the device in the touchlink scan, but it doesn't join my network (join enabled) even after a reset (either way with touchlink or with 5x power on/off) . Is the expected behavior when a device is not supported?

VRennebergBK, 2023-10-03:
Hello, for me it took some attempts. And it is 6!!! Turn on 6 times and leave it on on the 6th time. It joins the network, I was able to configure a binding from a switch to this lamp. But zigbee2mqtt does not know it, yet. Cheers Volker 

> Just came from ikea today to find this issue :'( ... I can see the device in the touchlink scan, but it doesn't join my network (join enabled) even after a reset (either way with touchlink or with 5x power on/off) . Is the expected behavior when a device is not supported?

Koenkk, 2023-10-03:
Please add the color temperature range as described on https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_3-adding-converter-s-for-your-device

closed 2023-10-03 by Koenkk

theodiem, 2023-10-03:
Many thanks for adding the device. 
I was going to give a shot on the converters for the weekend. Looking into the changed files (and all the imports/references) helps a lot to understand.

nbduck, 2023-11-14:
> Just came from ikea today to find this issue :'( ... I can see the device in the touchlink scan, but it doesn't join my network (join enabled) even after a reset (either way with touchlink or with 5x power on/off) . Is the expected behavior when a device is not supported?

Somehow its still not working for me. I have the same problem.

Zigbee2MQTT-Version
    [1.33.2](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.33.2) commit: [unknown](https://github.com/Koenkk/zigbee2mqtt/commit/unknown)

Coordinator-Typ
    zStack3x0

Coordinator-Version
    20220219

Coordinator IEEE Adresse
    0x00124b00258db86e

Frontend-Version
    0.6.142

Zigbee-herdsman-converters version
    15.106.0

Zigbee-herdsman version
    0.21.0