Koenkk/zigbee2mqtt #19119: Again: Error: {"sequence":-1} after 10000ms

Ankarii, 2023-09-30:
### What happened?

Zigbee2mqtt does not start in Home Assistant Operating System with Sonoff ZBDongle-E

### Zigbee2MQTT version

1.33.0

### Adapter firmware version

7.3.1.0 build 176

### Adapter

Sonoff ZBDongle-E

### Debug log

Zigbee2MQTT:info  2023-09-30 09:11:03: Starting zigbee-herdsman (0.18.5)
Error: Reset error: Error: {"sequence":-1} after 10000ms
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:299:23
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)

Ankarii, 2023-09-30:
**hassio/system log:**
```
23-09-30 10:06:08 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on zigbee2mqtt/zigbee2mqtt-amd64 with version 1.33.0-1
23-09-30 10:06:09 ERROR (MainThread) [supervisor.api.ingress] Ingress error: Cannot connect to host 172.30.33.1:8099 ssl:default [Connect call failed ('172.30.33.1', 8099)]
23-09-30 10:06:21 ERROR (MainThread) [asyncio] Task exception was never retrieved
future: <Task finished name='Task-1693' coro=<Addon.watchdog_container() done, defined at /usr/src/supervisor/supervisor/addons/addon.py:1069> exception=AddonsJobError('Rate limit exceeded, more then 10 calls in 0:30:00')>
Traceback (most recent call last):
  File "/usr/src/supervisor/supervisor/addons/addon.py", line 1083, in watchdog_container
    await self._restart_after_problem(event.state)
  File "/usr/src/supervisor/supervisor/jobs/decorator.py", line 255, in wrapper
    raise on_condition(
supervisor.exceptions.AddonsJobError: Rate limit exceeded, more then 10 calls in 0:30:00
23-09-30 10:06:21 ERROR (SyncWorker_2) [supervisor.docker.manager] Container addon_45df7312_zigbee2mqtt is not running
```

Ankarii, 2023-09-30:
In debug mode:
`[16:44:38] INFO: Preparing to start...
[16:44:39] INFO: Socat not enabled
[16:44:39] INFO: Zigbee Herdsman debug logging enabled
[16:44:39] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:debug 2023-09-30 16:44:40: Loaded state from file /config/zigbee2mqtt/state.json
Zigbee2MQTT:info  2023-09-30 16:44:40: Logging to console and directory: '/config/zigbee2mqtt/log/2023-09-30.16-44-40' filename: log.txt
Zigbee2MQTT:debug 2023-09-30 16:44:40: Removing old log directory '/config/zigbee2mqtt/log/2023-09-30.16-34-16'
Zigbee2MQTT:info  2023-09-30 16:44:40: Starting Zigbee2MQTT version 1.33.0 (commit #unknown)
Zigbee2MQTT:info  2023-09-30 16:44:40: Starting zigbee-herdsman (0.18.5)
Zigbee2MQTT:debug 2023-09-30 16:44:40: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[61,62,128,100,30,142,33,236],"networkKey":"HIDDEN","panID":40290},"serialPort":{"adapter":"ezsp","path":"/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20230803181737-if00"}}'
2023-09-30T11:44:40.924Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'
2023-09-30T11:44:40.925Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[129,3,117,159,163,88,122,186,193,177,219,61,196,161,81,69],"panID":40290,"extendedPanID":[61,62,128,100,30,142,33,236],"channelList":[11]},"serialPort":{"path":"/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20230803181737-if00","adapter":"ezsp"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
2023-09-30T11:44:40.926Z zigbee-herdsman:adapter:ezsp:uart Opening SerialPort with {"path":"/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20230803181737-if00","baudRate":115200,"rtscts":false,"autoOpen":false}
2023-09-30T11:44:40.929Z zigbee-herdsman:adapter:ezsp:uart Serialport opened
2023-09-30T11:44:40.929Z zigbee-herdsman:adapter:ezsp:uart Uart reseting
2023-09-30T11:44:40.929Z zigbee-herdsman:adapter:ezsp:uart --> Write reset
2023-09-30T11:44:40.930Z zigbee-herdsman:adapter:ezsp:uart --> [1ac038bc7e]
2023-09-30T11:44:40.931Z zigbee-herdsman:adapter:ezsp:uart -?- waiting reset
2023-09-30T11:44:42.018Z zigbee-herdsman:adapter:ezsp:uart <-- [c1020b0a527e]
2023-09-30T11:44:42.018Z zigbee-herdsman:adapter:ezsp:uart <-- RSTACK: c1020b0a527e
2023-09-30T11:44:42.019Z zigbee-herdsman:adapter:ezsp:uart RSTACK Version: 2 Reason: RESET_SOFTWARE frame: c1020b0a527e
2023-09-30T11:44:42.019Z zigbee-herdsman:adapter:ezsp:uart -+- waiting reset success
2023-09-30T11:44:42.020Z zigbee-herdsman:adapter:ezsp:ezsp ==> version: {"desiredProtocolVersion":4}
2023-09-30T11:44:42.020Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"version","_id_":0,"_isRequest_":true,"desiredProtocolVersion":4}
2023-09-30T11:44:42.020Z zigbee-herdsman:adapter:ezsp:uart --> DATA (0,0,0): 00000004
2023-09-30T11:44:42.021Z zigbee-herdsman:adapter:ezsp:uart --> [004221a850ed2c7e]
2023-09-30T11:44:42.021Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (1)
2023-09-30T11:44:42.025Z zigbee-herdsman:adapter:ezsp:uart <-- [0142a1a8582805c1ae247e]
2023-09-30T11:44:42.025Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (0,1,0): 0142a1a8582805c1ae247e
2023-09-30T11:44:42.025Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (1)
2023-09-30T11:44:42.025Z zigbee-herdsman:adapter:ezsp:uart --> [8160597e]
2023-09-30T11:44:42.026Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 0080000c021073
2023-09-30T11:44:42.026Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x0: {"_cls_":"version","_id_":0,"_isRequest_":false,"protocolVersion":12,"stackType":2,"stackVersion":29456}
2023-09-30T11:44:42.026Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (1) success
2023-09-30T11:44:42.026Z zigbee-herdsman:adapter:ezsp:ezsp Switching to eszp version 12
2023-09-30T11:44:42.027Z zigbee-herdsman:adapter:ezsp:ezsp ==> version: {"desiredProtocolVersion":12}
2023-09-30T11:44:42.027Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"version","_id_":0,"_isRequest_":true,"desiredProtocolVersion":12}
2023-09-30T11:44:42.027Z zigbee-herdsman:adapter:ezsp:uart --> DATA (1,1,0): 01000100000c
2023-09-30T11:44:42.027Z zigbee-herdsman:adapter:ezsp:uart --> [7d314321a9542a1989fb7e]
2023-09-30T11:44:42.027Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (2)
2023-09-30T11:44:44.028Z zigbee-herdsman:adapter:ezsp:uart --> Error: Error: {"sequence":2} after 2000ms
2023-09-30T11:44:44.028Z zigbee-herdsman:adapter:ezsp:uart -!- break waiting (2)
2023-09-30T11:44:44.028Z zigbee-herdsman:adapter:ezsp:uart Can't send DATA frame (1,1,0): 01000100000c
2023-09-30T11:44:44.529Z zigbee-herdsman:adapter:ezsp:uart ->> DATA (1,1,1): 01000100000c
2023-09-30T11:44:44.529Z zigbee-herdsman:adapter:ezsp:uart --> [194321a9542a197d3a567e]
2023-09-30T11:44:44.530Z zigbee-herdsman:adapter:ezsp:uart -?- rewaiting (2)
2023-09-30T11:44:46.530Z zigbee-herdsman:adapter:ezsp:uart --> Error: Error: {"sequence":2} after 2000ms
2023-09-30T11:44:46.531Z zigbee-herdsman:adapter:ezsp:uart -!- break rewaiting (2)
[... cut ...]
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:562:23
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
2023-09-30T11:44:46.576Z zigbee-herdsman:adapter:ezsp:uart Port closed
2023-09-30T11:44:46.576Z zigbee-herdsman:adapter:ezsp:ezsp Close ezsp
2023-09-30T11:44:46.576Z zigbee-herdsman:adapter:ezsp:driv Close driver
2023-09-30T11:44:46.576Z zigbee-herdsman:adapter:ezsp:ezsp Close ezsp
2023-09-30T11:44:46.576Z zigbee-herdsman:adapter:ezsp:driv Close driver`

xhemp, 2023-10-13:
I have this very same error with the latest Zigbee2MQTT:

 Error: Failure send version:{"type":"Buffer","data":[1,0,1,0,0,12]}

If I reboot the box for whatever reason, when the HA OS comes back, Zigbee2MQTT doesn't come back up automatically, it fails with error above.
I had to try and restart it twice, manually, and it came back up.

lazmo88, 2023-10-17:
Same error, using Skyconnect dongle. It seems latest updated have broken something with the ezsp adapters?

xelemorf, 2023-10-28:
> Same error, using Skyconnect dongle. It seems latest updated have broken something with the ezsp adapters?
> 
> Edit: removing recommended baudrate and rtscts values from the config allowed HA addon to start normally.
> 
> Values removed from the config: baudrate: 115200 rtscts: true

I am using Dongle-E, never had those config items added on my side.
For ref, my conf below.

```
data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt:
  server: mqtt://core-mosquitto:1883
  user: 
  password: 
serial:
  port: >-
    /dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20230324141749-if00
  adapter: ezsp
```

xhemp, 2023-10-28:
> I am using Dongle-E, never had those config items added on my side.
> For ref, my conf below.
>
> ```
> data_path: /config/zigbee2mqtt
> socat:
>   enabled: false
>   master: [...]

Same here.
Still have issues if I restart the HA OS.
Takes me 2 or 3 restarts to get z2m to start properly.

xelemorf, 2023-10-29:
There was a confuision on my side apparently on the config part yesterday, so there are two configuration places
- Z2M Add On configuration
- Z2M configuration.yaml

The configuration.yaml file indeed has the below properties for me:
-   baudrate: 115200
-   rtscts: true

For the Z2M restarts, I have similar issues but for different reasons generally because using ZBDongle-E and by the recent updates in HAOS and Core which made everything highly unstable when using ZBDongle-E, anyway the point is, this below article helped me in the adventure to somewhat stabilize things, at least I don't need to restart Z2M manually now.

In this article there is a Watchdog automation, and I have posted a script based on that aswell which makes things less miserable: 
https://community.home-assistant.io/t/tenacious-watchdog-for-add-ons-example-for-zigbee2mqtt/521888

xhemp, 2023-10-29:
> There was a confuision on my side apparently on the config part yesterday, so there are two configuration places
> 
> * Z2M Add On configuration
> * Z2M configuration.yaml
> 
> The configuration.yaml file indeed has the below properties for me:
> 
> * baudrate: 115200
> * rtscts: true
> 
> For the Z2M restarts, I have similar issues but for different reasons generally because using ZBDongle-E and by the recent updates in HAOS and Core which made everything highly unstable when using ZBDongle-E, anyway the point is, this below article helped me in the adventure to somewhat stabilize things, at least I don't need to restart Z2M manually now.
> 
> In this article there is a Watchdog automation, and I have posted a script based on that - these two will make things less miserable: https://community.home-assistant.io/t/tenacious-watchdog-for-add-ons-example-for-zigbee2mqtt/521888

Thanks for that, until we find out why this happens, I thin it's a great way of certifying it's running.

pbassut, 2023-11-24:
I'm having the same issue. Any solutions?

xelemorf, 2023-11-26:
> I'm having the same issue. Any solutions?

Forget Sonoff Zigbee Dongle-**E** to be used as Zigbee coordinator, note that it's still experimental which (still) means as no adequate support on it. The device itself can used as router though, and it's actually quite good at it.

Buy a Dongle-**P** diectly from Sonoff on AliExpress.
Please refer to my posts in this article for all relevant details on how to get started with Dongle-P:
https://github.com/Koenkk/zigbee2mqtt/issues/17412#issuecomment-1803690948

nicola, 2023-12-25:
Did you manage to resolve this?

changyeamoon, 2024-01-13:
I ended up trying all the adapter options until it worked! I used adapter: zstack to get it working!

All options in the documentation - https://www.zigbee2mqtt.io/guide/configuration/adapter-settings.html#configuration-of-the-zigbee-adapter

mortab, 2024-01-13:
Not sure if this is any help to anyone, but my Zigbee2MQTT started working after I disabled the default "SONOFF Zigbee 3.0 USB Dongle Plus" integration that is setup when you first connect the dongle to the system.  Possibly the two integrations not willing to share the device?

adde23, 2024-01-13:
> Not sure if this is any help to anyone, but my Zigbee2MQTT started working after I disabled the default "SONOFF Zigbee 3.0 USB Dongle Plus" integration that is setup when you first connect the dongle to the system. Possibly the two integrations not willing to share the device?

Thanks!!! This helped me after hours of frustration. I disabled the SONOFF Zigbee integration and then Zigbee2 MQTT started without any errors.

joepresti, 2024-01-14:
> There was a confuision on my side apparently on the config part yesterday, so there are two configuration places
>
> * Z2M Add On configuration
> * Z2M configuration.yaml
>
> The configuration.yaml file [...]

Excellent! It worked for me changing the config file /homeassistant/zigbee2mqtt/configuration.yaml

changyeamoon, 2024-01-14:
@mortab are you talking about the integration inside home assistant?

mortab, 2024-01-14:
> @mortab are you talking about the integration inside home assistant?
> 
> I still get trouble with home assistant not being able to find the sonoff usb when my computer restarts

@changyeamoon Yes HA was able to see the Sonoff Zigbee dongle when I connected on my setup which HA added the integration for me. 
 
[image]

kihaki, 2024-01-16:
I fixed my issues by editing the configs in the file editor instead of the zigbee2mqtt addon page, somehow the files did not get saved correctly from there.

Essentially this: https://www.reddit.com/r/homeassistant/comments/1926wbb/zigbee2mqtt_with_sonoff_dongle_30p_not_working/

Urikadze, 2024-01-28:
> Not sure if this is any help to anyone, but my Zigbee2MQTT started working after I disabled the default "SONOFF Zigbee 3.0 USB Dongle Plus" integration that is setup when you first connect the dongle to the system. Possibly the two integrations not willing to share the device?

It work!!!!

closed 2024-08-27 by github-actions[bot]

githuballrutz, 2024-10-01:
> I ended up trying all the adapter options until it worked! I used adapter: zstack to get it working!
> 
> All options in the documentation - https://www.zigbee2mqtt.io/guide/configuration/adapter-settings.html#configuration-of-the-zigbee-adapter

adapter: zstack did it for me too. I don't get it, the log tells me to use "ember".

baabercrombie, 2025-01-07:
> I ended up trying all the adapter options until it worked! I used adapter: zstack to get it working!
> 
> All options in the documentation - https://www.zigbee2mqtt.io/guide/configuration/adapter-settings.html#configuration-of-the-zigbee-adapter

I love you. This worked for me. Finally the last straw in getting my Zigbee setup back up and running fully after being down with various issues for 3 days.

marcvangend, 2025-04-23:
> There was a confuision on my side apparently on the config part yesterday, so there are two configuration places
> 
> * Z2M Add On configuration
> * Z2M configuration.yaml
> 
> The configuration.yaml file indeed has the below properties for me:
> 
> * baudrate: 115200
> * rtscts: true

Yes! Finally after many hours of trying different configurations, zigbee addons, firmware versions and what not, this comment put me on the right path. Once I had deleted the `baudrate` and `rtscts` properties from `config/zigbee2mqtt/configuration.yaml`, Zigbee2MQTT was able to start up. Previously, it kept failing with `z2m: Error while starting zigbee-herdsman` and `Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.` errors.