Koenkk/zigbee2mqtt #19120: [Feature request]: Turn off backlight of Tyua TS0001, TS0002, TS130F

SmolikOne, 2023-09-30:
### Is your feature request related to a problem? Please describe

Three is no possibility to turn off backlight of Tuya based devices TS0001, TS0002, TS130F and my wife wants to kill me :)

### Describe the solution you'd like

Add a option to turn off backlight

### Describe alternatives you've considered

 I didn't found any

### Additional context

I'm new with Z2M and this forum,but when I searched the topics I didn't found how to solve this problem, so I apologize if the solution is somewhere. On the exposed section of the mentioned devices there is a option to set the backlight to low, med, high, but the only thing that is changing are colors of led's from red to blue. No change incintensivity of led's. On the original Tyua environment and it's app from where I migrated recently, there was a option to turn off backlight completly of this devices. I really appriciate your help here, as my wife can't sleep due to lights that are glowing directly on her :)

hardshell92DE, 2023-10-03:
@Koenkk could you look into this? I have the same problem with TS130F. In the settings there
[image]
 is only the possibility to set the backlight_mode to low. But indeed you should have the possibility to set it off like in the tuya app.

PedroLageTavares, 2024-01-31:
I've bought a couple switches from the same seller in AliExp and the ones being identified as TS0001 (_TZ3000_ehgouyvu) are not exposing the "Led indicator mode" option (as opposing with the TS0002 / TZ3000_aa5t61rh).

I did pair the TS0001 with Tuya Zigbee controller (TYGWZ-01) and there I can control the led indicator. 

If anything else is required from my side, let me know and I will do my best to provide it.

ty

Trebor87, 2024-05-21:
Hello 

Any update on this 

Like a previous comment, I'm getting stick as my wife hates lights in the bedroom, and as the curtain switch backlight is always on....

In fact not only can you not turn off the backlight but also they three settings high, medium or low do not work 

Many thanks

hardshell92DE, 2024-05-21:
For me now the toggle buttons appear, but it seems they are not working so I can't shut off the damn leds. This cryptic error message is shown, someone could help fixing this please?

> Publish 'get 'backlight_mode' to 'Rolladen Gästezimmer' failed: 'Error: ZCL command Oxa4c138c38a264a78/1 genOnOff.read(['tuyaBacklightSwitch],
('timeout":10000,'disableResponse":false'di sableRecovery":false 'disableDefaultRespon se"true,'direction":0,'srcEnd point":null,"reser vedBits":0,'manufacturerCode"null transact ionSequenceNumber':null'writeUndiv':false failed (Status "UNSUPPORTED LATTRIBUTE)'

> Exception while calling fromZigbee converter: Expected string or number, got: undefined}

ausfas, 2024-10-21:
Have you guys tried this option as this doesn't work for me 
https://community.home-assistant.io/t/manually-setting-backlight-attribute-on-zigbee-moes-tuya-lightswitch-with-zha-toolkit-ts0014/575802

ausfas, 2025-01-01:
Any solution yet?

closed 2025-03-11 by github-actions[bot]